pyvisa/pyvisa #609: WARNING Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::22102::2114::???::0::INSTR

RobinTry, 2021-06-26:
WARNING Found a device whose serial number cannot be read. The partial VISA resource name is: USB0::22102::2114::???::0::INSTR

Instrument details
------------------
* Model:UPO7104Z
* Communication: USB

DavidLutton, 2021-06-26:
What operating system are you running?

I had an USB related permissions issue on Linux, that needed some udev rules to work.

RobinTry, 2021-06-28:
ARMv7 Processor rev 4 (v7l)
Raspbian  Debian linux

RobinTry, 2021-06-28:
Use NI to check , result: USB0::0x5656::0x0842::e415f6c928a7::instr

LongnoseRob, 2021-06-28:
What kind of instrument is that?
could you please also give us the compelete output of pyvisa-info ?

DavidLutton, 2021-06-30:
Run `id` to check what groups your login user is in. I've used 'plugdev'.
udev rules to create/add to "/lib/udev/rules.d/51-usbtmc.rules"
```
KERNEL=="usbtmc[0-9]", MODE="0770",  GROUP="plugdev"
SUBSYSTEM=="USB", MODE="0770", GROUP="plugdev"
```
These are probably excessively permissive permissions, [Python usbtmc](https://github.com/python-ivi/python-usbtmc/blob/master/README.md) implies that "0660" is sufficient

Reload rules with `udevadm control --reload-rules`  
Disconnect and reconnect the device, to trigger the rules. 

Using the first rule got me to your current state, adding the SUBSYSTEM=="USB" rule got it working for users in the plugdev group.

MatthieuDartiailh, 2021-06-30:
@DavidLutton Would you mind making a PR to add this to the pyvisa-py docs ? Your explanation seems more comprehensive that what we have.

RobinTry, 2021-07-02:
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
import pyvisa
rm = pyvisa.ResourceManager()
rm.list_resources()
Out[29]: ('ASRL/dev/ttyAMA0::INSTR', 'USB0::22102::2114::e415f6c928a7::0::INSTR')
instr = rm.open_resource('USB0::22102::2114::e415f6c928a7::0::INSTR')
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/IPython/core/interactiveshell.py", line 2878, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-30-03adab51d425>", line 1, in <module>
    instr = rm.open_resource('USB0::22102::2114::e415f6c928a7::0::INSTR')
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/highlevel.py", line 3304, in open_resource
    res.open(access_mode, open_timeout)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/resources/resource.py", line 298, in open
    self._resource_name, access_mode, open_timeout
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/highlevel.py", line 3232, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa_py/highlevel.py", line 167, in open
    sess = cls(session, resource_name, parsed, open_timeout)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa_py/sessions.py", line 323, in __init__
    self.after_parsing()
  File "/usr/local/lib/python3.7/dist-packages/pyvisa_py/usb.py", line 84, in after_parsing
    self.parsed.serial_number,
  File "/usr/local/lib/python3.7/dist-packages/pyvisa_py/protocols/usbtmc.py", line 292, in __init__
    self.usb_dev.reset()
  File "/usr/local/lib/python3.7/dist-packages/usb/core.py", line 949, in reset
    self._ctx.backend.reset_device(self._ctx.handle)
  File "/usr/local/lib/python3.7/dist-packages/usb/backend/libusb1.py", line 911, in reset_device
    _check(self.lib.libusb_reset_device(dev_handle.handle))
  File "/usr/local/lib/python3.7/dist-packages/usb/backend/libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 2] Entity not found

RobinTry, 2021-07-02:
Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0] on linux
import pyvisa
rm = pyvisa.ResourceManager()
rm.list_resources()
Out[32]: ('ASRL/dev/ttyAMA0::INSTR', 'USB0::6833::1603::DG9A212400326::0::INSTR')
instr = rm.open_resource('USB0::6833::1603::DG9A212400326::0::INSTR')
print(instr.query("*IDN?"))
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/IPython/core/interactiveshell.py", line 2878, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-34-e57dac15f296>", line 1, in <module>
    print(instr.query("*IDN?"))
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/resources/messagebased.py", line 644, in query
    return self.read()
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/resources/messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/resources/messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa_py/highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

MatthieuDartiailh, 2021-07-02:
Regarding your second error, you probably have to set the termination character used when writing ans reading. The manual of your instrument should have that information and pyvisa documentation has a section dedicated to how to debug that kind of issue.

LongnoseRob, 2021-07-02:
I think the problem is that the manual is at the moment only available in Chinese..
Here is a link: http://www.uni-trend.com.cn/uploadfile/2018/1120/20181120082510698.pdf
I had a quick look across it, but did not see any quick reference in regards to VISA or remote commands.

The UPO7000-series is not listed (yet) on the english page of unit-trend.
Maybe it shares some of the VISA interface/command structure with other products  series.

The English manual (https://www.uni-trend.com/uploadfile/cloud/English%20manual/Benchtop%20Instrument/UPO3000E%20English%20Manual.pdf) does not mention anything about VISA complience.

Maybe it would be good to snoop the communication between the instrument and the manufacturers orginal PC-software,
to see if this unit is actually talking VISA.

RobinTry, 2021-07-03:
> Regarding your second error, you probably have to set the termination character used when writing ans reading. The manual of your instrument should have that information and pyvisa documentation has a section dedicated to how to debug that kind of issue.

> I think the problem is that the manual is at the moment only available in Chinese..
> Here is a link: http://www.uni-trend.com.cn/uploadfile/2018/1120/20181120082510698.pdf
> I had a quick look across [...]

> I think the problem is that the manual is at the moment only available in Chinese..
> Here is a link: http://www.uni-trend.com.cn/uploadfile/2018/1120/20181120082510698.pdf
> I had a quick look across [...]

> I think the problem is that the manual is at the moment only available in Chinese..
> Here is a link: http://www.uni-trend.com.cn/uploadfile/2018/1120/20181120082510698.pdf
> I had a quick look across [...]

UPO7000 is a new product, I have consulted UNI-T technology, and Visa is supported.Test on Win10, NI can work normally.

DavidLutton, 2021-09-22:
Did some more work with a Linux host and getting permissions to talk to a USBTMC device.

Run `id` to check what groups your login user is in. I've used 'plugdev'.
udev rules to create in "/usr/lib/udev/rules.d/51-usbtmc.rules"

``` shell
DRIVERS=="usbtmc", GROUP="plugdev"

SUBSYSTEM=="usb", ATTRS{manufacturer}=="Keysight Technologies", GROUP="plugdev"
SUBSYSTEM=="usb", ATTRS{manufacturer}=="Agilent Technologies", GROUP="plugdev"
```

## udev 
Before you connect run `sudo udevadm monitor -e  | grep DEVNAME` then connect to find out what /dev/ are being created for your USBTMC device 
Find ATTRS of a given device `sudo udevadm info --name=/dev/bus/usb/003/026` including manufacturer
Check rule files are valid `sudo udevadm test 2>&1 | grep tmc`
Reload rules with `sudo udevadm control --reload-rules`  **then**
Check new rules grant permissions you expect `sudo udevadm trigger ; ls -l  /dev/bus/usb/*/ ;  ls -l /dev/usbtmc0`

## Note on USBTMC devices
USBTMC devices show up as **both** /dev/usbtmcN and /dev/bus/usb/NNN/NNN
Both instances need to have suitable permissions for pyvisa with pyvisa-py to be able to communicate

If only the /dev/usbtmcN has permissions then you get `WARNING Found a device whose serial number cannot be read`

@MatthieuDartiailh I will test this with the few USBTMC devices I have access to, with intent to create a PR on pyvisa-py.

MatthieuDartiailh, 2021-09-23:
Sounds good @DavidLutton Thanks for looking into it !