pyvisa/pyvisa #610: PyVisa Itself as a Remote Resource?

thestumbler, 2021-06-30:
I'm doing some simple tests using a single GPIB bench top multimeter (Keithley 195A). I have a Raspberry Pi Zero configured with a [GPIB adaptor](https://github.com/elektronomikon/raspizero_gpib_shield). All this is working locally from the Pi Zero... I can communicate and control the DMM using PyVisa just fine.

In my immediate application, I only need a very limited subset of functions from the meter, so I plan to query it from a host test computer over Ethernet using UDP or maybe TCP packets. As I was thinking about this, and reading the PyVisa documentation, I came across this resource type: `visa://host address[:server port]/remote resource`. That got me thinking... is there any way to configure PyVisa to be one of these remote servers? Kind of a bridge between whatever is connected on the local computer (the Pi Zero in my case) to the central test computer which is also running PiVisa? 

Note: I said Ethernet and I'm using a Pi Zero... that is correct. I'm using a USB-Ethernet dongle at the moment, because I found a bug in the WiFi radio firmware which renders it useless in South Korea. Waiting on Cypress to make an update, and until then wired LAN it is for my project.

MatthieuDartiailh, 2021-07-25:
Sorry for the delay. 

Some VISA implementation supports remote VISA resources, however those resource names are not supported by PyVISA currently and PyVISA does not offer a server to act as a sort of remote VISA. I suspect I will be easier to simply proxy what you need rather than having a remote VISA server. The VISA protocol is rather low level and if speed is a concern having something at a higher level inducing less traffic will likely be a better option.

thestumbler, 2021-07-25:
Thanks for the reply. I’m doing just that on my project, which turned out to be surprisingly simple. I’m using an old Keithley 195A. I only need a handful of commands like Read, Write and Query, sent using UDP packets.