pyvisa/pyvisa #612: [COM] Communication issue:sample a long period waveform cannot completed with  'MEAS:ARRAY:CURR?'

lucialiu2020, 2021-07-15:
We know query command means viwrite + viread. The depth of my instrument (Keysight 66319D) fifo is 4096，the minum interval is 15.6E-6. 

I want to sample a waveform that contiues about 3 mins. So I try to use this query command ('MEAS:ARRAY:CURR?') to multiple sample.However, I found viRead(read the sampled data) costs too many times. When viRead spend many time to return sampled data, it will lose many waveform data before the next viRead start. This is unreasonable. 

My environment:
Python 3.9.5 (tags/v3.9.5:0a7dcbd, May  3 2021, 17:27:52)

Keysight 66319D
code
#!/usr/bin/env python
# -*- coding: utf-8 -*-
from list import InstrList    # for testing
from list import AvlInstrList
import pyvisa                 # for testing
from time import sleep        # for testing
import datetime               # for testing
import os as os               # for testing
import numpy as np
import matplotlib.pyplot as plt
from time import *

    rm = pyvisa.ResourceManager()
    rm.list_resources()
    test = rm.open_resource('GPIB1::0::3::INSTR')
    test.chunk_size = 10240
    test.timeout = 240000
    # success
    print(test.query('*IDN?'))  

    #test sample interval、 points, success
    test.write('SENS:SWE:TINT 15.6E-6')
    test.write('SENS:SWE:POIN 700')
    test.write('SENS:CURR:RANG MAX')
    test.write('SENS:FUNC ""CURR""')
    
    #use query command 
    test.write('MEAS:ARRAY:CURR?')
    
    ....   
    #save data in array and plot ..
    
I also open the keysight IO monitor to capture when run this py script.
This is the log:
[image]
In this log, viWrite costs 1.1769ms，but viRead costs 1893.15ms which is hard for me to understand.
sample period：15.6E-6 * 700 = 10.92ms, but return the sample data costs(1893.15-10.92)ms. 
My issue:
If I need sample a long period waveform so what should I do?
Many methods have been tried, but they can't solve the problem.

I've been bothered by this problem for a week. If you have any suggestions or ideas, please leave a comment. 
I would appreciate your comments!  Love & Peace!

MatthieuDartiailh, 2021-07-15:
First it appears your instrument is returning data formatted as ASCII which is a fairly inefficient. Configuring your instrument to transfer data in binary format ("FORM:DATA REAL,32") could help with. Second according to the manual 
https://literature.cdn.keysight.com/litweb/pdf/5964-8184.pdf using any MEAScommand will initiate a new measurement, for your use case you want to query the current points without starting everything anew (since you want to stream). As a consequence you should use "FETCH:ARRAY:CURR?". If you can describe your use case in more details, I may be able to provide further guidance.

lucialiu2020, 2021-07-16:
> First it appears your instrument is returning data formatted as ASCII which is a fairly inefficient. Configuring your instrument to transfer data in binary format ("FORM:DATA REAL,32") could help [...]

Hi，
I'm sorry for the misunderstanding caused by not giving enough details. Now the code details is updated.  I need to sample a three minute current， so try to use test.query('MEAS:ARRAY:CURR?') in a loop. 
I've tried the fetch command "FETCH:ARRAY:CURR?" in this loop, but it doesn't return new data. 
As for the returned data format  is assic or binary format, there is a difference in the time consumed by the two methods. I am trying to compare them and show you the results later.Thank you for your reply~~

lucialiu2020, 2021-07-19:
Hi,

Acoording to your advice, I did some tests to try to figure out some relationships.First of all, I still get these time values through keysight IO capture when run my py script.

**1.query_binary_values & query_ascii_values**
Tests show that using binary format is more efficient and time-saving than using assic format. But it still can't solve my core request. This method is still not efficient when multiple measures are needed and each measure will fill 4096 data buffers.
[image]
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:dt="uuid:C2F41010-65B3-11d1-A29F-00AA00C14882"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=OneNote.File>
<meta name=Generator content="Microsoft OneNote 15">
</head>

<body lang=zh-CN style='font-family:Calibri;font-size:11.0pt'>

<div style='direction:ltr'>

fuction | sample   interval | Sample   points | value   format | cost   time(ms)
-- | -- | -- | -- | --
MEAS:ARRAY:CURR? | 2ms | 500 | binary | 1571
MEAS:ARRAY:CURR? | 2ms | 500 | assic | 2434

</div>

</body>

</html>

**2.MEAS:ARRAY:CURR? & read_raw() & read_bytes()**
In this test, I use read_bytes(4*4096+8) cause "The data within the Arbitrary Block is coded as IEEE single precision floating point, with 4 bytes per value.". The purpose of test2 is to try to see if it will save time without the cost of format conversion.But the results show that the costs of three fuctions are similar.
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:dt="uuid:C2F41010-65B3-11d1-A29F-00AA00C14882"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=OneNote.File>
<meta name=Generator content="Microsoft OneNote 15">
</head>

<body lang=zh-CN style='font-family:Calibri;font-size:11.0pt'>

<div style='direction:ltr'>

fuction | interval | points | format | cost(ms)
-- | -- | -- | -- | --
Read_raw | 2ms | 4096 | binary | 12760.8
Read_bytes | 2ms | 4096 | binary | 12759
MEAS:ARRAY:CURR? | 2ms | 4096 | binary | 12691.6

</div>

</body>

</html>

3. **TRIG:ACQ:COUN:CURR <number> + MEAS:ARRAY:CURR?**
I learned this instruction(TRIG:ACQ:COUN:CURR <number>) in the instrument documentation you gave me. It seems that it will solve my needs very well, so I have also tried it. 
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:dt="uuid:C2F41010-65B3-11d1-A29F-00AA00C14882"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=OneNote.File>
<meta name=Generator content="Microsoft OneNote 15">
</head>

<body lang=zh-CN style='font-family:Calibri;font-size:11.0pt'>

<div style='direction:ltr'>

fuction | interval | points | number(trigger   times | format | cost(ms)
-- | -- | -- | -- | -- | --
MEAS:ARRAY:CURR? | 2ms | 100 | 5 | binary | 1585.13
MEAS:ARRAY:CURR? | 2ms | 100 | 20 | binary | 6249.74
MEAS:ARRAY:CURR? | 2ms | 1000 | 5 | binary | timeout   error

</div>

</body>

</html>
The result shows that when trigger times*sample points ≤ the length of buffer(4096), this methods works well. But when trigger times*sample points > the length of buffer(4096), timeout occurs(I think it's also logical).

**All in all,**

 I'm still trying to find a solution.If you have any suggestions or ideas, please let me know as soon as possible. Thank you very much~~

MatthieuDartiailh, 2021-07-19:
Except if you instrument has a dedicated mode in which it can stream data (i.e. write them directly on the bus), I do not think you will be able to get measurements over such long period of time without any gap due to the limit on the buffer size. This is not a problem with VISA but more with the capabilities of the hardware. I would suggest to try to reach Keysight directly since they will know more about the capabilities of your instrument than me that never used it.

lucialiu2020, 2021-07-20:
@MatthieuDartiailh  Thanks~ I've asked keysight for help as you suggested~