pyvisa/pyvisa #614: how to write data to serial port with pyvisa?

bactone, 2021-07-21:
Hi, I am trying to communicate with serial port using pyvisa, and the following is the demo code:
```python
import pyvisa
from pyvisa.resources import serial

rm = pyvisa.ResourceManager('@py')
print(rm)
devlist = rm.list_resources()
print(devlist)  # device list feedback is ('ASRL7::INSTR', 'ASRL8::INSTR', 'ASRL9::INSTR')
RS232 = serial.SerialInstrument('ASRL9::INSTR')
RS232.write("V")
print(RS232.read())
```
but when i run the code, it reports error:
```
RS232 = serial.SerialInstrument('ASRL9::INSTR')
TypeError: __init__() missing 1 required positional argument: 'resource_name'
```

so how to communicate with serial port?

MatthieuDartiailh, 2021-07-21:
You should not create resource (i.e. serial.SerialInstrument) directly but use the resource manager to open the resource. It will select the right class based on the address. So:
```python
import pyvisa

rm = pyvisa.ResourceManager('@py')
print(rm)
devlist = rm.list_resources()
print(devlist)  # device list feedback is ('ASRL7::INSTR', 'ASRL8::INSTR', 'ASRL9::INSTR')
RS232 = rm.open_resource('ASRL9::INSTR')
# You may need to configure further the resource by setting explicitly the
# baud rate, parity, termination character, etc before being able to communicate
# Those depend on your instrument.
RS232.write("V")
print(RS232.read())
```
You can also have a look at the documentation https://pyvisa.readthedocs.io/en/latest/

MatthieuDartiailh, 2021-09-01:
@bactone did you solve your issue ?