pyvisa/pyvisa #615: I got "pyvisa.errors.VisaIOError: VI_ERROR_INP_PROT_VIOL (-1073807305): Device reported an input protocol error during transfer."

doiuec, 2021-07-22:
Hi, I want to communicate with the instruments with pyvisa, but the error occurs when I use "read_raw" command.
I want to read the data of wave in the oscilloscope as the binary data, but it seems like the "raw_data" does not work well in my code.
I searched the error code, but I couldn't find the hint about the solution of that.

codes(main)
------------------
``` python
import struct
import serial
import time
import numpy as np
import MSO5074 as OSCILLO  #the code of MSO5074 is below.

results = []

oscillo = OSCILLO.MSO()

pts = np.load('file name')
cts = np.load('file name')
usb = serial.Serial('COM11',115200,timeout = 2)

count = 0
for i in pts:
    count += 1
    send = bytes(list(i))
    print('PT:',send.hex())
    usb.write(send)
    get = usb.read(16)
    print("CT:",get.hex())
    print(count,'/1000')
    time.sleep(0.5)
    oscillo.scope.write(":STOP")
    oscillo.scope.write(":WAVeform:SOURce CHAN1")
    oscillo.scope.write(":WAV:MODE NORMal")
    oscillo.scope.write(":WAVeform:FORMat BYTE")
    oscillo.scope.write(":WAVeform:POINts 1250")

    binary = oscillo.get_data()  #the problem is here.

    wave = binary[10:-1]
    l = len(wave)
    fm = '<' + str(l) + 'B'
    tapple = (struct.unpack(fm,wave))
    _list = list(tapple)
    results.append(_list)
    oscillo.scope.write(":RUN")
    time.sleep(0.2)
usb.close()

oscillo.close()

results = np.array(results,dtype=object)
filename = './data/WAVE/waves_perfect.npy'
np.save(filename,results)
```

codes(MSO5074)
------------------
``` python
import pyvisa
import numpy
import matplotlib.pyplot as plot
import sys
import time

class MSO:
    def __init__(self): 
        rm = pyvisa.ResourceManager() 
        inst = rm.list_resources() 
        print(inst)
        usb = list(filter(lambda x: "USB" in x, inst)) 
        if len(usb) != 1:
            print('Bad instrument list', inst)
            sys.exit(-1)
        self.scope = rm.open_resource(usb[0], timeout=10, chunk_size=10 * 1024 * 1024) 

    def get_data(self):
        self.scope.write(":WAVeform:DATA?")
        time.sleep(2)
        tmp = self.scope.read_raw() #the problem is here
        return tmp

    def close(self):
        self.scope.close()
```

error code
------------------
```
c:/Users/usr/main.py
('USB0::0x1AB1::0x0515::MS5A221401414::INSTR', 'ASRL1::INSTR', 'ASRL11::INSTR')
PT: 82c4ce86ae48c89e4d0d04f47a21f697
CT: 841e9c8dd543a192226333de91449823
1 /1000
USBInstrument at USB0::0x1AB1::0x0515::MS5A221401414::0::INSTR

Traceback (most recent call last):
  File "c:\Users\usr\DPA\for_DOI\getPulse.py", line 42, in <module>
    binary = oscillo.get_data()
  File "c:\Users\usr\DPA\for_DOI\MSO5074.py", line 62, in get_data
    tmp = self.scope.read_raw()
  File "C:\Users\usr\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 405, in read_raw
    return bytes(self._read_raw(size))
  File "C:\Users\usr\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\usr\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\usr\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\usr\AppData\Local\Programs\Python\Python39\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_INP_PROT_VIOL (-1073807305): Device reported an input protocol error during transfer.
```

Instrument details
------------------
* Model:RIGOL MSO5074
* Communication: GPIB(USB)
* Link to the documentation (if available):https://cdn-reichelt.de/documents/datenblatt/D100/RIGOL_MSO5000_PG-EN.pdf

Output of `pyvisa-info`

```
 C:\Users\usr>pyvisa-info

Machine Details:
   Platform ID:    Windows-10-10.0.19041-SP0
   Processor:      AMD64 Family 23 Model 8 Stepping 2, AuthenticAMD

Python:
   Implementation: CPython
   Executable:     c:\users\usr\appdata\local\programs\python\python39\python.exe
   Version:        3.9.5
   Compiler:       MSC v.1928 64 bit (AMD64)
   Bits:           64bit
   Build:          May  3 2021 17:27:52 (#tags/v3.9.5:0a7dcbd)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
   py:
      Version: 0.5.2
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```
-----------------------

MatthieuDartiailh, 2021-07-22:
I have never seen that one I must say. Can you read other response from the instrument ? the response to *IDN? for example. Also what happens if you try to read a fixed number of bytes (1 for example) using `read_bytes` ?

doiuec, 2021-07-24:
Thank you for your reply!
I have solved this problem by uninstalling the application which is conflicted to pyvisa.

However, I get the another problem...
I want to get the wave data as binary data from the oscilloscope(RIGOL MSO5074).
But I cannot get the full length of data, that is, only the partial data is appear...

This is one of the data which I want to get as binary.
[image]

And this is a graph created from the binary data obtained.[image]

Perhaps this graph only shows  the first part of the waveform.
I have changed the number of points by `":WAVeform:POINts 10000"`, but over 4000000 will time out.(The memory depth is 4000000.)
I checked that 2000000 points could work, and the half of waveform was converted to graph.

main code
---------------
``` python
import struct
import serial
import time
import numpy as np
import sample as OSCILLO

results = []

oscillo = OSCILLO.MSO()

pts = np.load('file name')
cts = np.load('file name')
usb = serial.Serial('COM11',115200,timeout = 2)

count = 0
for i in pts:
    count += 1
    send = bytes(list(i))
    print('PT:',send.hex())
    usb.write(send)
    get = usb.read(16)
    print("CT:",get.hex())
    print(count,'/1000')
    time.sleep(0.5)
    oscillo.scope.write(":STOP")

    oscillo.scope.write(":WAVeform:SOURce CHAN2")
    oscillo.scope.write(":WAV:MODE RAW")
    oscillo.scope.write(":WAVeform:FORMat BYTE")
    oscillo.scope.write(":WAVeform:POINts 10000")
    
    binary = oscillo.get_data()
    wave = binary[10:-1]
    l = len(wave)
    fm = '<' + str(l) + 'B'
    tapple = (struct.unpack(fm,wave))
    _list = list(tapple)
    results.append(_list)

    resultss = np.array(results)
    filename = './data/WAVE/waves_perfect.npy'
    np.save(filename,resultss)

    oscillo.scope.write(":RUN")
    time.sleep(0.2)
usb.close()

oscillo.close()

results = np.array(results)
filename = './data/WAVE/waves_perfect.npy'
np.save(filename,results)
```

oscilloscope.py
----------------
```python
import pyvisa
import numpy
import matplotlib.pyplot as plot
import sys
import time

class MSO:
    def __init__(self):
        rm = pyvisa.ResourceManager() 
        inst = rm.list_resources() 
        usb = list(filter(lambda x: "USB" in x, inst)) 
        if len(usb) != 1: 
            print('Bad instrument list', inst)
            sys.exit(-1)
        self.scope = rm.open_resource(usb[0], timeout=30000, chunk_size=10 * 1024 * 1024) 

    def get_data(self):
        
        self.scope.write(":WAV:DATA?") 
        time.sleep(2)
        rawdata = self.scope.read_raw() 
        rawdata = rawdata[ 10 : ] 
        data_size = len(rawdata)
   
        sample_rate = self.scope.query(':ACQ:SRAT?')[0]
        print('Data size:', data_size, "Sample rate:", sample_rate)
        return rawdata
        
    def close(self):
        self.scope.close()
```

graph code
----------------
``` python
import numpy as np
import matplotlib.pyplot as plt
import matplotlib as mpl

data = np.load('file name')

size = 9991

for i in range(size):
    data1[i] = data[0,i]

labels = np.zeros(size)

for i in range(size):
    labels[i] = i

plt.plot(labels,data1)
plt.show()
```

MatthieuDartiailh, 2021-07-25:
First note that `read_raw` will stop if it encounters the termination characters (which is likely present in tour binary data). So I would suggest you to give a look at using `read_binary_values` that can perform the conversion for you in nice way (see https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html#reading-binary-values). Also note that typically oscilloscope do not return values in volt but as integers corresponding to the internal resolution of their analog digital converters. You need to convert to volt using the channel offset and volt per division ratio.

closed 2022-09-27 by side-realms

kglaraiza, 2023-11-06:
What would be the application conflicted to pyvisa that you uninstalled?