pyvisa/pyvisa #616: [TCPIP] Communication issue with FLUKE 8845A : can set but not read

lplPerard, 2021-07-28:
Hello,

I am currently trying to control a Fluke 8845A through LAN using pyVISA.
I can correctly send commands such as : "CONF:VOLT:DC" and it works perfectly well !

But ...

I can't find a way to read from the instrument, I have a timeout error.

pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

I tried different commands such as : 
 
                    ressource.write('MEAS:VOLT:DC?')         
                    voltage = float(ressource.read())
 
Does anyone have the same issue ?
A solution ?
 
Thank you ! :)

Instrument details
------------------
* Model: FLUKE 8845A
* Communication: TCPIP

lplPerard, 2021-07-28:
Update : Using 

while toto :
     voltage = self.instrument.ressource.read_bytes(2)

I have the following output : 

start
b'+9'
b'.3'
b'12'
b'40'
b'00'
b'0E'
b'-0'
b'4\r'
b'\n+'
b'8.'
b'98'
b'08'
b'00'
b'00'
b'E-'
b'04'
b'\r\n'
b'+8'
b'.9'
b'59'
b'40'
b'00'
closed

MatthieuDartiailh, 2021-07-28:
Setting the `read_termination` character to `\r\n` should do the trick: `ressource.read_termination = "\r\n"`

lplPerard, 2021-07-28:
That was the fastest best answer I had :)
It works perfectly fine !

closed 2021-07-28 by lplPerard

MatthieuDartiailh, 2021-07-28:
Happy to help