pyvisa/pyvisa #617: [COM] wait_on_event()  time out

michaelshow, 2021-08-03:
I have two NI GPIB converters connected to each other: A (GPIB-ENET) and B (GPIB-USB). A simulates device access in a non-control role, and B serves as a control access. A and B can communicate normally, read and write each other。
I want to test wait_on_event(), that is, B waits for A to issue an SRQ event. I simulated sending SRQ by use `set_visa_attribute(pyvisa.constants.ResourceAttribute.device_status_byte, 108)`
 on the A side. SRQ events can be successfully monitored through NI tool, but pyvisa's wait_on_event() returns a timeout: `pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.`。
Shown on NI IO-trace: Both pyvisa and NI tool finally call the viWaitOnEvent() method of visalib

### **code for B**
```
from pyvisa import ResourceManager, constants
import pyvisa
# pyvisa.log_to_screen()

rm = ResourceManager()
# print(rm.list_resources())
with rm.open_resource("GPIB1::5::INSTR") as instr:
    instr.clear()
    print('start wait...')
    event_type = constants.EventType.service_request
    event_mech = constants.EventMechanism.queue
    instr.enable_event(event_type, event_mech)
    while True:
        try:
            res = instr.wait_on_event(event_type, 3000)
            print('event SQR.....')
            print('stb1---{}'.format(instr.stb))
            print('res---{}--{}---{}'.format(res.ret.name, res._event_type.name, res._context))
            # res = instr.wait_for_srq(3000)
            # print('stb2---{}'.format(instr.stb))
            # break
        except Exception as e:
            print('error---{}'.format(e))
            print('stb2---{}'.format(instr.stb))
```

MatthieuDartiailh, 2021-08-03:
I have no real idea what is happening here. Could you also post your code for A and a screen shot of NI IO-trace ?

michaelshow, 2021-08-03:
### **code for A**
```
# in python shell
from pyvisa import ResourceManager, constants

rm = ResourceManager()
bd = rm.open_resource("GPIB0::INTFC")
bd.is_system_controller = False
bd.set_visa_attribute(constants.ResourceAttribute.device_status_byte, 108)
```
### **screenshot** 
**NI-trace**
[image]

michaelshow, 2021-08-03:
[image]

MatthieuDartiailh, 2021-08-03:
From your NI IO screen shot it looks like NI agrees and your request does indeed timeout (according to the error code)., or am I missing something ?

MatthieuDartiailh, 2021-08-03:
> Uploading NI-trace.png…

This simply rounds trip to this page, I assume it is not intentional.

michaelshow, 2021-08-03:
[image]

MatthieuDartiailh, 2021-08-03:
Setting the bit on the INTFC, I would expect a single service request to be generated so if you have 2 programs waiting for one I am not sure that both will get it. What happens if you close NI tools ?

michaelshow, 2021-08-03:
I set the bit on INTFC many times, but only the NI tool gets it, even if the NI tool is closed, pyvisa always timeout.
In another case, open two NI tools at the same time(pyvisa is not call wait_on_event()), they can get it

michaelshow, 2021-08-03:
The following screen shot  shows the NI tool calling event
[image]

MatthieuDartiailh, 2021-08-03:
Can you confirm that you see in NI IO the viEnableEvent performed by PyVISA ? In your last screenshot it is not visible (I see only the one for NI) so I want to be sure it is there.

michaelshow, 2021-08-03:
I can confirm that I see the viEnableEvent executed by PyVISA in NI IO, and the execution result is "0 (VI_SUCCESS)".
If you execute wait_on_event when enable_event is not executed, an error will be reported: "pyvisa.errors.VisaIOError: VI_ERROR_NENABLED (-1073807313): You must be enabled for events of the specified type in order to receive them."

MatthieuDartiailh, 2021-08-03:
I am a bit at a loss I must say. 

One minor difference you have between NI and PyVISA is the address that PyVISA attempts to normalize but I could never find an authoritative source for the secondary address and so currently GPIB::15::INSTR is considered equivalent to GPIB::15::0::INSTR. You could try to edit pyvisa/resources/resource.py line 144-148 to skip the normalization.

michaelshow, 2021-08-04:
Thank you. Your guess is correct. skip the default value (0) of the secondary address of the resource_name and skip the normalization. pyvisa can get event. 
The code I modified is pyvisa/resources/resource.py line 163-167，The __init__ method of the Resource class:
```
try:
    # Attempt to normalize the resource name. Can fail for aliases
    self._resource_name = str(rname.ResourceName.from_string(resource_name))
except rname.InvalidResourceName:
    self._resource_name = resource_name
self._resource_name = resource_name  # my new line
```

MatthieuDartiailh, 2021-08-04:
Great ! Thanks for testing !

So going again through a bunch of GPIB, I found a couple hinting at the fact that a secondary address of 0 and no secondary address are not equivalent, along with your situation, it makes a strong argument for changing how we normalize GPIB addresses and what default value to use. It may take me some time before I can look into it so if you want to make a PR to address this feel free to beat me to it.

A reasonable plan would be to make the default be None instead of "0" and make sure that the code handling conversion back and forth will ignore the None which may be a bit more tricky since this code handles all resource names. The relevant file is rname.py