pyvisa/pyvisa #618: [COM] Communication to 34980A is very slow with only 2 write and query commands

techgirl-1, 2021-08-05:
I am trying to communicate with GPIB using pyvisa, and the following is the demo code:

import pyvisa

rm = pyvisa.ResourceManager('filepath')
devlist = rm.list_resources()
print(devlist)  # this prints quickly 

switchStation= rm.open_resource('GPIB0::21::INSTR')
switchStation.read_termination = '\n'
switchStation.write_termination = '\n'
print(switchStation.query('*IDN?')) # This operation also happens quickly 

## this is where the code really starts to lag, it takes about 45 seconds for the equipment to respond 
switchStation.write('ROUT:OPEN (@1001)') 
print(switchStation.query('ROUT:OPEN? (@1001); *OPC?'))

## And then takes another 1 minute to respond to this 
switchStation.write('ROUT:CLOS (@1001)')
print(switchStation.query('ROUT:CLOS? (@1001); *OPC?'))

The operation completes and everything prints correctly, however its seems to take a little more than 2 minutes for just these operations to run. Is this normal? I'm not sure why it is communicating to slowly, could it be the way I installed PyVISA? I feel like these are rather simple operations for python to handle and execute in a timely manner.

MatthieuDartiailh, 2021-08-05:
It does seem quite slow for the switch to operate. Do you observe a similar behavior querying multiple times the IDN ?

techgirl-1, 2021-08-05:
Yes, when I query IDN? for multiple instruments in a row it is similarly slow

MatthieuDartiailh, 2021-08-05:
What other instrument do you have on the GPIB bus ? Are they all affected similarly ? Since GPIB is dispatched to all connected instrument, a single bad sheep can wreak havoc across the whole bus.

techgirl-1, 2021-08-05:
I have a power supply and a picoammeter connected to the GPIB bus. I am using Spyder to run the program, could it be any file paths? Also, what is the typical response time for a query/write in PyVISA?

techgirl-1, 2021-08-05:
Should PyVISA be a sub directory in my Python folder or should it be its own folder on the same directory level as Python ?

MatthieuDartiailh, 2021-08-05:
Except if you plan to hack into pyvisa I would recommend to simply install it using pip. The response time is rather instrument dependent but can be in the 1-10ms range for simple query (IDN should be one). Do you observe the same latency issue using your other instruments ? 

Can you post the output of `pyvisa-info` (run from the cmd line once you install PyVISA using pip ? That way I will know on what system you are working.

techgirl-1, 2021-08-05:
When writing to other instruments, the latency is less. It's roughly 20 seconds total for execution and response, which is still long but not as long as the switch station.  The computer is offline so I cannot use pip. If you know what directory PyVISA should be in or if there is a filepath I need to add to my Python folder, I think that could help. I know my code is able to find the file path because my program executes, but I am curious to know if it takes awhile because python or pyvisa is having a hard time looking for a particular library.

techgirl-1, 2021-08-05:
Also, my code did not use the read_termiantion and write_termination characters when set them in the beginning. Which was why I had to add *OPC? in my query commands. Do you suspect it could be a EOI problem?

MatthieuDartiailh, 2021-08-09:
If you have a cloned version of pyvisa on the computer you can simply run `python setup.py install` from the root of the PyVISA repo.  

I fail to see the relation between termination characters and the use of OPC. Do you get something notably different without OPC ?

techgirl-1, 2021-08-09:
Hi I just solved the issue by unplugging and re-plugging in the USB cable. I should've known, that's IT 101.

MatthieuDartiailh, 2021-08-09:
Happy you solved it !

closed 2021-08-09 by MatthieuDartiailh