pyvisa/pyvisa #619: [COM] Communication issue with Keithely 2000 DMM using GPIB @py

sjlukacs, 2021-08-06:
hello one and all,

i'm trying to reproduce "A more complex example" with an actual Keithley 2000 using GPIB0::20::INST as in https://pyvisa.readthedocs.io/en/latest/introduction/example.html under python3 and pyvisa 1.11.

thank you in advance.  lucas.. .

if i copy the exact 18 lines of code, exactly, i get the exception on the implementation of the keithley.wait_for_srq() line:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/resources/gpib.py", line 168, in wait_for_srq
    constants.EventType.service_request, constants.EventMechanism.queue
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/resources/resource.py", line 530, in enable_event
    self.visalib.enable_event(self.session, event_type, mechanism, context)
  File "/usr/local/lib/python3.7/dist-packages/pyvisa/highlevel.py", line 909, in enable_event
    raise NotImplementedError
NotImplementedError

where i will see on the SRQ status label on the 2000 display.

here is my pyvisa-info output:
Machine Details:
   Platform ID:    Linux-5.10.17-v7l+-armv7l-with-debian-10.9
   Processor:      

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.7.3
   Compiler:       GCC 8.3.0
   Bits:           32bit
   Build:          Jan 22 2021 20:04:44 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.2
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.4')
      GPIB INTFC: Available via Linux GPIB (b'4.3.4')

LongnoseRob, 2021-08-07:
pyvisa with the pyvisa-py backend does not support wait_for_srq() yet.
See also https://github.com/pyvisa/pyvisa-py/issues/142 & https://github.com/pyvisa/pyvisa-py/pull/175

sjlukacs, 2021-08-08:
so, i read the posts.  i believe this feature is not part of pyvisa-py but tivek is working on it but its been stalled because he doesn't have access to his windows based ni-visa to learn and compare on how to implement the coding within pyvisa-py.  if i have that correct?

there was some mention of application level event handling because tivek seems unsure how to implement it in python on the pyvisa-py class level, if i understood that correctly.  how would we implement the srq event handling within our applications?

LongnoseRob, 2021-08-08:
@sjlukacs yes the work seems stalled for the moment.
I also had offered help with testing with my 82357B USB-GPIB interface and my instruments, however I am also waiting on feedback from tivek.

I know that SRQ is a actually a dedicated signal on the GPIB, I have no idea if this is also implemented in a VISA session via TCPIP.
Maybe someone with more details on that could explain this to us.

sjlukacs, 2021-08-08:
hey longnoserob,

I've been thinking about this.  and know that I haven't looked at any code whatsoever in pyvisa-py.

the wait_for_srq() function simply holds the calling application on that line until the srq register is set on the GPIB bus, correct?  if that is the case, can't we just write some hack-ish code under wait_for_srq() in pyvisa-py that loops/sleeps and queries the srq status register on the GPIB bus but on the host computer side until it is set.  once the srq register is set, the wair_for_srq() function breaks out of its loop and allows the calling application to continue on its way.

in this way, we're not polling the instrument directly, but only the srq status line/register from within the GPIB driver software and once set, move along.  its not perfect because it isn't based on the perfect timing of an event handler, but it would only probably require a few lines of code under wait_for_srq() function.  what do you think?  Lucas

MatthieuDartiailh, 2021-08-09:
Service request are supported on USB-TMC and TCPIP using the VXI11 (and I assume hislip too). Ideally I would prefer a solid base for handling event (which can be queue or handler base), but if you can come up with a reasonably clean first implemenattion for GPIB. I would definitively consider it since @tivek work has stalled for a bit now.

sjlukacs, 2021-12-15:
ok, sorry for the huge delay.  I had to build a machine and order a NI-GPIB-PCIe card for my Mac.  I did this so I can install the native NI drivers for and use NI-I/O-Trace to follow every send and receive from the instrument, still the Keithley 2000.  as a reminder, I'm literally reproducing example https://pyvisa.readthedocs.io/en/latest/introduction/example.html all to figure out why @py does NOT wait on wait_for_srq().  so I follow the example for the PCIe/NI-Drivers/Mac setup/configuration with the effective lines under python being:

	dmm.write(":initiate")
	print("dmm.assert_trigger()...")
	dmm.assert_trigger()
	ti = datetime.now()
	print("dmm.wait_for_srq(timeout)...")
	dmm.wait_for_srq(timeout)

where NI-Trace produces:

36.  viAssertTrigger (GPIB0::20::0::INSTR (0x00000001), 0 (0x0)) | Status: 0 (VI_SUCCESS)
37.  viEnableEvent (GPIB0::20::0::INSTR (0x00000001), 0x3FFF200B (VI_EVENT_SERVICE_REQ), 1 (0x1), 0 (0x0)) | Status: 0 (VI_SUCCESS)
38.  viWaitOnEvent (GPIB0::20::0::INSTR (0x00000001), 0x3FFF200B (VI_EVENT_SERVICE_REQ), 24999 (0x61A7), 0x3FFF200B (VI_EVENT_SERVICE_REQ), 0x00013001) | Status: 0 (VI_SUCCESS)
39.  viClose ("GPIB0::20::0::INSTR Event" (0x00013001)) | Status: 0 (VI_SUCCESS)

where line 36 is from assert_trigger() and lines 37 through 39 are from wait_for_srq().  i ask the dmm to get 25 measurements which takes about 10s with a timeout of 20s.  the computer clearly waits nicely on line 38 until the 25 measurements are complete.  when I do hard polling with dmm.stb, in loop waiting on return 65, instead of wait_on_srq(), lines 37-39 do not show up, which makes sense and is proper.  so the NI-Drivers are doing lines 37 - 39 within the DLL drivers when wait_to_srq() is executed.  this is not happening under the @py python drivers.

so, I figured that I should hard code lower-level lines to reproduce what NI-DLL-Drivers are doing to replicate wait_on_srq().  I was able to get line 37 to work via, dmm.enable_event(pyvisa.constants.VI_EVENT_SERVICE_REQ, 0x1), with a correct 0 return status.  but I can not reproduce lines 38 where the computer just waits until the SRQ is asserted.  I did try, dmm.enable_event(pyvisa.constants.VI_EVENT_SERVICE_REQ, 0x61A7) as per line 38, but I get a status return of 0xBFFF0027.  and if I try instead, dmm.wait_on_event(pyvisa.constants.VI_EVENT_SERVICE_REQ, timeout), I do not get a status error but the program just does not wait and the NI-Trace does not show viWaitOnEvent.

so, I need some advice and what direction to proceed.  thank you in advance and have a good day.  Lucas