pyvisa/pyvisa #620: Pyvisa enabling  Over current and over voltage protection on a DC  electronic Load  (Chroma 6314A)

LydieNsangou, 2021-08-06:
hi everyone,
please can someone help me out?. i am trying to enable the over current and over voltage protection on my electronic load automatically with pyvisa using python as programming language. i have checked already in the Manual but could not rally get how to enable the over current and over voltage protection. 
if any one has already had to do with this, i will appreciate your help

kind regards

LongnoseRob, 2021-08-07:
If I read the manual pg. 39 (the only pulblic available, I could find was https://xdevs.com/doc/Chroma/63103a/CHROMA631xxSeries_um.pdf ) correctly, the over-current/-voltage protections set depeding on the output mouldes parameters.

According to my understanding these protections are **always active** to protect the modules.

The limits for the different protections are listed as part of the spectification table from page 15 of above linked PDF.

LydieNsangou, 2021-08-09:
@LongnoseRob  thanks very much for your response . you are actually right the over-current /- voltage Protection are active on the output modules. but your could also do an OCP/OPP Test to check if it properly works

MatthieuDartiailh, 2021-08-09:
Closing since it appears the issue is resolved

closed 2021-08-09 by MatthieuDartiailh