pyvisa/pyvisa #624: [COM] Communication issue with Keithley 2000 using GPIB using SRQ / STB

sjlukacs, 2021-08-29:
hello one and all,

this is related to and grown out of issue: https://github.com/pyvisa/pyvisa/issues/619

as a first step in learning this trigger and SRQ bit, i have replaced the wait_for_srq() function, which currently isn't working or coded for under pyvisa-py, with the following code:
```
	ti = datetime.now()
	while 1:
		stb = dmm.query('*stb?')
		#stb = dmm.read_stb() #also tested with dmm.stb
		print('*sre?', dmm.query('*sre?'), '*stb?', stb, bin(int(stb)))
		if (stb == '4'):
			print('error:', dmm.query(':system:error?'))
		sleep(interval_in_ms/(2*1000.))
		if (stb == '65'):
			#0b1000001=bin(65)=6:1|5:0|4:0|3:0|2:0|1:0|0:1
			print('service request SRQ:', stb, ':::', bin(int(stb)), ':::', '|'.join([f'{i}:{b}' for i, b in enumerate(bin(int(stb))[len(bin(int(stb))):1:-1])][::-1]))
			break
		elif ((datetime.now() - ti).seconds > (10 + number_of_readings*(interval_in_ms/1000.))):
			print('loop timeout:::')
			break

```
where dmm is the pointer to the visa resource, or "dmm = visa.open_resource('GPIB0::20::INST')", and the "stb = dmm.query('*stb?')" polls the Keithley 2000 instrument for when the SRQ bit 6 is set, which signals to the host computer that the meter is done with its data collection bit.  this code works, but i'm moving to be more efficient in process and code by getting away from constantly polling the instrument.

as the next development step, i tried accessing the stb byte by using "stb = dmm.read_stb()" or "stb = dmm.stb" which i had hoped it was polling only the gpib bus instead the remote instrument.  i was hoping that the instrument, when it reached SRQ that it would set that bit on the bus also, simultaneously, thus only having to poll the bus for the bit and foregoing having to poll the instrument itself, thus reducing comm traffic and processing on the remote instrument.  doesn't the SRQ do that?  doesn't the instrument push the SRQ on the bus as well as the instrument at the same time?

ok, nonetheless, read_stb() nor stb did not work in either case, as commented above.  the above loop simply continued until it reached the timeout condition.  so, what is wrong with the read_stb() or stb reading of that integer?  it stays 0b0 the entire time.

thank you in advance and have a great week, lucas

sjlukacs, 2021-09-01:
did I ask a dumb question?

ok, the reason I'm wanting this to work well is because I'm trying to control my array of lab instruments with a $80 raspberry pi, and since it is arm processors, NI's VISA backend can't be installed.  so pyvisa-py is the only backend for a raspberry.

I'd like to contribute to pyvisa-py and I've programmed instruments on various levels, including assembler, since the 80s including labview.  anyway, I'm no expert on the intricacies of GPIB but many many instruments still use the bus and in some ways its still superior to tcp/ip.  not looking for a debate there.  anyway, I've also setup my windows 7 computer with a NI PCI-GPIB card with python 3.6 and the NI VISA backend so that I can run parallel tests and runs and learn this stuff and translate what I can learn into pyvisa-py.  basically, I'm serious about this but I'll also need guidance and such from you all.  what do you think?

btw, running the code above, the read_stb() or stb works fine on NI backend while not polling the instrument but only polling the bus, I believe.  but it doesn't work at all on pyvisa-py.

MatthieuDartiailh, 2021-09-01:
Sorry for the lack of answer. Your question is not dumb. Since @tivek knows definitely more about GPIB than me I had hoped he would participate. 

Regarding the problem at hand, the first thing I would like to know is what GPIB adapter you are using since I assume you do not have a PCI board on your Pi. Also can you test using this adapter on your windows computer to ensure we are not dealing with some funky adapter issue. 

If this test comes back ok, you can then look at the following:
- follow the discussion in https://github.com/pyvisa/pyvisa/issues/617 and see if preventing PyVISA to insert a secondary address of 0 fix your issue (it may very well do given the issue). If it does, I would really appreciate if you could make a PR to address it.
- if it does not help, try getting a linux computer with a PCI board and test using your PCI card to cross check that linux-gpib does its job properly

sjlukacs, 2021-09-01:
hey Matthieu,

ok, yes, on the Pi i'm using the NI GPIB-USB-HS which is challenging to configure the Pi for GPIB but once its done it works great.  the directions are at: https://iquanta.org/iquanta/main/c_blogitem/4.

i did install the NI PCI-GPIB card into my windows computer and read_stb does read the state off of the bus, instead of the instrument itself.  i've verified this that when the Keithley DMM is triggered collecting, it is NOT flashing talk/listen during collection.  it does flash talk/listen when i use "dmm.query('*stb?')" which makes sense because i'm polling the instrument directly and not the bus.

i then attached my NI GPIB-USB-HS to the windows computer and verified both behaviors.  and the USB adaptor works exactly the same at the PCI card.  do it is not an adaptor difference or problem.

i will not get into the wait_on_event() timeout issue you mentioned.  lucas

MatthieuDartiailh, 2021-09-01:
When you run tests on Windows, you run them through pyvisa or through NI-MAX ?

MatthieuDartiailh, 2021-09-01:
Also I cannot access the blog you linked.

sjlukacs, 2021-09-01:
my tests are run in python 3.6, importing pyvisa, and open_resource and such.  so the only touch to NI-MAX or NI-VISA should only be when pyvisa accesses it on the backend.  correct?

oh yeah, are you in France or outside the USA?  years ago, I put an IP fence around my web server because Russia, china, and North Korea were trying to hack my little server with thousands of challenges.  the IP fence did the trick.  maybe I can upload a PDF to this site of that blog page.

MatthieuDartiailh, 2021-09-01:
I am indeed currently in France. Good to know that both are tested through pyvisa even though it may make #617 less relevant (but maybe NI is doing magic that linux GPIB is not). Keep me posted.

sjlukacs, 2021-09-22:
hey all,

I'm trying to understand the interface between PyVisa and the backend.  right now, I'm using the "NI-488.2 19.5.dmg" driver on a MacPro 10.14 Mojave with a NI PCIe-GPIB interface card.  took a while to find and get that card but I wanted to go old-school on the interface for most of my instruments use GPIB.  I have my test code working and well with the wait_on_srq() function is working properly with the NI backend as opposed to the pyvisa_py backend on the raspberry, which started me on this whole journey.

getting into the nuts and bolts and I'm browsing the source code of pyvisa, first, where or how does pyvisa interface with the backend?  also, does wait_on_srq() under pyvisa simply do serial polling or does it wait for an event trigger?  I think I saw code for the latter in resources/gpib.py.  is the backend firing the event or simply setting the srq state somewhere?

I guess I have no idea right now and I need some direction.  Lucas

MatthieuDartiailh, 2021-09-23:
In PyVISA, a backend must provide an implementation of the `pyvisa.highlevel.VisaLibraryBase` class. When you use the `ivi` backend in combination with a proprietary implementation of VISA (by NI, Keysight, R&S, Tektronik, etc), pyvisa use the implementation of `pyvisa.highlevel.VisaLibraryBase` found in `pyvisa.ctwrapper.highlevel` which uses `ctypes` to interface with the vendor implementation. Typically when waiting on an srq, the `viWaitOnEvent` function will be called. The implementation of this function is up to the vendor.

Hope this helps, and let me know if you have further questions.