pyvisa/pyvisa #625: Issue with saving screen image from scope -

bardgu, 2021-08-31:
while trying to save the image - I ran into an error  and the try not succsses.
    def save_screen_image(self, file_path):
        file_format = file_path.split('.')[-1]

        format_list = ['bmp', 'gif', 'tif', 'png', 'jpeg']
        if file_format not in format_list:
            root = Tk()
            root.withdraw()     # use to hide tkinter window

            root.overrideredirect(True)
            root.geometry('0x0+0+0')
            root.deiconify()
            root.lift()
            root.focus_force()

            messagebox.showerror(parent=root, title='Error',
                                 message=f'Not supported image format')
            root.destroy()
        file_path = File_handel.check_file_exist(file_path)
        if file_path == '':
            return
        self._rm.write(':STOP')
        self._rm.query('*OPC?')      # Wait for instrument to finish
        try:
            read_data = self._rm.query_binary_values(f':DISPlay:DATA? {file_format}, SCReen',
                                                     datatype='B',
                                                     is_big_endian=False,
                                                     container=bytearray)
            File_handel.save_binary_file(file_path, read_data)
        except:
            root = Tk()
            root.withdraw()     # use to hide tkinter window

            root.overrideredirect(True)
            root.geometry('0x0+0+0')
            root.deiconify()
            root.lift()
            root.focus_force()

            messagebox.showerror(parent=root, title='Error',
                                 message=f'Error retrieving scope data.')
            root.destroy()

Im allways getting into the except  any idea why ?
thanks alot

MatthieuDartiailh, 2021-08-31:
Can you post the traceback you get if you add a raise statement at the end of your except block so we know what error you encounter ?

bardgu, 2021-08-31:
Thanks for fast response 

Traceback (most recent call last):
  File "X:\General\Characterization\Python\CHAZ_Environment\Python39\lib\contextlib.py", line 135, in __exit__
    self.gen.throw(type, value, traceback)
  File "X:\General\Characterization\Python\CHAZ_Environment\PyCHAZ\lib\site-packages\pyvisa\highlevel.py", line 298, in ignore_warning
    yield
  File "X:\General\Characterization\Python\CHAZ_Environment\PyCHAZ\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "X:\General\Characterization\Python\CHAZ_Environment\PyCHAZ\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "X:\General\Characterization\Python\CHAZ_Environment\PyCHAZ\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "X:\General\Characterization\Python\CHAZ_Environment\PyCHAZ\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
python-BaseException

MatthieuDartiailh, 2021-08-31:
Ok so it is a timeout. First can you confirm that you checked you could communicate without issue with your instrument (querying *IDN? for example) ? 
If yes, you can have a look at https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html#reading-binary-values and https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html#when-things-are-not-what-they-should-be. In particular, can you read any response from the instrument one byte at a time (to ensure the instrument did answer) ? If the instrument does answer, check if the termination character is included at the end of the block and adapt your query. Finally you may need to increase the default timeout since your may just be transferring a lot of data.

bardgu, 2021-08-31:
Thanks a lot the delay solve the issue.

MatthieuDartiailh, 2021-08-31:
Happy to help

closed 2021-08-31 by MatthieuDartiailh