pyvisa/pyvisa #626: MacBook issue

paolovezio, 2021-09-08:
hello, I have an issue to read the usb address of the oscilloscope Siglent SDS2354X Plus. The MacBook gives other devices address but not the oscilloscope one. I use the code
import pyvisa
rm = pyvisa.ResourceManager('@py')
print(rm.list_resources())
I think I have a connectivity/communication problem because also the NI MAX software crashing all time I try to use it.

some one can help me.
thanks

MatthieuDartiailh, 2021-09-09:
The fact that NI-Max crashes, is not a good sign indeed. Can you post the output of `pyvisa-info` ? Also looking at the manual of your instrument it appears that it also support communication over TCPIP. Could you give that a shot ? It is generally more portable but note that you will need to know the address since pyvisa won't list it.

dominikfigueroa, 2021-12-23:
@MatthieuDartiailh 
> I think I have a connectivity/communication problem because also the NI MAX software crashing all time I try to use it.

Not sure if it's related but, I was also having problems using NI-Max on macOS. I was not able to open any NI-VISA related application without it crashing while starting up. Just today I found this solution on stackoverflow which solved my problems.

https://stackoverflow.com/a/68447825/4302911

LongnoseRob, 2021-12-23:
@dominikfigueroa 
I would recommend to add this to the FAQ part of the documentation, somewhere around [here](https://github.com/pyvisa/pyvisa/blob/main/docs/source/faq/faq.rst#where-can-i-get-more-information-about-visa).

Would you be able to ppen a PR for the documentation enhancement regarding the use use of NI-applciations on MacOs?