pyvisa/pyvisa #627: Trigger setting for an oscilloscope with pyvisa

LydieNsangou, 2021-09-15:
Hi everyone,
please i have an issue. i am trying to do a trigger automatically using pyvisa but the problem is hen i trigger manually it works but when i do the same process automatically using pyvisa it does not work anymore. 
here is my code please:

#setting the trigger parameters!
  !1)  oszi.write('HORIZONTAL:SCALE 10E-3')# setting the horizontal scale to 20ms
    oszi.write('CH3:SCALE 5')#setting the y-axis to 5v per division
    oszi.write('TRIGGER:A:TYPE EDGE') #specifies that the trigger is an Edge triggering

    oszi.write('TRIGGER:A:EDGE:COUPLING DC') #setting the edge trigger coupling to DC
    sleep(0.5)
    oszi.write("TRIGGER:A:EDGE:SLOPE RISE") #setting trigger as rising edge

    oszi.write('TRIGGER:A:MODE:NORMAL') #sets the A trigger mode(specifies that a valid trigger event must occur before a trigger is generated)

    oszi.write('TRIGGER:A:EDGE:SOURCE CH3') #specifies the Channel Waveform as the source waveform
    sleep(0.5)
    oszi.write(f'TRIGger:A:LEVel:CH3 6') #sets the trigger level 

    oszi.write('FPAnel:PRESS:SINGleseq') #pressing the single mode on the oszilloskop
    
   2) print(psu.set_voltage(10,output_num=0)) #setting voltage level
    sleep(2)    
   
    oszi.write('ACQUIRE:STATE ON')#switching the acquire state on
    sleep(5)
    
    oszi.write(f"SAVE:WAVEFORM ALL, \"E:/TEST/triggercsv\"") #saves waveform in the  file directory
    
    
  3)  while '1' in oszi.query('BUSY?'):  
        sleep(0.1)
    else:
        print('the Oszi. is currently not busy')

in 1) its the oscilloscope basic settings
by 2) i want to jump the voltage from 0V to 10V and trigger at 6V but it doesnot work.
my trigger mode is normal and i try pressing the single button automatically but it doesnot work to
3)when i am done with the trigger i want to store the waveform as .csv in my file test_folder 

please can someone help me?

MatthieuDartiailh, 2021-09-15:
Could you please the model and ideally the manual of your instrument ?

LydieNsangou, 2021-09-15:
here is a link for the Handbuch
[https://de.tek.com/oscilloscope/mso2000-dpo2000-manual/mso2000b-dpo2000b-mso2000-and-dpo2000-series](url)

LongnoseRob, 2021-09-15:
Have you tried pyvisa-shell or running your progam line by line ?

> `oszi.write(f'TRIGger:A:LEVel:CH3 6') #sets the trigger level `

is this **_f_** a typo or placed intentionally?

same a bit further down:

> `oszi.write(f"SAVE:WAVEFORM ALL, \"E:/TEST/triggercsv\"") #saves waveform in the  file directory`

same about the **_f_** here

If I remember correctly the Tek-Scopes show the trigger threshold setting on the display, so it would be an easy thing to first manually set the trigger level to lets say 0V and trigger-source  Channel 1 and then check after your program ran, if these values have changed.

LydieNsangou, 2021-09-15:
@LongnoseRob please ignore the f its a mistake i made when copying the code
i did manually and it works. but automatically nothing seems to change on the oscilloscope although i used the right commands as in the manual

LongnoseRob, 2021-09-15:
Have you tried switching on your power supply manually?
Are the probes/cables connected correctly?
This line also looks suspicious:

> `print(psu.set_voltage(10,output_num=0)) #setting voltage level`

I would recommend to remove the `print()` in that line and try again

MatthieuDartiailh, 2021-09-15:
Looking at the documentation I would suggest you try to query `TRIGGER:STATE?` after the simulated button press. If you do not get `READY` it would explain why you do not observe the expected behavior. But at this point I am not sure what you might be doing wrong.

LydieNsangou, 2021-09-16:
@yes i checked it an everything is connected correctly
this line is a command for the power supply

LydieNsangou, 2021-09-16:
@MatthieuDartiailh  thanks for your suggestion i will try it now