pyvisa/pyvisa #628: Why even use VISA?

seppestas, 2021-09-20:
Is here still a reason to use VISA?

I have been using this library mainly to connect to LAN based test equipment, but I'm starting to think it's more trouble than it's worth. I only use it for LAN devices since the Prolix GPIB controller nor the USB interface of the Minicircuits RF switch I use are supported by VISA.

This means all VISA is doing for me is opening sockets, and it's doing a pretty shitty job at that. Depending on the VISA implementation I use, sockets inexplicitly close after a certain time, crippling long running tests. This seems to be the worst with Keysight's (x64) implementation, but also occasionally happens with NI's implementation. NI only seems to support a 32-bit implementation requiring I use a 32-bit Python (at least according to our internal docs).

Sockets and serial device programming is my bread and butter. I know how to do everything by heart and it is easy with Python's built in socket library and PySerial. PyVISA, albeit well documented, requires me to look up how thing work all the time.

I can understand VISA making things easier back when GPIB was sill getting tracking and a lot of outdated protocols still needed to be supported. However now almost everything runs on top of already standardised systems like sockets, RS-232 or USB HID.

Does it make sense to just use Python sockets instead when only using LAN based devices, or is there still a good reason I'm missing for using VISA?

Sorry if this is not the right forum, I was not able to find a mailing list.

MatthieuDartiailh, 2021-09-20:
Hi,

If you only ever need RAW sockets and serial and never switch the kind of communication for a given instrument I can understand your feeling. However VISA still has value beyond GPIB, I believe (and you may be surprised how many people still use GPIB).

First, using VISA means that you can pretty easily switch the hardware connection, allowing to develop using for example USB TMC but running using over TCPIP (VXI-11). Second, TCPIP protocols beyond raw sockets (VXI-11 and hislip) which provides for example serial requests are not something one wants to re-implement from scratch every other day. Finally, if you do not want to deal with installing a VISA distribution, you can also have a look (and maybe contribute) to pyvisa-py which offers a pure Python backend.

So if your use case is limited to TCPIP raw instrument without any perspective of ever running something else, you can probably just use sockets manually. Otherwise I would still use it.

seppestas, 2021-09-20:
Thanks a lot for your response.

My use-case is primarily raw TCPIP sockets (R&S FSG and FSP spectrum analysers, Rigol PSUs and data loggers, ...) and the odd couple of GPIB devices connected through a Prologix USB to GPIB controller, for which I made my own "driver" anyway since it is not supported by VISA, see #112.

The interface I need for my test programs is fairly straight forward, most devices use a SCPI interface or some very similar command based protocol. It's fairly trivial to make "drivers" that implement this interface without needing VISA.

I have issues with the socket "sessions" expiring, not sure why but it seems to be related to the VISA library. When I use my own raw sockets I can at least debug why and re-establish the connection automatically, while with PyVISA I need to go through 2 layers of obscurity, the pyVISA library and the VISA implementation, to try to figure out what is going on.

Maybe I might switch back to pyVISA when I need to talk to some obscure hardware, but for now raw sockets seem to just work better.

An other feature I never understood is device detection. It does not seem to find any LAN test equipment automatically (though maybe I have something configured incorrectly). Some DDNS or DNS-SD like feature would be interesting to automatically find devices and address them without relying on an IP address that might change at the DHCP server's whim. Instead, all `ResourceManager.list_resources()` shows me are virtual COM ports, which are either not test equipment or test equipment not directly usable by VISA / pyVISA anyway. I understand these are all just limitations of VISA, not pyVISA.

I guess the usability depends on the setup and e.g "proper" GPIB adapters actually work great with VISA and support for obscure protocols could be a lifesaver.

Sorry for the rant, my goal is not to bash your great work, I just want to check if it actually makes sense for me to use it.