pyvisa/pyvisa #629: [COM] Communication issue with XXX using XXX

sixtemesseven, 2021-09-23:
I have a HP3577A which I was able to communicate with at some point (1-2 years ago) but now I only get timeouts. 
Back then I did not set any kind of line terminations. I just used the example and some simple commands.

The instrument will go to "remote" as indicated on the front panel. But it gets stuck there (local key does not work) and needs  a reset. 

I have the option to display the GPIB pin states. From ATN high it goes to REN high NDAC high and then back to ATN high. Then it crashes. No "invalid" warning is displayed whatsoever on the instrument.

I can read the status byte once, after that the instrument crashes (no warning, just no answer to queries and a dead front panel including local key).

I tried to go back to the old ni-visa and 488-2 installations which did not help either.

Other instruments work fine. I tried to disconnect the 3577 from the cable bus and connect directly to the NI GPIB USB Adapter as well.

Has there been an update on pyvisa which could explain this new behavior?

Instrument details
------------------
HP3577A
HP-IB / GPIB
https://www.pearl-hifi.com/06_Lit_Archive/15_Mfrs_Publications/20_HP_Agilent/HP_3577A_3-Chan_VNA/HP_3577A_Operators.pdf

Output of `pyvisa-info`
-----------------------
```
Machine Details:
   Platform ID:    Windows-10-10.0.19042-SP0
   Processor:      AMD64 Family 23 Model 113 Stepping 0, AuthenticAMD

Python:
   Implementation: CPython
   Executable:     C:\Users\justrandom\anaconda3\envs\simplevisa\python.exe
   Version:        3.9.7
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Sep 14 2021 01:11:01 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
```

MatthieuDartiailh, 2021-09-23:
I do not have any specific change in mind that could have caused your issue. However there is a known issue regarding GPIB address normalization that in some corner cases can be problematic. Try to have a look at https://github.com/pyvisa/pyvisa/issues/617#issuecomment-891754549 and let me know if it helps.

sixtemesseven, 2021-09-23:
> I do not have any specific change in mind that could have caused your issue. However there is a known issue regarding GPIB address normalization that in some corner cases can be problematic. Try to have a look at [#617 (comment)](https://github.com/pyvisa/pyvisa/issues/617#issuecomment-891754549) and let me know if it helps.

Thank you for your answer!
I use this format to address the instrument:

`
rm = pyvisa.ResourceManager()
print(rm.list_resources())
visaID = "GPIB0::11::INSTR"
i = rm.open_resource(visaID)
i.close()
`

Prints: 
`_('ASRL1::INSTR', 'ASRL10::INSTR', 'GPIB0::11::INSTR')_`

Interestingly, opening the instrument also causes a crash.
So the above works once, but not twice.

`
rm = pyvisa.ResourceManager()
print(rm.list_resources())
visaID = "GPIB0::11::INSTR"
i = rm.open_resource(visaID)
print(i.read_stb())
`

Returns:
```
('ASRL1::INSTR', 'ASRL10::INSTR', 'GPIB0::11::INSTR')
16
```

Also works once, returns the status byte and then crashes.
So strange...

sixtemesseven, 2021-09-23:
I just noticed that running the code above twice actually results in:  
```
('ASRL1::INSTR', 'ASRL10::INSTR', 'GPIB0::11::INSTR')
16
('ASRL1::INSTR', 'ASRL10::INSTR')
```

So the instrument disappears from the GPIO0 bus completely?

sixtemesseven, 2021-09-23:
Damit!! Now it works.
I downgraded nivisa so far that I tought my fake NI GPIB USB adapter might work again and it did. Now I can read the instrument with the fake adapter but not with the original.. It is on GPIB1::11::INSTR so I have to reboot and test it on GPIB0.

MatthieuDartiailh, 2021-09-23:
That's a weird one for sure. Good luck with your tests.