pyvisa/pyvisa #630: 8 Channel Scope Screen Image transfer failure

clee1606, 2021-09-30:
I have been using this below python command 
"binary = scope.query_binary_values(':DISPlay:DATA? %s,%s,%d,%s' % ('JPG', 'SCReen', 0, 'NORMal'), 'B', False)"
to transfer screen image from 4 channel scope into my pc successfully all the while. 

But it is not working for my 8 channel scope and it return below error message :
===================================================================================
Traceback (most recent call last):
  File "C:/Users/clee16/Desktop/Python/test.py", line 41, in <module>
    scope_waveform()
  File "C:/Users/clee16/Desktop/Python/test.py", line 33, in scope_waveform
    binary = scope.query_binary_values(':DISPlay:DATA? %s,%s,%d,%s' % ('JPG', 'SCReen', 0, 'NORMal'), 'B', False)
  File "C:\Python38\lib\site-packages\pyvisa\resources\messagebased.py", line 747, in query_binary_values
    return self.read_binary_values(
  File "C:\Python38\lib\site-packages\pyvisa\resources\messagebased.py", line 575, in read_binary_values
    block = self._read_raw(chunk_size)
  File "C:\Python38\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Python38\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Python38\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Python38\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

===================================================================================

I am thinking the 8 channel binary files are much more bigger than 4 channel binary files. Any fix for it ?

LongnoseRob, 2021-09-30:
Based from what I see,
You are trying to read the screen data as a JPG.
Is this really what you want?

From my basic understanding this should work independent how many channels displayed.
After the command failed, it might be wise to check the error register of the instrument usIng the :SYSTem:ERRor? command
However as you did not mention the type of Oscilloscpe and what interface you are using.

Normally modern oscilloscopes have a special command as part of their SCPI implementation, which would give you the raw data for each channel.
On modern Keysight scope this would be the command block WAVeform, esp. WAVeform:DATA?

MatthieuDartiailh, 2021-09-30:
Since the two lines of products are different there may be some incompatibilities between them. However, it is also possible that the image is just bigger and the transfer time exceed the default timeout. I would recommend you to:
- try increasing the timeout (`instr.timeout = 30`)
- if it fails follow the methodology outlined in https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html#reading-binary-values and https://pyvisa.readthedocs.io/en/latest/introduction/rvalues.html#when-things-are-not-what-they-should-be. In prticular try to manually read the message using `read_bytes` to determine whether or not the termination character is appended. I have seen different behavior in that respect for different Keysight instruments.

clee1606, 2021-09-30:
> Based from what I see, You are trying to read the screen data as a JPG. Is this really what you want? From my basic understanding this should work independent how many channels displayed. After the [...]

Thank you for the look in . Yes , i want to read the screen image of the scope .

clee1606, 2021-09-30:
> Since the two lines of products are different there may be some incompatibilities between them. However, it is also possible that the image is just bigger and the transfer time exceed the default [...]

Great thanks ! 
i added 
scope.timeout = 4000 at the opening and it works !!! 
Fantastic , matthieu

closed 2021-09-30 by clee1606

MatthieuDartiailh, 2021-09-30:
You are welcome !