pyvisa/pyvisa #633: [COM] Communication issue with MSO44 using USB

yonixiong, 2021-11-04:
Instrument details
------------------
* Model: MSO44
* Communication: USB
* Link to the documentation (if available): [https://download.tek.com/manual/4-5-6Series-MSO-Programmer-Interface-Manual_EN-US_077130508.pdf](url)

Output of `pyvisa-info`
-----------------------
```
Microsoft Windows [Version 10.0.19042.1288]
(c) Microsoft Corporation. All rights reserved.

Machine Details:
   Platform ID:    Windows-10-10.0.19041-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 12, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\rerus\AppData\Local\Programs\Python\Python39\python.exe
   Version:        3.9.4
   Compiler:       MSC v.1928 64 bit (AMD64)
   Bits:           64bit
   Build:          Apr  6 2021 13:40:21 (#tags/v3.9.4:1f2e308)
   Unicode:        UCS4

PyVISA Version: 1.10.0.dev0

Backends:
   ni:
      Version: 1.10.0.dev0 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 22020096
         Spec. Version: 5244928
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 22020096
         Spec. Version: 5244928
```

Description of issue: 
I am doing continuous and autonomous frequency measurement on some test devices. Every couple of hours or so (anywhere form 1 hour to 50 hours) my oscilloscope will freeze and data communication will cease to work. Other users online report that this may be because scopes do not respond well to communication errors. I did some capture tracing with NI I/O trace to give some insight on what is happening. It looks like the scope just fails to respond to a *OPC? query after some irregular amount of time... I also included the measurement function in my python code where the issue seems to occur. My python script attempts to close the resource and reconnect to it, but this does not work because the scope is not responsive after this supposed communication error. Any queries or reads like *IDN? will timeout (which is set to 7 minutes for my program). Any help or suggestions would be greatly appreciated! 

Measurement Code: 
```
# single shot acq is configured before this function call using 
#    scope.write('acquire:state off\n')
#    scope.write('acquire:stopafter sequence\n')
def measureFreq(scope, ro, channel, numberOfTimes):
    freqList = [] 
    success = 0
    tried = 0

    # set the waveview display, horizonal/vertical scaling based on ro and channel params

    scope.write('measurement:immed:type frequency\n')
    while success < numberOfTimes and tried < 50:
        time.sleep(0.01)
        scope.write("trigger force\n")
        scope.write('acquire:state on\n')
        # wait for acquisition to complete
        scope.query('*opc?',delay=0.1)

        # make the measurement
        freqValue = float(scope.query('measurement:immed:value?',delay=0.1))
        if freqValue < 9.9E+37: 
            freqList.append(freqValue)
            success += 1
        tried += 1
    scope.write('acquire:state off\n')
    return freqList
```
[image]
[image]
[image]

LongnoseRob, 2021-11-04:
This could be related to the instrument itself.

And  quick look trough the [release notes](https://download.tek.com/software/supporting_files/releaseNotes_066201811.txt) for the  MSO firmware shows in line 219:  
_On 4 Series MSO with eScope, *OPC? command does not work sometimes on raw socket connection_

I would recommend to do a firmware update on the instrument, if possible.

yonixiong, 2021-11-05:
I have updated the firmware to the most recent version (v1.36.2) like you mentioned. I will see it that fixes the issue, but the problem does seem to be exactly described in the release notes. Thank you for your help!

yonixiong, 2021-11-07:
The scope was frozen again after 2 days of running after the firmware update. I am restarting the setup now with an NI I/O trace to see if it was due to another *OPC timeout. Any ideas in the meantime?

LongnoseRob, 2021-11-07:
@yonixiong running the NI I/O trace, is a good idea.
Another Idea would be to cyclic close the conneciton to the device, if this would be possible.
If it still would be the same problem, would it be possible for you to switch your setup and try TCP/IP or GPIB communication with the instrument?

yonixiong, 2021-11-07:
By cyclic close do you mean I would close and reopen the connection every so often? I could set a timer in my main loop to reset the connection every 6 hours. The scope does not have a GPIB port, but I used to use the ethernet port for TCP/IP connection and the OPC timeout was still a problem. Would you recommend switching back to LAN?

LongnoseRob, 2021-11-07:
> By cyclic close do you mean I would close and reopen the connection every so often? I could set a timer in my main loop to reset the connection every 6 hours.
Exactly, something like this.

> The scope does not have a GPIB port, but I used to use the ethernet port for TCP/IP connection and the OPC timeout was still a problem. Would you recommend switching back to LAN?

Did you also try ethernet and TCP/IP conenction since the FW update.
But it oculd be that the problem is not related to the communication hardware, so it would make maybe no difference between USB or TCP/IP.

Another option is still to take the NI I/O trace data and communicate to the Manufacturer abiut this problem, as these Instruments are still being sold.

yonixiong, 2021-11-08:
It looks like I need to take this issue to the manufacturer. I added a cyclic close to my program, and an OPC timeout happened 15 minutes after a successful refresh of the connection. Hopefully, they can help me... At least it does not seem to be an issue stemming from PyVISA. Thanks for the valuable discussion here!

Trace for after firmware update:
[image]

Trace for after adding cyclic close: 
[image]