pyvisa/pyvisa #636: [COM] Communication issue with Keithley 2602A using TCPIP

Max1993914, 2021-11-17:
Instrument details
------------------
* Model: Keithley 2062A
* Communication: TCPIP

Hello MatthieuDartiailh:

I was trying to write an auto-scan ip function recently, I use the socket try to connect with each instruments to confirm the status. but after that when I use visa to send command again, it reports error:

(This error only occur when use 2602A. 2450 has no such problem)

`In [1]: import socket

In [2]: s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)

In [3]: s.connect(('10.235.58.43', 5025))

In [4]: s.send("*idn?\n".encode())
Out[4]: 6

In [5]: s.recv(2048)
Out[5]: b'Keithley Instruments Inc., Model 2602A, 1219110, 2.2.6\n'

In [6]: import visa

In [7]: rm = visa.ResourceManager()

In [8]: s.close()

In [9]: a = rm.open_resource('TCPIP0::10.235.58.43::INSTR')

In [10]: a.query('*idn?')
---------------------------------------------------------------------------
VisaIOError                               Traceback (most recent call last)
<ipython-input-10-cf1f493b55a8> in <module>
----> 1 a.query('*idn?')

c:\python37-32\lib\site-packages\pyvisa\resources\messagebased.py in query(self, message, delay)
    611         if delay > 0.0:
    612             time.sleep(delay)
--> 613         return self.read()
    614
    615     def ask(self, message, delay=None):

c:\python37-32\lib\site-packages\pyvisa\resources\messagebased.py in read(self, termination, encoding)
    425         if termination is None:
    426             termination = self._read_termination
--> 427             message = self._read_raw().decode(enco)
    428         else:
    429             with self.read_termination_context(termination):

c:\python37-32\lib\site-packages\pyvisa\resources\messagebased.py in _read_raw(self, size)
    398                     logger.debug('%s - reading %d bytes (last status %r)',
    399                                  self._resource_name, size, status)
--> 400                     chunk, status = self.visalib.read(self.session, size)
    401                     ret.extend(chunk)
    402             except errors.VisaIOError as e:

c:\python37-32\lib\site-packages\pyvisa\ctwrapper\functions.py in read(library, session, count)
   1582     buffer = create_string_buffer(count)
   1583     return_count = ViUInt32()
-> 1584     ret = library.viRead(session, buffer, count, byref(return_count))
   1585     return buffer.raw[:return_count.value], ret
   1586

c:\python37-32\lib\site-packages\pyvisa\ctwrapper\highlevel.py in _return_handler(self, ret_value, func, arguments)
    191
    192         if ret_value < 0:
--> 193             raise errors.VisaIOError(ret_value)
    194
    195         if ret_value in self.issue_warning_on:

VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
`

MatthieuDartiailh, 2021-11-17:
A socket instrument cannot accept more than one connection at any given time. As a consequence you need to close your socket before you can open the visa resource.

Let me know if that helps.

Max1993914, 2021-11-17:
Wow, your answer is really fast, thanks a lot for your response.

I have terminated the socket in line 8 in the code. but still I cannot use visa.query.
And If I use the visa.query first and close it, and then use socket, it cannot connect either.
This is really weird, this problem only occur for Keithley 2602A.

MatthieuDartiailh, 2021-11-17:
Sorry for missing the s.close...

That is quite weird. Can you try opening and closing multiple times the visa resource ? I would like to know if it is somehow related to using different libraries or simply disconnecting and reconnecting too quickly. If you see the same issue closing and reopening with visa try to add a sleep between the 2 (1 to 5 s maybe) and see if that helps.

Max1993914, 2021-11-18:
Thanks, I already solved the problem.
When I reconnect the instrument with visa, I have to send 'abort' to the instrument and then the old connect can be completely released.

Hope this can help someone who has the same issue in the future.

MatthieuDartiailh, 2021-11-18:
You mean a scpi abort command right ? That's weird, it sounds like you did not complete reading the instrument response. Could you check that the output buffer is fully empty (before closing or after reopening using read_bytes).