pyvisa/pyvisa #637: [COM] Communication issue with HP 3562A using Prologix GPIB-USB

bfailing, 2021-11-19:
I'm trying to dump trace data in ASCII format (command "DDAS").  This is discussed on pages 3-2 (PDF page 40) and 3-5 to 3-8 (PDF pages 43-46) of the below-linked Programming Manual.

In Tera Term, the only changes that need to be made are those mentioned in the Prologix GPIB-USB FAQ (number 17 on http://prologix.biz/gpib-usb-6.0-faq.html).  Specifically: "Open Setup | Terminal dialog. Set Transmit to CR+LF, and check Local echo. Click OK."  With those changes, Tera Term returns the data header and data (each element on a separate line) responsive to a DDAS command.  At a minimum, this is 468 elements (1 element that indicates the total number of elements, 66 elements for the header, and 401 elements for the data).

However, the following entered in the Python Interpreter only results in the return of a single element:
```
import pyvisa
rm = pyvisa.ResourceManager()
my_instrument = rm.open_resource('ASRL7::INSTR')
print(my_instrument.query('DDAS'))
```
And each subsequent query operation returns the next element - as if they are being queued by the instrument.  For example, issuing a "RDY?" command (discussed on page 6-15 of the below-linked Programming Manual at PDF page 141) or another "DDAS" command via a query operation will return the next element only.

In an attempt to configure PyVISA like Tera Term, I tried the following:
```
my_instrument.write_termination = '\r\n'
```
But it still behaves the same.

Continually issuing read operations may work as each read operation returns the next element.  However, I'm wondering why this is necessary given that Tera Term returns the entire ASCII data dump at once.

Is it possible to access the entire ASCII output with a single query operation?

Instrument details
------------------
* Model: HP 3562A
* Communication: Serial via Prologix GPIB-USB
* Link to the documentation (if available): https://xdevs.com/doc/HP_Agilent_Keysight/HP%203562A%20Programming.pdf

Output of `pyvisa-info`
-----------------------
Machine Details:
   Platform ID:    Windows-10-10.0.19042-SP0
   Processor:      AMD64 Family 23 Model 113 Stepping 0, AuthenticAMD

Python:
   Implementation: CPython
   Executable:     C:\Program Files\Python\Python310\python.exe
   Version:        3.10.0
   Compiler:       MSC v.1929 64 bit (AMD64)
   Bits:           64bit
   Build:          Oct  4 2021 19:00:18 (#tags/v3.10.0:b494f59)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments

LongnoseRob, 2021-11-19:
Hello,
did you have look at #112?
or try the prologix adapter in [pymeasure](https://pymeasure.readthedocs.io/en/latest/api/adapters.html#prologix-adapter)

douardda, 2023-02-10:
When using a prologix compatible adapter, you may try my old pygpibtoolkit project. Never got it to the point I wanted back then, but I wrote a GUI interface to control the HP 3562A:

[image]
[image]

See https://pygpibtoolkit.readthedocs.io/en/latest/index.html and the code is now there https://hg.sr.ht/~douardda/pygpibtoolkit (yes it's a mercurial repo :-) )

There is also HPGL plotter in there:

[image]