pyvisa/pyvisa #638: [COM] Communication issue with SR830 Using RS232 with Usb adapter

horoshho, 2021-11-23:
CODE:
import pyvisa
rm = pyvisa.ResourceManager()
print(rm.list_resources())
my_instrument = rm.open_resource('ASRL5::INSTR')
my_instrument.write('*IDN?')
while True:
    print(my_instrument.read_bytes(1))

ERROR:
Traceback (most recent call last):
  File "C:/Users/averin/Desktop/08.11.2021/07.11.2021/_project/Newlaser/lockin.py", line 11, in <module>
    print(my_instrument.read_bytes(1))
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 371, in read_bytes
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

MESSAGE:
Could you please help me to resolve this issue?

horoshho, 2021-11-23:
UPD:
Same output for commands:
my_instrument.write('*IDN? <lf>')
my_instrument.write('*IDN?<lf>')

MatthieuDartiailh, 2021-11-23:
Several things you can try (base on the manual):
- set the termination character (`write_termination="\n", read_termination="\n"`)
- set the baud_rate when opening the resource
- set other serial specific configuration (bitness, parity, etc)

horoshho, 2021-11-23:
Thank you for quick response!

CODE:
import pyvisa
from pyvisa.constants import StopBits, Parity
rm = pyvisa.ResourceManager()
print(rm.list_resources())
my_instrument = rm.open_resource('ASRL5::INSTR',
    baud_rate=9600, data_bits=8,
    parity=Parity.odd)
my_instrument.read_termination = '\n'
my_instrument.write_termination = '\n'
my_instrument.query('*IDN? <lf>')

ERROR:
Traceback (most recent call last):
  File "C:/Users/averin/Desktop/08.11.2021/07.11.2021/_project/Newlaser/lockin2.py", line 16, in <module>
    my_instrument.query('*IDN? <lf>')
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 644, in query
    return self.read()
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

Process finished with exit code 1

MESSAGE:
I have also tried without data bits and (baud_rate=9600, data_bits=7, flow_control=2,
parity=Parity.odd, stop_bits=StopBits.one). I am not completely aware what should be that parameters, on device
I can only adjust cable, boud, and parirty.
I put the same values but it doesnt work(

horoshho, 2021-11-23:
I also have another device PM320E Thorlabs. 
It reads print(my_instrument.query('*IDN?\n'))
>>>Thorlabs,PM320E,M00257794,2.2.0
But dont reads
my_instrument.query(':POW[1]:VAL?')
>>>Traceback (most recent call last):
  File "C:/Users/averin/Desktop/08.11.2021/07.11.2021/_project/Newlaser/lockin.py", line 7, in <module>
    (my_instrument.query(':POW[1]:VAL?'))
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 644, in query
    return self.read()
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

Process finished with exit code 1

May be it makes something more clear, idk sorry, I am not experienced in that.

horoshho, 2021-11-23:
Okay, I will try.
And what about 1-st one?

MatthieuDartiailh, 2021-11-23:
Try the delay and if you share the manual, I can try to cross-check the parameters you use for the serial connection.

horoshho, 2021-11-23:
I have tried delay, same error
CODE:
>import pyvisa
rm = pyvisa.ResourceManager()
print(rm.list_resources())
my_instrument = rm.open_resource('USB0::0x1313::0x8022::M00257794::INSTR')
print(my_instrument.query('*IDN?\n', delay=1))
(my_instrument.query(':POW[1,2]:VAL?\n', delay=1))
>
OUTPUT:
>('USB0::0x1313::0x8022::M00257794::INSTR', 'ASRL1::INSTR', 'ASRL5::INSTR')
Thorlabs,PM320E,M00257794,2.2.0
>
ERROR:
>Traceback (most recent call last):
  File "C:/Users/averin/Desktop/08.11.2021/07.11.2021/_project/Newlaser/lockin.py", line 8, in <module>
    (my_instrument.query(':POW[1,2]:VAL?\n', delay=1))
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 644, in query
    return self.read()
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
>
Link to manual:https://www.thorlabs.com/drawings/8791fc5acddd9aba-037101EB-A058-EFD3-C9BEFD8BE8C564EA/PM320E-Manual.pdf

MatthieuDartiailh, 2021-11-23:
I believe the proper command is `my_instrument.query(':POW1:VAL?\n')` for the channel 1 and `my_instrument.query(':POW2:VAL?\n')` for the channel 2. The manual could be a bit more clear...

horoshho, 2021-11-23:
Yeeei, that works)
Do you have any ideas what could be wrong with first device?
Thank you!!

MatthieuDartiailh, 2021-11-23:
As I suggested try the delay for the first device and if you can also share the manual.

horoshho, 2021-11-23:
https://www.thinksrs.com/downloads/pdfs/manuals/SR830m.pdf- manual
>import pyvisa
from pyvisa.constants import StopBits, Parity
rm = pyvisa.ResourceManager()
print(rm.list_resources())
my_instrument = rm.open_resource('ASRL5::INSTR',
    baud_rate=9600,
    parity=Parity.odd)
my_instrument.read_termination = '\n'
my_instrument.write_termination = '\n'
my_instrument.query('*IDN? <lf>',delay=1)
>
ERROR:
>Traceback (most recent call last):
  File "C:/Users/averin/Desktop/08.11.2021/07.11.2021/_project/Newlaser/lockin2.py", line 16, in <module>
    my_instrument.query('*IDN? <lf>',delay=1)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 644, in query
    return self.read()
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

Process finished with exit code 1

>

horoshho, 2021-11-24:
May be cable is damaged? Or I need to install additional drivers? BTW I use connector from rs232 to USB. If I will try use GPIB and buy USB connector to it do I need install some drivers or additional libraries. As I am concerned there is interface for that device to connect via GPIB from NI. But do I need something else if I will use adapter?

MatthieuDartiailh, 2021-11-24:
If you are on windows and you have NI-VISA, Keysight VISA or any other major provider VISA implementation you should not need anything else.

The SRS mention a debug mode in its manual that displays the character received by the instrument on the front panel you could try to use that to check if the right command is received.

horoshho, 2021-11-24:
If you mean queue bottom page 4-24, which should displays received message, it shows nothing.

MatthieuDartiailh, 2021-11-24:
Then it may be an adapter issue. I must say I am out of ideas at this point.

LongnoseRob, 2021-11-24:
Here are some more hints/ideas what to check:

- Have you double checked the correct com-port in the device manager?
- Have you tried to use Teraterm or other terminal software to see if you can communicate with the device?
- Could it be that some other software block access to the port? 
- have you tried it after a reboot?
- assuming this is a 9-pin D-Sub connector, if you jumpers pin 2-3 you should receive all the text you send(type) with a terminal software like Teraterm

horoshho, 2021-12-17:
I have changed cable to first device.
Now I can recieve signal from it in YAT.
e.g. 
>*IDN?
SStanford_Research_Systems,SR830,s/n43480,ver1.07 <CR>Stanford_Research_Systems,SR830,s/n43480,ver1.07 <CR>0<CR>0<CR>6.10356e-005<CR>-0.000122071<CR>Stanford_Research_Systems,SR830,s/n43480,ver1.07 <CR>
OUTP? 1 
6.10356e-005<CR><

But pyvisa still shows error:
CODE:

>import pyvisa
from pyvisa.constants import StopBits, Parity
rm = pyvisa.ResourceManager()
print(rm.list_resources())
my_instrument = rm.open_resource('ASRL5::INSTR',
    baud_rate=9600, data_bits=8,
    parity=Parity.none)
my_instrument.read_termination = '\n'
my_instrument.write_termination = '\n'
my_instrument.write('*IDN? <lf>',delay=1)<

ERROR

>Traceback (most recent call last):
  File "C:/Users/averin/Desktop/08.11.2021/07.11.2021/_project/Newlaser/lockin2.py", line 10, in <module>
    parity=Parity.none)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\highlevel.py", line 3304, in open_resource
    res.open(access_mode, open_timeout)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\resource.py", line 298, in open
    self._resource_name, access_mode, open_timeout
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\highlevel.py", line 3232, in open_bare_resource
    return self.visalib.open(self.session, resource_name, access_mode, open_timeout)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 1852, in open
    session, resource_name, access_mode, open_timeout, byref(out_session)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_RSRC_BUSY (-1073807246): The resource is valid, but VISA cannot currently access it.
>
Settings in YAT attache
[image]
[image]
d

horoshho, 2021-12-17:
I can send command to change state of device via pyvisa and it will changes, but the same error still ocuurs and *IDN? command doesnt work

MatthieuDartiailh, 2021-12-17:
From the code you pasted it looks like there is a space after the IDN command or am I misreading. If the command is correct, can you read a single bytes using read_bytes(1) ?

horoshho, 2021-12-21:
>import pyvisa
from pyvisa.constants import StopBits, Parity
rm = pyvisa.ResourceManager()
print(rm.list_resources())
my_instrument = rm.open_resource('ASRL5::INSTR',
    baud_rate=9600, data_bits=8,
    parity=Parity.none)
my_instrument.read_bytes(1)<

Error

>Traceback (most recent call last):
  File "C:/Users/averin/Desktop/08.11.2021/07.11.2021/_project/Newlaser/lockin2.py", line 12, in <module>
    my_instrument.read_bytes(1)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\resources\messagebased.py", line 371, in read_bytes
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\averin\PycharmProjects\pythonProject\venv\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.<

horoshho, 2021-12-21:
I have tried that way:

>import pyvisa
from pyvisa.constants import StopBits, Parity
rm = pyvisa.ResourceManager('@py')
print(rm.list_resources())
my_instrument = rm.open_resource('ASRL5::INSTR',
    baud_rate=9600, data_bits=8, write_termination = '\n',read_termination='\n',
    parity=Parity.none)
my_instrument.timeout=10000
my_instrument.write('*IDN? <lf>')
my_instrument.read_bytes(1)>

Also, the same error

horoshho, 2021-12-21:
The problem why I cannot use standart NI driver for this device is that NI has driver only for GPIB connection of this device, when I am using RS232.

LongnoseRob, 2021-12-21:
> The problem why I cannot use standart NI driver for this device is that NI has driver only for GPIB connection of this device, when I am using RS232.

NI MAX should also support RS232 connections via NI VISA.

Just had a look at the [manual of the SR830, especially page 4-24](https://courses.physics.illinois.edu/phys406/sp2017/Lecture_Notes/P406POM_Lecture_Notes/SR830_Manual.pdf) and came up with two points to check for you:
1. is the device propely set to RS232?
2. do Speed & Parity setting match with your code? 
3. have you tried to check if the QUEUE matches the string sent from your code(e.g. `*IDN?`) ?
4. page 4-27 mentions to use a straight RS232-cable, no null-modem-cable for the connection. have you double checked that the cable is wired correctly?

horoshho, 2021-12-21:
1. Yes
2. Yes
3. In queue there are always some codes like(3c4c 463E 10R0R)
But its doing commands I send to it, just not returning anything.
4. I use adapters for rs232-USB and gender changer for rs232, but as I previously mentioned it recieves commands, and in YAT app it works completely properly, it both sends and recieves.

LongnoseRob, 2021-12-21:
Your string seems to be 

```
0x3C  <
0x4C  L
0x46  F
0x3e  >
0x10  'DEL' [backspace]
0x0A  'LF' [linefeed]
0x0A  'LF'
```
 (if we read the 0R0R as actually 0A0A as mentioend in the manual).

So try to find out what is sending `<LF>`,a Backspace and two linefeeds.
Are you 100% sure that no other piece of software is accessing the COM port associated with `ASRL5::INSTR` and could maybe also block reading from the port..

horoshho, 2021-12-21:
SO when sending via YAT, I get for
OUTP? 1 <lf>-3F20 2120 0A0A
FMOD 1 <lf>-4420 3120 0A0A
*IDN? <lf>- 444E 3F20 0A0A
Via pyvisa:
3c4c 463E 0R0R- same for all commands mentioned above,
and after restart symbol 1 disappeard

horoshho, 2021-12-21:
Okay, while I am using command without lf in brackets now queue same for both YAT and pyvisa

horoshho, 2021-12-21:
But there are still error
And I have turned off all other terminals and restart pc and sr

MatthieuDartiailh, 2021-12-21:
As I mentioned earlier you send `my_instrument.write('*IDN? ')` instead of `my_instrument.write('*IDN?')` (no trailing space). Can you check if this fixes your issue ?

horoshho, 2021-12-22:
Okay I can get response if I initiate code:
>import pyvisa
from pyvisa.constants import StopBits, Parity
rm = pyvisa.ResourceManager('@py')
print(rm.list_resources())
my_instrument = rm.open_resource('ASRL5::INSTR',
    baud_rate=9600, data_bits=8, write_termination = '\n',read_termination='\n',
    parity=Parity.none)
my_instrument.timeout=1000
my_instrument.write('*IDN? ')
###print(my_instrument.read_bytes(100))

And then initiate:
>
>import pyvisa
from pyvisa.constants import StopBits, Parity
rm = pyvisa.ResourceManager('@py')
print(rm.list_resources())
my_instrument = rm.open_resource('ASRL5::INSTR',
    baud_rate=9600, data_bits=8, write_termination = '\n',read_termination='\n',
    parity=Parity.none)
my_instrument.timeout=1000
my_instrument.write('*IDN? ')
print(my_instrument.read_bytes(100))

If I start last code second time error occurs again.

horoshho, 2021-12-22:
It also works this strange way:
>import pyvisa
from pyvisa.constants import StopBits, Parity
rm = pyvisa.ResourceManager('@py')
print(rm.list_resources())
my_instrument = rm.open_resource('ASRL5::INSTR',
    baud_rate=9600, data_bits=8, write_termination = '\n',read_termination='\n',
    parity=Parity.none)
my_instrument.timeout=1000
import time
my_instrument.write('*IDN? ')
time.sleep(3)
my_instrument.write('*IDN? ')
print(my_instrument.read_bytes(100))

horoshho, 2021-12-22:
But that is dont work for OUTP? 1  command

horoshho, 2021-12-22:
If I send query OUTP? 1  and instantly open YAT terminal I will immediately recive answer, but not in pyvisa

horoshho, 2021-12-22:
Okay, I have changed amount of bytes to read and it seems that it works

MatthieuDartiailh, 2021-12-22:
If this works you can go back to simply using query (possibly with a delay), which is equivalent to write followed by read.  read_bytes is mostly meant for debugging purposes.

hanwang365, 2024-04-26:
> I believe the proper command is `my_instrument.query(':POW1:VAL?\n')` for the channel 1 and `my_instrument.query(':POW2:VAL?\n')` for the channel 2. The manual could be a bit more clear...

Thank you for sharing a workable reference. The manual is too rough, even no example.
I met a problem when I wanted to modify the currently setting wavelength to 850 nm. My command is ''my_instrument.write(':WAVEL1:850\n')" or ''my_instrument.write(':WAVEL1:8.50E+02\n')", but didn't work.
Would you mind giving me some ideas? Thank you for your help in advance.

hanwang365, 2024-04-26:
> Thank you for sharing a workable reference. The manual is too rough, even no example. I met a problem when I wanted to modify the currently setting wavelength to 850 nm. My command is [...]

Update:
I completed this setting function through "my_instrument.write(':WAVEL1:VAL\s850\n')".
If I want to tuning the power range, I should use "my_instrument.write(':PRANGE1\sR1MW\n')" or "my_instrument.write(':PRANGE1\sAUTO\n')". The highest power range should be R100MW. When I use R1W, it doesn't work.

MatthieuDartiailh, 2024-05-15:
This thread has become hard to follow for me. Could you share what instrument you are using and a working link to the manual ?

hanwang365, 2024-05-15:
> This thread has become hard to follow for me. Could you share what instrument you are using and a working link to the manual ?

Thank you very much for the reply, Matthieu. I solved this problem and it works well now. No worries. Still thanks for the reply.

closed 2024-05-16 by MatthieuDartiailh