pyvisa/pyvisa #639: [COM] Communication issue with GPIB Proxy/Mapper/Master

bytelinker, 2021-11-25:
Hello,
some commercial programs (closed src, os Windows 7,10) only accept GPIB0 as connection route and only the device number can be chosen.

Is there any method available to map a usb or tcp scipy device on GPIB0:n?  This might be solves with NI-VISA.

Is it possible to run some code (python) as GPIB0:n device? So this could be used as proxy to send and return commands to e.g. a tcp device. This setup could also be used to emulate a device needed in the commercial software by translating the commands and answers to an alternative device.

Regrads Bytelinker

MatthieuDartiailh, 2021-11-25:
I am not sure I follow your question so let me rephrase.

Because you have a blackbox that expects an instrument on GPIB0:n, you would like a way to emulate using python such an instrument. Is that right ?

 If the closed source program does not use pyvisa, you will have to find a solution through the VISA library used by the program. If it is dynamically linked and NI-VISA allow you perform such a redirection then you are set. If you want to use Python, you can make a use a socket on 127.0.0.1 (loop back). Ideally you would need to redirect to a RAW Socket resource to avoid dealing with the VXI-11 protocol.

In all that I am not sure where pyvisa can help. In Pyvisa-py you may find examples to help you write your mock-up instrument though.

bytelinker, 2021-11-25:
pyvisa can connect to a GPIB0 device, this is the normal way of using it. I would like to make pyvisa acting and registering as a GPIB0, that can be used by the blackbox via NI-VISA on an Windows device. So i could proxy the communication to a second device including command modification.
 
Regarding the redirection I didn't find much.

MatthieuDartiailh, 2021-11-25:
PyVISA is basically a wrapper for VISA and is meant to act as controller. It cannot act as a device. Honestly I have no idea if this possible without hacking VISA by replacing the gpib-32.dll by your own, and even that may not be possible.

Anyway I wish you good luck and let me know if your figure out anything.