pyvisa/pyvisa #640: [COM] Communication issue with XXX using XXX

sixtemesseven, 2021-12-01:
I use a HP3577 connected via a genuine NI USB-GPIB-HS adapter to my host PC. It used to work last week, then I suddenly couldn't run my scripts any more. I noticed that even a minimal code:

`
import pyvisa as visa

rm = visa.ResourceManager()
print(rm)
hp = rm.open_resource('GPIB0::11::INSTR')
print(hp)
hp.write('AB4')
`

`
Resource Manager of Visa Library at C:\WINDOWS\system32\visa32.dll
GPIBInstrument at GPIB0::11::0::INSTR
Traceback (most recent call last):

  File "D:\Github_Sixtimesseven\VisaInstruments\Simplevisa\SimpleVisa\untitled0.py", line 14, in <module>
    hp.write('AB4')

  File "C:\Users\justrandom\anaconda3\envs\ba21\lib\site-packages\pyvisa\resources\messagebased.py", line 197, in write
    count = self.write_raw(message.encode(enco))

  File "C:\Users\justrandom\anaconda3\envs\ba21\lib\site-packages\pyvisa\resources\messagebased.py", line 157, in write_raw
    return self.visalib.write(self.session, message)[0]

  File "C:\Users\justrandom\anaconda3\envs\ba21\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2796, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))

  File "C:\Users\justrandom\anaconda3\envs\ba21\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore

  File "C:\Users\justrandom\anaconda3\envs\ba21\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)

VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
`

Causes the instrument to go into remote and lock up (power cycle required), it does not set average mode 4 (AB4 command) which is tested and should work and has worked last week.
I tried another instrument and it was not working as well (Keithley 238). 

I tried another PC and uninstalling visa and the 488-2 drivers. I reinstalled the oldest (17.6) version and updated pyvisa (it was already up to date). Right now I can talk to the other instruments but the HP3577 locks up as before.

Instrument details
------------------
* Model: HP3577A
* Communication: GPIB
* Link to the documentation (if available): [https://www.pearl-hifi.com/06_Lit_Archive/15_Mfrs_Publications/20_HP_Agilent/HP_3577A_3-Chan_VNA/HP_3577A_Operators.pdf](url)

Output of `pyvisa-info`
-----------------------
`
C:\Users\justrandom\anaconda3\envs\ba21\lib\site-packages\gpib_ctypes\gpib\gpib.py:54: UserWarning: GPIB library not found. Please manually load it using _load_lib(filename). All GPIB functions will raise OSError until the library is manually loaded.
  warnings.warn(message)
Machine Details:
   Platform ID:    Windows-10-10.0.19042-SP0
   Processor:      AMD64 Family 23 Model 113 Stepping 0, AuthenticAMD

Python:
   Implementation: CPython
   Executable:     C:\Users\justrandom\anaconda3\envs\ba21\python.exe
   Version:        3.9.7
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Sep 23 2021 07:24:41 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
   py:
      Version: 0.5.2
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
`

MatthieuDartiailh, 2021-12-01:
Not sure why that would have spontaneously changed but I would suggest setting both write_termination and read_termination to \n since the instrument expect EOI or \n at the end of a command.

sixtemesseven, 2021-12-02:
> 

Interesting thank you, strange that it worked before without. To test it I changed the script to:

# -*- coding: utf-8 -*-
"""
Created on Tue Nov 30 16:26:52 2021

@author: justrandom
"""

import pyvisa as visa

```
rm = visa.ResourceManager()
print(rm)
hp = rm.open_resource('GPIB0::11::INSTR')
hp.read_termination = '\n'
hp.write_termination = '\n'
print(hp)
hp.write('AB4')
```

But it still locks up the instrument.

sixtemesseven, 2021-12-02:
Well now it's back to the old error. I hate gpib :(

```
runfile('D:/Dropbox (icts)/Dropbox (icts)/Github_Sixtimesseven/VisaInstruments/Simplevisa/SimpleVisa/testVisa.py', wdir='D:/Dropbox (icts)/Dropbox (icts)/Github_Sixtimesseven/VisaInstruments/Simplevisa/SimpleVisa')
Resource Manager of Visa Library at C:\WINDOWS\system32\visa32.dll
GPIBInstrument at GPIB0::11::0::INSTR
Traceback (most recent call last):

  File "D:\Dropbox (icts)\Dropbox (icts)\Github_Sixtimesseven\VisaInstruments\Simplevisa\SimpleVisa\testVisa.py", line 16, in <module>
    hp.write('AB4')

  File "C:\Users\justrandom\anaconda3\envs\ba21\lib\site-packages\pyvisa\resources\messagebased.py", line 197, in write
    count = self.write_raw(message.encode(enco))

  File "C:\Users\justrandom\anaconda3\envs\ba21\lib\site-packages\pyvisa\resources\messagebased.py", line 157, in write_raw
    return self.visalib.write(self.session, message)[0]

  File "C:\Users\justrandom\anaconda3\envs\ba21\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2796, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))

  File "C:\Users\justrandom\anaconda3\envs\ba21\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore

  File "C:\Users\justrandom\anaconda3\envs\ba21\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)

VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

MatthieuDartiailh, 2021-12-02:
Since your are on Windows with NI VISA, can you test the communication with NI-MAX ?

sixtemesseven, 2021-12-02:
> Since your are on Windows with NI VISA, can you test the communication with NI-MAX ?

Well, the Instrument shows up on NI-Max, however, as far as I understand you cannot send anything from NI-Max?

[image]

MatthieuDartiailh, 2021-12-02:
You can using the interactive control button (with the device selected)

sixtemesseven, 2021-12-03:
> You can using the interactive control button (with the device selected)

Ok, I found the Visa Test Panel. This seems to work. Instruments resets and "remote" and "listen" led light up. If I press the LCL key the instruments returns to local controll, "remote" led is off and "listen" is still on.

[image]

I just checked if it might be working today (sometimes it does not work on a machine for days, but then it works on another machine usually).

```
import pyvisa as visa

rm = visa.ResourceManager()
print(rm)
hp = rm.open_resource('GPIB0::11::INSTR')
hp.read_termination = '\n'
hp.write_termination = '\n'
print(hp)
hp.write('IPR')
```

Still no good. The instrument does not reset, the "remote" light comes on, I get locked out and I get a timeout error.

MatthieuDartiailh, 2021-12-03:
I don't see why it would have worked before and would be failing now but there is one bug in how PyVISA normalize GPIB address by always inserting a secondary address of 0. I cannot seem to find the issue in which this is discussed right now but you could try to bypass the normalization logic at https://github.com/pyvisa/pyvisa/blob/main/pyvisa/resources/resource.py#L163 and see if it fixes your issue.

Another thing you can investigate is using NI IO Monitor compared the messages sent by PyVISA and NI Interactive control to understand why one work and not the other.

sixtemesseven, 2021-12-03:
This is the trace sent by my script:

[image]

If I send it directly.

[image]

sixtemesseven, 2021-12-03:
Interesting, it seems it really does use a secondary GPIB address. If I communicate with my Keithley 238, it works and it does not use a secondary address despite being the same adapter and the same bus!

[image]

MatthieuDartiailh, 2021-12-03:
I am not sure we understood one another.

 The issue you may have is that PyVISA always set the secondary address as 0 even though a secondary address of 0 and no secondary address are not fully equivalent (It took me some time to dig through the documentation). Which is why I suggest you manually edit PyVISA source to omit the secondary address to see if it is the root of the issue.

LongnoseRob, 2021-12-11:
@sixtemesseven could you please try with 
```
my_inst.read_termination = "\r\n"
my_inst.write_termination = "\r\n"
```
This is what I need to use for my HP instruments of similiar vintage as your HP3577.

sixtemesseven, 2021-12-13:
> @sixtemesseven could you please try with
> 
> ```
> my_inst.read_termination = "\r\n"
> my_inst.write_termination = "\r\n"
> ```
> 
> This is what I need to use for my HP instruments of similiar vintage as your HP3577.

Unfortunatly, this does not work. Same Problem.

sixtemesseven, 2021-12-13:
> I am not sure we understood one another.
>
> The issue you may have is that PyVISA always set the secondary address as 0 even though a secondary address of 0 and no secondary address are not fully [...]

Well, I'm not sure how exactly, but  I tried to fix the name in the resource.py file as follows: 

 ```
self._resource_name: str
        try:
            # Attempt to normalize the resource name. Can fail for aliases
            self._resource_name = str(rname.ResourceName.from_string(resource_name))
            self._resource_name = "GPIB0::11::INSTR"
        except rname.InvalidResourceName:
            self._resource_name = resource_name
```

interestingly: self._resource_name will be set to: 

`
GPIB0::11::0::INSTR
`

So I guess the problem is the secondary address.

Now it should be set to:

`"GPIB0::11::INSTR"`

However, the instrument still locks up as before.

sixtemesseven, 2021-12-14:
Well, hmm. I'm pretty sure I rebooted the instrument yesterday. Maybe I did not open a new spyder console... I did unplug and change the gpio adapter usb location (usb 2 to usb 3). Not sure why but now it seems to be working.

Test script:
```
import pyvisa as visa

rm = visa.ResourceManager()
print(rm)
hp = rm.open_resource('GPIB0::11::INSTR')
#hp.read_termination = "\r\n"
#hp.write_termination = "\r\n"
print(hp)
hp.write('IPR')
```

My fix in resource.py line 162
```
        self._resource_name: str
        try:
            # Attempt to normalize the resource name. Can fail for aliases
            self._resource_name = str(rname.ResourceName.from_string(resource_name))
            position = self._resource_name.find("::0")
            self._resource_name = self._resource_name[0: position:] + self._resource_name[(position+2) + 1::]
        except rname.InvalidResourceName:
            self._resource_name = resource_name
```

Not that great but it works...

MatthieuDartiailh, 2021-12-14:
Since you did a bunch of other changes can you confirm you need the resource name change ? I am asking since it used to work for you and that behavior is not exactly recent in PyVISA.

sixtemesseven, 2022-01-23:
> self._resource_name: str
>         try:
>             # Attempt to normalize the resource name. Can fail for aliases
>             self._resource_name = [...]

I just did a completely fresh install of a machine. And the first thing I did was test the issue above. I got the same problem, and succesfully applied the same fix.

For reference, the following is currently installed:
Pyvisa 1.11.3
NI-Visa 21.0
NI-488-2 19.0.0 
Python 3.9.7
Running on Win10, using Anaconda.

closed 2022-01-23 by sixtemesseven

MatthieuDartiailh, 2022-01-23:
Thanks for the confirmation. It is very nice to know that the resource name is truly in cause here. 

Ideally, a fix should be applied in the GPIB resource name class to omit the secondary address when it is irrelevant. Given how the code is currently working, it may not be exactly obvious how to fix it, but feel free to have a look and make a PR if you have time. Otherwise, I will try to get to it ASAP, but I would appreciate if you could test the patch.

MatthieuDartiailh, 2022-01-24:
In case you are interested in trying to fix the resource name, here are  some thoughts. 

Currently, the resource name base class assumed that a fix formatting string can be computed based on the class fields and then filled with either user values or default values. This assumption breaks for GPIB since the secondary address must not appear. A minimally disruptive way to handle things could be to make the canonical format an ordered dictionary with keys as field names and values as formats, and then only use it if the field value is not None. Changes would need to be made here https://github.com/pyvisa/pyvisa/blob/main/pyvisa/rname.py#L104, here https://github.com/pyvisa/pyvisa/blob/main/pyvisa/rname.py#L158 and there https://github.com/pyvisa/pyvisa/blob/main/pyvisa/rname.py#L318

Finally, the secondary address would be marked as Optional[str] with a default of None, with an explanation that could reference this issue.

sixtemesseven, 2022-01-30:
> Thanks for the confirmation. It is very nice to know that the resource name is truly in cause here.
>
> Ideally, a fix should be applied in the GPIB resource name class to omit the secondary address [...]

Sure, if I find some time first I will look into it.
Otherwise I would be happy to test your patch.