pyvisa/pyvisa #641: Memory error with write_binary_values() on 2e9 int16 sized array

jonnojohnson, 2021-12-08:
I'm able to read a 2e9 sized data block from an instrument okay using query_binary_values(datatype='h', is_big_endian = True, container = np.array, data_points = 2e9) with no problems.
However on the same machine I'm getting a memory error using write_binary_values() when util.py gets to:
block = struct.pack(fullfmt, *iterable)

Below is the output of python -m visa info:

Machine Details:
   Platform ID:    Windows-10-10.0.19042-SP0
   Processor:      Intel64 Family 6 Model 165 Stepping 2, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\marnj\Anaconda3\python.exe
   Version:        3.8.12
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Oct 12 2021 03:01:40 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: Keysight Technologies
         Spec. Version: Keysight Technologies
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: Keysight Technologies
         Spec. Version: Keysight Technologies

MatthieuDartiailh, 2021-12-09:
Can you post the full traceback ? Also if the data is already in a numpy array can you try to replicate the logic but using numpy to perform the packing rather than struct. It may be interesting to special numpy arrays when writing just like we do when reading.

jonnojohnson, 2021-12-09:
This is the full traceback:
```Traceback (most recent call last):

  File "C:\Users\uname\load_wfm_pyvisa.py", line 134, in <module>
    scope.write_binary_values(f":WAVeform:DATA {x_origin},{x_increment},{y_origin},{y_increment},",data, datatype='h', is_big_endian = True)

  File "C:\Users\uname\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 308, in write_binary_values
    block = util.to_ieee_block(values, datatype, is_big_endian)

  File "C:\Users\uname\Anaconda3\lib\site-packages\pyvisa\util.py", line 775, in to_ieee_block
    return to_binary_block(iterable, header, datatype, is_big_endian)

  File "C:\Users\uname\Anaconda3\lib\site-packages\pyvisa\util.py", line 741, in to_binary_block
    block = struct.pack(fullfmt, *iterable)

MemoryError
```
I will try to look into using numpy.

LongnoseRob, 2021-12-09:
The struct.pack and unpack functions seem to be known for MemoryErrors on big data-sets, at leat this is that my quick search during the lunchbreak showed.
Numpy might be helpful here, but It might be also worth looking into the options for the transfer, e.g. a packed stream vs. uncompressed raw bytestream..

Could you tell us which Maker/Type of Instrument you are working with?

Here is an idea, that might be not 100% the solution you would like to achive, but it could be a work-around:
As you are propably connecting via TCPIP to the instrument, would it be an option to transfer the data from the PC to the Instruments harddisk (or orther way around) over the network and just issue the load from disk /write to disk command via pyVisa?
I could imagine that transferring some Gigabytes of sampled data via VISA conection might be actually slower then copying it over the Samba/WindowsNetwork Share.

jonnojohnson, 2021-12-10:
@LongnoseRob appreciate the suggestions and I realize there may be other ways to achieve a result but in this case I'm trying to see if it's possible to write the full 2Gpoints to the instrument ideally using Pyvisa.

I guess I will need to write to the instrument in multiple commands if I use numpy. How do I write to the instrument in pyvisa without a termination character?

MatthieuDartiailh, 2021-12-10:
No my suggestion would be to use the `tobytes` method of the array to build the message to send in place of struct. You would only modify the way the message is prepared but send the exact same message as if struct worked. You may however need to play with the endianess of the dtype to match your instrument expectations, this might help https://numpy.org/doc/stable/user/basics.byteswapping.html

jonnojohnson, 2021-12-10:
@MatthieuDartiailh Yes I had found the tobytes method and fortunately it worked with defaults with the endianness I was using. Haven't tested it with 2e9 values yet.
Now I see that using write_binary_values uses write_raw so I just need to convert the string messages I need to binary, combine everything and use write_raw. Will update. Guess I should file a ticket for the struct.pack issue.

MatthieuDartiailh, 2021-12-10:
Having an issue tracking the use of numpy routines in `to_binary_block` would be nice. 

For the rest it should a matter of calling encode and adding the 2 byte strings.

jonnojohnson, 2021-12-10:
Ok I got it working with `numpy.tobytes` (my data is int16, big-endian so the `tobytes` defaults worked) by essentially copying the code in `write_binary_data`.
It worked great for smaller sized arrays (1e8 size was 8x faster than struct.pack). However I got a different error with a 2e9 array:
```Traceback (most recent call last):

  File "C:\Users\uname\load_wfm_pyvisa.py", line 141, in <module>
    scope.write_raw(msg)

  File "C:\Users\uname\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 157, in write_raw
    return self.visalib.write(self.session, message)[0]

  File "C:\Users\uname\Anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2796, in write
    ret = library.viWrite(session, data, len(data), byref(return_count))

  File "C:\Users\uname\Anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore

  File "C:\Users\marnj\Anaconda3\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)

VisaIOError: VI_ERROR_USER_BUF (-1073807247): A specified user buffer is not valid or cannot be accessed for the required size.```

MatthieuDartiailh, 2021-12-10:
I have no upfront idea why it is so. Since you said earlier that NI-MAX worked could you capture the data as sent by NI-IO Trace to see if NIMax does anything fancy behind the scenes. You can also try to split the message in two (using write_raw who can omit the termination char). Since the IEEE block specify length it may just work.

MatthieuDartiailh, 2022-02-01:
ping @jonnojohnson

jonnojohnson, 2022-02-01:
@MatthieuDartiailh I'm using Keysight IO Libraries not NI-MAX.
This code works well up to a certain size:
```
with open(f"{src_dir}/block_data.npy", "rb") as block_data_file:
    data = np.load(block_data_file)
scope.write(":WAVeform:STReaming ON")

data_length = len(data)
if waveform_format == "BYTE":
    element_length = 1
else:   # waveform_format == "WORD":
    element_length = 2
header = "%d" % (data_length * element_length)
header = "#%d%s" % (len(header), header)
header = bytes(header, "ascii")
block = data.tobytes()
block = header + block
message = f":WAVeform:DATA {x_origin},{x_increment},{y_origin},{y_increment},"
msg = message.encode('ascii') + block
scope.write_raw(msg)
```
I used the IO Libraries IO Monitor feature to capture the data and this is the ASCII interpretation of the viWrite command:
```
:WAVeform:DATA -0.0039062500019255,3.90625e-12,0.00213894,6.68419e-05,#104000000000�.����.���������V._..�.�..{...����+.�..�@�Q.B..�.��.�.d���r....�W����.�.�.'.�.��4���..�.�.@..�E�.���t�.�7...��u�(���d�M��.u.i�P�~...����.�.�D.q...�.��...�.�u�....�.�...n�t�.
```

MatthieuDartiailh, 2022-02-01:
Thanks. Out of curiosity did you capture the traffic when sending a message that would crash pyvisa but work with Keysigh Interactive control ?

jonnojohnson, 2022-02-03:
@MatthieuDartiailh I'm not sure what case you're referring to. The memory error?

MatthieuDartiailh, 2022-02-03:
Sorry apparently I confused this issue with a different one in which the user could do something with NI MAX but the same thing failed in PyVISA.

Since your issue is now related to the buffer size you can try to play with viSetBuf (https://documentation.help/NI-VISA/viSetBuf.html). I do not remember off the the top of my head if it is exposed at the resource level, you may have to go through low level library calls. Let me know if you need help to figure it out and if this helps.

Also it would be great if you could make a short PR with the changes to write_binary you made. Even if it is minimal it will be a nice starting point to add the feature.

jonnojohnson, 2022-02-04:
@MatthieuDartiailh I will try to create a PR. It will be my first!

Re: viSetBuf I've found the set_buffer() method for the Resource Manager visalib.
https://pyvisa.readthedocs.io/en/latest/api/visalibrarybase.html?highlight=set_buffer#pyvisa.highlevel.VisaLibraryBase.set_buffer
But I can't figure out how to set the mask BufferType:
https://pyvisa.readthedocs.io/en/latest/_modules/pyvisa/constants.html?highlight=buffertype#

jonnojohnson, 2022-02-05:
Tried this:
```
lib = rm.visalib
write_mask = pyvisa.constants.BufferType.write
lib.set_buffer(1,write_mask,2000000000)
*** pyvisa.errors.VisaIOError: VI_ERROR_ALLOC (-1073807300): Insufficient system resources to perform necessary memory allocation.
```

MatthieuDartiailh, 2022-02-05:
That looks correct but you should pass `instr.session` instead of 1 as first argument also you may use the `io_out` buffer since I am not sure you actually use buffer write.

Finally if that still fail did you try splitting the message in multiple write as we evoked earlier ?

jonnojohnson, 2022-02-06:
@MatthieuDartiailh same result using `io_out` and `instr.session`
I won't have time to try anything else or do the PR for the next week.

MatthieuDartiailh, 2022-02-07:
No problem