pyvisa/pyvisa #642: [COM] Communication issue with XXX using XXX

Atmosphire, 2021-12-09:
Hello, I'm trying to connect an electronic load (TTI LD400P) to a raspberry PI, unfortunately I can't do it. I tried to connect from my pc I manage to have access to the web server created by the power supply but nothing on the raspberry pi. The IP address is given by the DHCP but I cannot connect to it which is strange because I can have access to the DHCP but not to the rest ! HELP ME PLZ

LongnoseRob, 2021-12-09:
Hello @Atmosphire.

I dont know about the setup you are using with this load, PC and the RasPi, but if all are on the same network, you should be able to ping it from the PC and the RasPi.
Have you tried this? 
For TCP/IP communcation with VISA you do not need the internal webserver of the load.

It could also be the that the instrument only allows one active connection per time, so that if the Webinterface is accessed by your PC the other connection could be blocked.

Here are some more points of information you could help up with diagnosing this problem:
1. please give us the output of `pyvisa-info` run in a shell on your RasPi
2. show us your code of the pyhton program you are tyring to connect to the load with.
3. share a link to the manual of this load with us (if freely available on-line)

Atmosphire, 2021-12-09:
I'am back ! 

It's good I solved my problem my DHCP server gave me an address 169.X.X.X, from what I understood the addresses 169 are addresses with an error, my server rejected the electronic load. However, so i gave hime a static IP but I have one more questions.

`import vxi11
instr = vxi11.Instrument(172.20.42.203)
print(instr.ask("*IDN?"))
print(instr.ask("*MODE?"))`

Gave me the same thing, HOW ?

LongnoseRob, 2021-12-09:
Could you please post the traceback of your python command?
Are you sure about the IP-adress of the instrument?
this 172.x.x.x adress seems to be an adress which is probably not in your local network (168.x.x.x)

Atmosphire, 2021-12-09:
Instead of using an address given by the DHCP I give it a static address

[Traceback](
Instead of using an address given by the DHCP I give it a static address)