pyvisa/pyvisa #643: Is there anyone know that  how to plot real time data with pyvisa, matplotlib or other library so that we can know some measurement error at the beginning.

GitQer, 2021-12-15:
I see some method on internet about real time plotting, but did anyone tried real plotting with instrument? And will it have some time_related impact on measurement?

Satertek, 2021-12-16:
PyVISA would just be once piece of your solution. You would utilize it to send and receive data from your instrument, but storing and viewing that data would require additional packages.

[PyMeasure](https://github.com/pymeasure/pymeasure)  is one project I know of that aims to combine all of that functionality that may be useful to take a look at. (it utilizes PyVISA and pyqtgraph)

GitQer, 2021-12-16:
Thank you @Satertek. Are you familiar with pymeasure? After reading the document, I have some confusion about Pymeasure in their adding instrument section. If I want to use it by SCPI command, I can just use the self. ask( )and self. write() and put the command in the bracket like what I do using pyvisa? And the self. control means that we write part of SCPI command so that we can just insert value when we use the API created by self control?