pyvisa/pyvisa #644: [COM] Communication issue with KEYSIGHT34465A using GPIB

janetmayle, 2021-12-17:
Instrument details
------------------
* Model: KEYSIGHT 34465A（a digit multimeter）
* Communication: GPIB
* Link to the documentation (if available):

My Question
------------------
Thank you for your valuable time.

When I execute the following code，the output is `()`，I don't know why my instrument is not listed......

```
import pyvisa

if __name__ == '__main__':
    rm = pyvisa.ResourceManager('@ivi')
    print(rm.list_resources())
```

Output of `pyvisa-info`
-----------------------
```
Machine Details:
   Platform ID:    Linux-5.10.0-5-generic-x86_64-with-glibc2.31
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/sugar/Work/Software/anaconda3/bin/python
   Version:        3.9.7
   Compiler:       GCC 7.5.0
   Bits:           64bit
   Build:          Sep 16 2021 13:09:58 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.21.0.0:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments

```

LongnoseRob, 2021-12-17:
For GPIB on Linux you will need to install linux-gpib and ctypes-gpib.

a few questions to be able to guide you wiht more details:
Which GPIB-Interface-adapter are you using?
Which distribution are you using?

MatthieuDartiailh, 2021-12-23:
I remember cases where GPIB instruments were not listed properly by NI on Linux but could be accessed nonetheless.

Did you try to connect directly ?

janetmayle, 2021-12-24:
> For GPIB on Linux you will need to install linux-gpib and ctypes-gpib.
> 
> a few questions to be able to guide you wiht more details: Which GPIB-Interface-adapter are you using? Which distribution are you using?

After many attempts, things have made some progress, but the problem is still not solved. The current situation is as follows.

I have installed linux-gpib and gpib-ctypes. 

If I use python-py as backend, I can use GPIB-USB-HS+ to communicate with KEYSIGHT multimeter successfully. The python code and output are as follows:
```python
import pyvisa

if __name__ == '__main__':
    rm = pyvisa.ResourceManager('@py')
    # rm = pyvisa.ResourceManager('@ivi')
    print(rm)
    print(rm.list_resources())
    inst = rm.open_resource('GPIB0::22::INSTR')
    print(inst.query("*IDN?"))

```

```bash
/home/sugar/Work/Software/anaconda3/bin/python /home/sugar/Work/Project/Python/PyVISA-Test/main.py
Resource Manager of Visa Library at py
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
libgpib: invalid descriptor
[... 60 lines cut ...]
but it failed:
```bash
/home/sugar/Work/Software/anaconda3/bin/python /home/sugar/Work/Project/Python/PyVISA-Test/main.py
Resource Manager of Visa Library at /usr/lib/x86_64-linux-gnu/libvisa.so.21.0.0
('ASRL1::INSTR', 'ASRL2::INSTR', 'ASRL3::INSTR', 'ASRL4::INSTR', 'ASRL5::INSTR', 'ASRL6::INSTR', 'ASRL7::INSTR', 'ASRL8::INSTR', 'ASRL9::INSTR', 'ASRL10::INSTR', 'ASRL11::INSTR', 'ASRL12::INSTR', 'ASRL13::INSTR', 'ASRL14::INSTR', 'ASRL15::INSTR', 'ASRL16::INSTR', 'ASRL17::INSTR', 'ASRL18::INSTR', 'ASRL19::INSTR', 'ASRL20::INSTR', 'ASRL21::INSTR', 'ASRL22::INSTR', 'ASRL23::INSTR', 'ASRL24::INSTR', 'ASRL25::INSTR', 'ASRL26::INSTR', 'ASRL27::INSTR', 'ASRL28::INSTR', 'ASRL29::INSTR', 'ASRL30::INSTR', 'ASRL31::INSTR', 'ASRL32::INSTR')
Traceback (most recent call last):
  File "/home/sugar/Work/Project/Python/PyVISA-Test/main.py", line 8, in <module>
    inst = rm.open_resource('GPIB0::22::INSTR')
  File "/home/sugar/Work/Software/anaconda3/lib/python3.9/site-packages/pyvisa/highlevel.py", line 3271, in open_resource
    info = self.resource_info(resource_name, extended=True)
  File "/home/sugar/Work/Software/anaconda3/lib/python3.9/site-packages/pyvisa/highlevel.py", line 3197, in resource_info
    ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
  File "/home/sugar/Work/Software/anaconda3/lib/python3.9/site-packages/pyvisa/ctwrapper/functions.py", line 2077, in parse_resource_extended
    ret = library.viParseRsrcEx(
  File "/home/sugar/Work/Software/anaconda3/lib/python3.9/site-packages/pyvisa/ctwrapper/highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "/home/sugar/Work/Software/anaconda3/lib/python3.9/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_INTF_NUM_NCONFIG (-1073807195): The interface type is valid but the specified interface number is not configured.

Process finished with exit code 1

```

Please give me some advice. I really don't know how to do it. Thank you very much.

janetmayle, 2021-12-24:
### output of `pyvisa-info`

```
(base) sugar@suagr:~$ pyvisa-info
Machine Details:
   Platform ID:    Linux-5.10.0-5-generic-x86_64-with-glibc2.31
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /home/sugar/Work/Software/anaconda3/bin/python
   Version:        3.9.7
   Compiler:       GCC 7.5.0
   Bits:           64bit
   Build:          Sep 16 2021 13:09:58 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.21.0.0:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
   py:
      Version: 0.5.2
      ASRL INSTR:
         Please install PySerial (>=3.0) to use this resource type.
         No module named 'serial'
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR: Available via Linux GPIB (b'4.3.4')
      GPIB INTFC: Available via Linux GPIB (b'4.3.4')

```

MatthieuDartiailh, 2021-12-24:
According to NI website it appears USB-GPIB os not supported on linux (and never was). So it appears you are stuck with linux-gpib. 

Reference https://www.ni.com/en-us/support/documentation/supplemental/06/ni-488-2-supported-versions-for-gpib-devices-and-modules.html#section--1752621460

closed 2021-12-31 by janetmayle