pyvisa/pyvisa #645: 'IVIVisaLibrary' object has no attribute 'viParseRsrcEx'

asebian, 2021-12-28:
Hi,
I am a newbie owning a GwInstek MSO-2104EA running a socket server listening at port 3000.
Trying to connect from Ubuntu 21.10  I get the error message below, but the same script works just fine from Windows.
Is this an Ubuntu issue?

Thanks for sharing!

To Reproduce
--------------
```
from pyvisa import ResourceManager

rm = ResourceManager()
instr = rm.open_resource('TCPIP0::192.168.8.107::3000::SOCKET')
instr.read_termination = '\n'
instr.write_termination = '\n'
print(instr.query('*IDN?'))
```
**Error and Traceback**
```
AttributeError                            Traceback (most recent call last)
<ipython-input-1-a50b1738897f> in <module>
      2 
      3 rm = ResourceManager()
----> 4 instr = rm.open_resource('TCPIP0::192.168.8.107::3000::SOCKET')
      5 instr.read_termination = '\n'
      6 instr.write_termination = '\n'

/usr/lib/python3/dist-packages/pyvisa/highlevel.py in open_resource(self, resource_name, access_mode, open_timeout, resource_pyclass, **kwargs)
   3269 
   3270         if resource_pyclass is None:
-> 3271             info = self.resource_info(resource_name, extended=True)
   3272 
   3273             try:

/usr/lib/python3/dist-packages/pyvisa/highlevel.py in resource_info(self, resource_name, extended)
   3195 
   3196         if extended:
-> 3197             ret, err = self.visalib.parse_resource_extended(self.session, resource_name)
   3198         else:
   3199             ret, err = self.visalib.parse_resource(self.session, resource_name)

/usr/lib/python3/dist-packages/pyvisa/ctwrapper/functions.py in parse_resource_extended(library, session, resource_name)
   2075     # [ViSession, ViRsrc, ViPUInt16, ViPUInt16, ViAChar, ViAChar, ViAChar]
   2076     # ViRsrc converts from (str, unicode, bytes) to bytes
-> 2077     ret = library.viParseRsrcEx(
   2078         session,
   2079         resource_name,

AttributeError: 'IVIVisaLibrary' object has no attribute 'viParseRsrcEx'

```
**Output of `pyvisa-info`**
```
Machine Details:
   Platform ID:    Linux-5.13.0-22-generic-x86_64-with-glibc2.34
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.9.7
   Compiler:       GCC 11.2.0
   Bits:           64bit
   Build:          Sep 10 2021 14:59:43 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.0.0.0:
         found by: auto
         bitness: 64
         Could not get more info:
            VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
   py:
      Version: 0.5.1
      ASRL INSTR: Available via PySerial (3.5b0)
      USB INSTR: Available via PyUSB (1.0.2). Backend: libusb1
      USB RAW: Available via PyUSB (1.0.2). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```
**Output `pip show <module>`**
```
$ pip show pyvisa
Name: PyVISA
Version: 1.11.3
Summary: Python VISA bindings for GPIB, RS232, TCPIP and USB instruments
Home-page: https://github.com/pyvisa/pyvisa
Author: Torsten Bronger, Gregor Thalhammer
Author-email: [snip]
License: MIT License
Location: /usr/lib/python3/dist-packages
Requires: 
Required-by: 
```
```
$ pip show pyvisa-py
Name: PyVISA-py
Version: 0.5.1
Summary: Python VISA bindings for GPIB, RS232, and USB instruments
Home-page: https://github.com/pyvisa/pyvisa-py
Author: Hernan E. Grecco
Author-email: [snip]
License: MIT License
Location: /usr/lib/python3/dist-packages
Requires: 
Required-by: 
```

MatthieuDartiailh, 2021-12-28:
This is a known issue with LibreVisa but I do not know if it was ever addressed. See https://github.com/pyvisa/pyvisa/issues/261

If you only need TCPIP resources you can use pyvisa-py by creatig the ResourceManager with "@py" as argument.

asebian, 2021-12-28:
Thanks for your quick reply, Matthieu. 
Cheers!

```
from pyvisa import ResourceManager

try:
    rm = ResourceManager('@py') # use 'pyvisa-py' backend
except ValueError:
    rm = ResourceManager()
instr = rm.open_resource('TCPIP0::192.168.8.107::3000::SOCKET')
instr.read_termination = '\n'
instr.write_termination = '\n'
print(instr.query('*IDN?'))
```
GW,MSO-2104EA,GER110150,V1.42

closed 2021-12-28 by asebian