pyvisa/pyvisa #649: If condition doesn't output proper data if if trigger:state is providing proper scope state

PoojaLangaliya, 2022-01-19:
<Even if scope is triggering and state is printing as 'trigger', python loop does't enter into correct state. The screen attached is showing the scope trigger state and the output of the execution file. Issue is when I print state, it provide output as 'trigger'. Right after that If condition is check that if scope triggered, it should print 'scope triggered' but instead it is going into else loop. I am not sure what is the issue. Same thing is happening with READY state.>

`#-------------------------------------------------------------------------------
# Name:        GetScopeInfo
# Author:      bbattye
#
# Created:     October 02, 2018

#-------------------------------------------------------------------------------
import pyvisa
import pyvisa as visa
import time
import sys
import timeit
import numpy as np

from timeit import default_timer as timer

rm = pyvisa.ResourceManager()
rm.list_resources()

scope = rm.open_resource("TCPIP0::10.232.243.39::INSTR")
scope.write("*RST")
scope.query("*OPC?")
scope.timeout = 9000
string = scope.query("*IDN?")
print ("Scope Model , Serial #, FW Version")
print (string)
scope.query("*OPC?")
scope.write("RECALL:SETUP 'C:\\Users\\Tek_Local_Admin\\Tektronix\\TekScope\\Setups\\TBP_PKP_CONFIG_Edge Trigger.set'")
scope.query("*OPC?")
scope.write(":TRIGger:A:MODe NORMAL")
scope.query("*OPC?")
state = scope.query("TRIGger:STATE?")
print(state)
if state == 'trigger':
    print("scope triggered")
    exit
else:
    print('scope not triggered')
    exit
print(state)

scope.close()

rm.close()

exit

`

To Reproduce
--------------
Steps to reproduce the behavior:
1. Scope Series: TEKTRONIX,MSO64,C014830,CF:91.1CT FV:1.26.5.7750
2. run the script above, make sure scope is running and in triggering state
3. check the execution output
I am not sure if the issue is with scope or the code I am using

MatthieuDartiailh, 2022-01-19:
I do not have access to such an instrument, could you provide the output of the script ?

PoojaLangaliya, 2022-01-19:
[image]
@MatthieuDartiailh  thanks for your response. I tried all different way to make if statement work but it seems to have issue in following correct if statement.

Let me know if any additional details are required

PoojaLangaliya, 2022-01-20:
I found the solution. 

One of the scope online manual it mentioned: "In this situation, first makes sure you can actually
read from the instrument by reading the answer using the read_raw function (you may need to call it multiple time),
and check that the advertized length of the block match what you get from your instrument (plus the header)."

The issue was scope was only providing data when asked using scope.query but code was not getting reading from the scope.
To get the reading from scope read_raw() needs to be used.
As a result code was getting trigger value assigned to state. Refer below code for solution.

#-------------------------------------------------------------------------------
import pyvisa
import pyvisa as visa
import time
import sys
import timeit
import numpy as np

from timeit import default_timer as timer

rm = pyvisa.ResourceManager()
rm.list_resources()

scope = rm.open_resource("TCPIP0::10.232.243.39::INSTR")
scope.write("*RST")
scope.query("*OPC?")
scope.timeout = 9000
string = scope.query("*IDN?")
print ("Scope Model , Serial #, FW Version")
print (string)
scope.query("*OPC?")
scope.write("RECALL:SETUP 'C:\Users\Tek_Local_Admin\Tektronix\TekScope\Setups\TBP_PKP_CONFIG_Edge Trigger.set'")
scope.query("*OPC?")
scope.write(":TRIGger:A:MODe NORMAL")
scope.query("*OPC?")
state = scope.read_raw(scope.write("TRIGger:STATE?")) #this is where the issue was
time.sleep(1)
print(state)
if state == b'TRIGGER\n': #scope returns value in this format
print("scope triggered")
exit
else:
print('scope not triggered')
exit
print(state)

scope.close()

rm.close()

exit

`
[image]

closed 2022-01-20 by PoojaLangaliya