pyvisa/pyvisa #654: [COM] Occasional VISA timeout when querying Keysight Power Supply using TCPIP (LAN)

oswinrodrigues, 2022-02-16:
Instrument details
------------------
* Model: [Keysight E36312A](https://www.keysight.com/ca/en/products/dc-power-supplies/bench-power-supplies/e36300-series-triple-output-power-supply-80-160w.html)
* Communication: TCPIP (LAN)
* Link to the documentation: [programming guide](https://www.keysight.com/ca/en/assets/9018-04577/programming-guides/9018-04577.pdf)

Output of `pyvisa-info`
-----------------------

<details><summary>Click to expand</summary>

```
Machine Details:
   Platform ID:    Linux-5.14.14-nuvation-sdi-x86_64-with-glibc2.29
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python3
   Version:        3.8.10
   Compiler:       GCC 9.3.0
   Bits:           64bit
   Build:          Nov 26 2021 20:14:08 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      Binary library: Not found
   py:
      Version: 0.5.2
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb1
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb1
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'
```

</details>

Instrument setup
------------------
<details><summary>Click to expand</summary>

```python
def connect(self, ip, socket):
        if socket is None:
            TCP_INST = "TCPIP0::" + ip + "::" + "inst0" + "::INSTR"
        else:
            TCP_INST = "TCPIP::" + ip + "::" + socket + "::SOCKET"
        rm = pyvisa.ResourceManager('@py')
        self.instrument = rm.open_resource(TCP_INST, open_timeout=5000)
        self.instrument.timeout = 10000
        self.instrument.query_delay = 0
        self.instrument.write_termination = "\n"
        self.instrument.read_termination = "\n"
```

</details>

Error Trace
-----------
<details><summary>Click to expand</summary>

```
    res = self.instrument.query(msg)
  File "/usr/local/lib/python3.8/dist-packages/pyvisa/resources/messagebased.py", line 644, in query
    return self.read()
  File "/usr/local/lib/python3.8/dist-packages/pyvisa/resources/messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "/usr/local/lib/python3.8/dist-packages/pyvisa/resources/messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/usr/local/lib/python3.8/dist-packages/pyvisa_py/highlevel.py", line 519, in read
    return data, self.handle_return_value(session, status_code)
  File "/usr/local/lib/python3.8/dist-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
```

</details>

Context
--------
- I run this code in an automated test sytem. 
- This happens only about 5-10% of the time.
- When it does happen, it's only when querying.
- Happens in a place where there are 3 back-to-back queries.

MatthieuDartiailh, 2022-02-16:
Does this happen using the SOCKET or INSTR resources type ?

MatthieuDartiailh, 2022-02-16:
So it is transient and occurs with the simplest transport layer when you "stress" the connection. Can you try adding a tiny sleep between your queries ? 1 ms or so.

It may be on the instrument the socket resource is really simple. 

Alternatively can you install a VISA distribution and see if you see also see the issue with the ici backend ? Keysight can be installed on Linux without too much hassle.

oswinrodrigues, 2022-02-16:
Will try that and update here accordingly. May take some time to confirm that the problem has truly gone away, since it didn't show up consistently in the first place.

Good point about the transience, I've edited the original post to explicitly clarify that. The problem would self-recover i.e. I didn't have to manually reset the Keysight. After some time (in the same script) future communications with the instrument (including queries) would work as expected.

oswinrodrigues, 2022-02-19:
> Can you try adding a tiny sleep between your queries ?

Instead of manually sleeping in between, would it also work to use `self.instrument.query_delay` (from my instrument setup code above) or to specify a `delay` paramter when invoking the `query()` method?

MatthieuDartiailh, 2022-02-19:
You can try that yes. On a raw socket I do not really expect issues when writing and reading back to back, but it will delay consecutive writes too so it may help yes.

oswinrodrigues, 2022-02-28:
This looks to have done the trick. Will keep monitoring, but shall close this issue for now. Thanks for the assistance!

MatthieuDartiailh, 2022-03-01:
Happy to help.

oswinrodrigues, 2022-12-07:
Re-opening this as it seems to have crept up again. Here are the changes I made as discussed above:
- Set up `instrument` with a `query_delay` of 0.1 seconds. This delay happens in between the write and read that make up a single query.
- Sleep for 0.1 seconds before each attempted query

MatthieuDartiailh, 2022-12-13:
Could somebody perform network wide scan (looking for instruments for example) ? Since your instrument seems rather touchy it may affect it. Is testing on local network an option ?

oswinrodrigues, 2023-01-14:
Hi, sorry for the late reply here. It turned out that in fact someone else was unknowingly controlling the instrument at the same time. By putting the device on a local network, these problems went away. Thank you!

closed 2023-01-14 by oswinrodrigues