pyvisa/pyvisa #655: Macbook , They are all 64bits, but tips 'no matching architecture in universal wrapper'

tangtangtang11, 2022-02-17:
Machine Details:
   Platform ID:    macOS-11.2.3-arm64-arm-64bit
   Processor:      arm

Python:
   Implementation: CPython
   Executable:     /opt/homebrew/opt/python@3.9/bin/python3.9
   Version:        3.9.9
   Compiler:       Clang 13.0.0 (clang-1300.0.29.3)
   Bits:           64bit
   Build:          Nov 21 2021 03:16:13 (#main)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: /Library/Frameworks/visa.framework/visa:
         found by: auto
         bitness: 64
         Could not get more info: Interpreter and library have different bitness.

MatthieuDartiailh, 2022-02-17:
I do not own a Mac M1 but my best guess is that your python is built for arm but that NI does not provide a native arm build.

You can try installing x86-64 python and see if it helps.

We should try to get more details than just the bitness now that Apple can have that kind of mismatch, too. If you figure out a way to get the full architecture for both python and the dylib feel free to make a PR.

closed 2023-03-28 by MatthieuDartiailh

akira-okumura, 2023-04-27:
PyVISA + NI-VISA works on M1/M2 Macs if x86_64 Python is used.

PR #733 (2023-03-28): Use architecture instead of `bitness` in `LibraryPath` and friends
- Replaced `bitness` of libraries and python interpreter
- Removed unused `LibraryPath.is32bit`, `LibraryPath.is_64bit`, and `LibraryPath.bitness` methods
- Introduced AARCH64 as tested-for architecture
- Slightly updated detection logic
- Removed IA64 and linux universal binary test scenarios since they are unrealistic
- Removed `linux2/3` checking since it should not exist past Python 3.3

I am not entirely happy with current enum-based approach, but I still think enums are more appropriate in this case than magic dicts. Mayb [... cut]
commit 404ef1f1 (2023-03-28): Use architecture instead of `bitness` in `LibraryPath` and friends

Fixes #655

- Replaced `bitness` of libraries and python interpreter
- Removed unused `LibraryPath.is32bit`, `LibraryPath.is_64bit`, and
  `LibraryPath.bitness` methods
- Introduced AARCH64 as tested-for architecture
- Slightly updated detection logic
- Removed IA64 and linux universal binary test scenarios since they are
  unrealistic
- Removed `linux2/3` checking since it should not exist past Python 3.3