pyvisa/pyvisa #656: Unable to use lock with device

gadese, 2022-02-28:
Hi, I am having issues creating using locks with a device. I can connect to my device (an Agilent power supply, if it makes any difference) without issue, and send/read data correctly. But the locks do not work for me.

Version:
Python 3.8
Pyvisa 1.11.3
Pyvisa-py 0.5.2

My code:
--------------
```
import pyvisa
rm = pyvisa.ResourceManager()
device = rm.open_resource('ASRL/dev/ttyUSB2::INSTR', baud_rate=9600, data_bits=8, parity=0, stop_bits=10)

device.lock()
```
As a note, I have also tried the following and still get an error:
setting ```access_mode=pyvisa.constants.AccessModes.shared_lock``` when opening the resource
device.lock_state
```
with device.lock_context():
        ...
```

**Calling device.lock() gives the following error message**
```
File ".../pyvisa/resources/resources.py", line 599, in lock
    return self.visalib.lock(
File ".../pyvisa_py/highlevel.py", line 706, in lock
    return key, self.handle_return_value(session, status code)
File ".../pyvisa/highlevel.py", line 251m in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_NSUP_OPER (-1073807257): The given session or object reference does not support this operation.
```

**In fact, simply trying to access the device.lock_state attribute gives the following error, which seems to hint an issue with the lock attribute itself**
```
File ".../pyvisa/attributes.py" line 176, in __get__
    return self.post_get(instance.get_visa_attribute(self.attribute_id))
File ".../pyvisa/resources/resource.py", line 370, in get_visa_attribute
    return self.visalib.get_attribute(self.session, name)[0]
File".../pyvisa_py/highlevel.py", line 630, in get_attribute
    return state, self.handle_return_value(session, status_code)
File ".../pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError:VI_ERROR_NSUP_ATTR (-1073807331): The specified attribute is not defined or supported by the referenced object.
```

MatthieuDartiailh, 2022-03-01:
PyVISA-py does not support locking VISA resources. What is your use case for wanting to lock the resource ?

gadese, 2022-03-01:
> PyVISA-py does not support locking VISA resources. What is your use case for wanting to lock the resource ?

I'm confused, isn't it implemented in here?https://github.com/pyvisa/pyvisa/blob/53a1d410f727e8d16212ceae3229e6219443f9aa/pyvisa/resources/resource.py#L573

I have multiple processes that can potentially interact with a single instance of a ressource, so I was thinking of using a lock to make sure only one process can read/write to it at a time. I've looked into threading.Lock but I'm not sure it's what I'm looking for

gadese, 2022-03-01:
While I'm at it, I've been randomly getting an issue when reading data from the device, where an error is raised saying the query was interrupted because the output buffer still contained data from a previous command.

This didn't happen on every run, and seemed to be caused by the system not having enough time to read all the data in the buffer before sending another command, and the problem disappeared if I added large (1 second) wait times between each command. Is there a way to make sure the full output buffer is read before continuing to execute the code?

gadese, 2022-03-01:
> Regarding your second point, I have seen this kind of error when multiple threads (processes) access the same resource and one thread send a command which will generate a response but another one send a command before the response was read. Does this fit the situation in which you are seeing the error ?

I'll have to do a bit more trouble-shooting, but it's possible that's the issue. However, I tried a threading.Lock (I'm running multiple threads) and still got the error, so I'm not sure. I assumed the lock would be enough to avoid sending multiple commands.

> PyVISA is a front-end for multiple backends. In your case, you are using the pure Python backend implemented in PyVISA-py. The built-in `@ivi` backend requires a VISA library implementation to be installed (from National instruments, Keysight, ...).

That's my bad, I didn't realize I wasn't using the right backend. Thanks!

MatthieuDartiailh, 2022-03-01:
To avoid the query interrupted error, you need to acquire the lock, write then read and finally release the lock.

gadese, 2022-03-02:
Yes, this is what I'm doing but still seeing issues

I'll close the thread because this is a somewhat unrelated discussion and the original issue is solved.

closed 2022-03-02 by gadese

MatthieuDartiailh, 2022-03-02:
Can you post a minimal script illustrating you workflow ?

gadese, 2022-03-03:
Hi Matthieu, I'm actually fairly certain the multithreading isn't causing my query issues. I reformatted my code to make sure only one thread makes write/read calls to the device, and I'm still getting the timeout error. Even with the simple code below, the error pops up:

MatthieuDartiailh, 2022-03-03:
Thanks for sharing. I'm not sure we can do anything; it looks like your instrument is just slow. We have a way to specify a delay between the write and read in a query but it looks like it would not really help here.

gadese, 2022-03-03:
Alright, thanks for the quick responses