pyvisa/pyvisa #657: mypy cannot figure out query()

abunimeh, 2022-03-06:
I cannot get mypy to understand pyvisa 

given the class method below

```
# snip
self.instr = self.pyvisa_rm.open_resource(address)
# snip
@property
def id(self) -> str: 
    return self.instr.query("*IDN?")
```

`mypy` mypy 0.930 gives this

```
Returning Any from function declared to return "str"mypy(error)
```

It seems that query() is not recognized at all
```
"Resource" has no attribute "query"mypy(error)
```

MatthieuDartiailh, 2022-03-06:
The `open_resource` method of the ResourceManager returns a resource whose class match the resource string provided as argument. Mypy cannot validate the type which depends on the value of the resource string and hence we use the most generic resource class Resource. Since it may represent a register based resource, it does not have a query method. 

This explains why Mypy is confused. The best way to handle this is to annotate the variable in which you store the resource with the resource class you know is appropriate (or MessageBasedResource if you not want to assume the underlying transport layer). Mypy will complain about the assignation but you can just silence with a type: ignore.

It is not ideal bit it is the best we can do at the moment I believe.

abunimeh, 2022-03-07:
Thanks for your suggestion. I think the solution provided above is pretty good. At least my IDE and mypy can trace `query()` now:

```
self.inst: pyvisa.resources.MessageBasedResource
self.inst = self.pyvisa_rm.open_resource(address)  # type: ignore
```
works with.
`self.inst.query("*IDN?")`

closed 2022-03-07 by abunimeh