pyvisa/pyvisa #660: [REQUEST] Support for integrating pyvisa-proxy

casabre, 2022-03-26:
Hi Pyvisa Devs,

I am currently [developing an extension](https://github.com/casabre/pyvisa-proxy) for pyvisa which should somehow reflect the behavior of the NI remote VISA functionality but without the constraint of running NI-VISA. Due to the fact that there are many VISA flavors out there and NI-VISA is a bloated installer, I want to develop an independent and generic solution based on zmq which are connecting legacy protocols as GPIB or some USB connection. In the end, with mostly and ethernet-based instruments and pyvisa-proxy, distributed, dockerized tests will be feasible like for instance with my corresponding [Docker Image](https://hub.docker.com/r/casabre/rsvisa-python).

The architecture is not thought to then end but I just want to see if the PoC is working. However, after starting, I realized that implementing the client side is pretty hard because I want to forward the attribute calls to the server. Without relying on a specific class implementation, it was almost impossible to get the right attributes for forwarding.

Thus, I am asking if you could support me in order to integrate the logic into pyvisa with either a more monkey-patching approach of `__get_attr__` or an embedded implementation. 

Regarding the resource string, I have something similar in mind like the NI-VISA string &rarr; `proxy://hostname:port/remote resource`. After parsing, the pyvisa-proxy extension has to be loaded.

If that sounds interesting to you and I believe you have questions, please contact me.

Thanks & best regards
Carsten

LongnoseRob, 2022-03-26:
Interesting idea, 

how different is you approach from [LXI](https://www.lxistandard.org/About/LXI-Protocols.aspx) ?
Or from using a socket resource as already supported by pyvisa and pyvisa-py (or other backends)?

casabre, 2022-03-26:
Hm, my approach would be quite lean compared to LXI. I want to create just a RPC tunnel to reach the real VISA wrapper. LXI defines complete protocols which is too heavy 😉.

I wasn't aware of the already available socket connections. Is that a PAIR connection (single peer &rarr; single peer)? 

My goal is to create an async server which can keep multiple connections, e.g. Raspberry Pi. Thus, I am using zmq because it puts a convenience layer on top regarding multi-peer requests.

casabre, 2022-03-26:
I don't want to reinvent the wheel and if something similar can do the job, I will check out.

For me, the managed-multi VISA connections are appealing and the rpc is just tunneled to the server. With that, you have just a single host rather multiple IPs and ports.

MatthieuDartiailh, 2022-03-26:
Hi @casabre,

Your idea sounds interesting. Having access to remote resources using non TCPIP transport layers can help in some situations. Since you are going to define custom resource names you will have to associate to them the corresponding resource classes so I am not sure you need much on the PyVISA side apart from maybe full access to resource string to when selecting the resource rather than just the transport layer and protocol as is currently done. Since your resources will be thin wrapper around existing resource you could simply automatically generate your wrapper by inspecting the resource class they wrap, no ?

I won't have time to commit any work on your behalf but I will be happy to discuss possible extensions you may need and to review PRs.

casabre, 2022-03-26:
@MatthieuDartiailh thanks for your input and the offer. 

I was hoping to get some guidance, pointers and reviewing rather than coding.

I am confident that a simple wrapper based on the resource name would work and there would be just the need for a if clause like

```
if proxy:
   from pyvisa_proxy import ProxyClient
   resource, host = split_resource(resource_name)
   cls = get_cls_from_resource(resource)
   return ProxyClient(resource, host, cls)
```

My candidates for implementing this would be `list_resources` and `open_resource`.

Do you agree? Are there better ways to implement this? Like a specific proxy backend? I just thought that a extended resource_name would be the easiest solution.

MatthieuDartiailh, 2022-03-26:
The natural way to implement your idea is as a different backend for pyvisa that people would select as below for example:
```python
rm = pyvisa.ResourceManager("@remote")
```
you could also specify the IP address and port of the default server before the backend type as:
```python
rm = pyvisa.ResourceManager("192.168.0.1:7878@remote")
```

Your backend would define remote resource and register them using `Resource.register` as is done here for example https://github.com/pyvisa/pyvisa/blob/main/pyvisa/resources/gpib.py#L136. 

Since VISA uses `::` as separator you could use `REMOTE.GPIB::{server_ip}::{server_port}::{gpib_address}::INSTR`. This is somewhat inelegant since you end up with many interface types and something like `REMOTE::{server_ip}::{server_port}::GPIB::{gpib_address}::INSTR` would look better. However it means the resource selection would occur based on more than just the interface_type and resource_cls which is not possible at the moment, but we could maybe work something out. The resource names can be registered in a similar manner to resources themselves. 

Since NI already has a format for remote resources it may be interesting to use the same if it is practical to do so.

casabre, 2022-03-27:
Yes, I agree that from software architecture perspective, the backend extension is more reasonable. And I also agree that the `resource_name` is inelegant and would break the common pattern. In the end, it would confuse user more than they would benefit from it.

Thus, I am more in favor of having a similar `resource_name` like the NI Remote VISA string &rarr; `proxy://{server_ip}:{server_port}/{gpib_address}::INSTR`.

I believe this is also quite easy integrate-able with a implementation like the snippet from previous comment
> ```python
>  if proxy:
>      from pyvisa_proxy import ProxyClient
>      resource, host = split_resource(resource_name)
>      cls = get_cls_from_resource(resource)
>      return ProxyClient(resource, host, cls)
>  ```

But I believe that the `list_resources` is somehow left out because we would need another `kwarg` &rarr; e.g. `proxy=(host,port)`

I prepared my code yesterday more or less to that approach. I would give it a try and if it turn out, there is too much left out, maybe the backend approach is the way to go.

What do you think?

casabre, 2022-04-01:
@MatthieuDartiailh I could need your help regarding some implementation details :).

I like the generic approach with
> `rm = pyvisa.ResourceManager("192.168.0.1:7878@remote")`

In the end, I want to stick with the current `resouce_name` pattern. Thus, I am going to implement the `open`function that way that the cls is forwarded into my `ProxyResource` class which is returned. Thus, I believe that I can stick with the current pattern.

Could you guide me how I can adapt the `__new__` function in `highlevel.py` of pyvisa in a non-intrusive way that I can retrieve the `IP` address and the `port` and forward it into the `pyvisa-proxy` `highlevel.py` class in order to establish the zmq connection.

MatthieuDartiailh, 2022-04-02:
Sorry for the low responsivity. Whatever you provide before the @ when creating a resource manager will be passed to the visa library under its `library_path` attribute so I do not think you need to actually play with new here.

casabre, 2022-04-02:
No worries and that definitely helps!

> Whatever you provide before the @ when creating a resource manager will be passed to the visa library under its library_path attribute so I do not think you need to actually play with new here. Does that help ?

So, I believe I have no need to patch anything in pyvisa rather that creating a fitting `WRAPPER`in `pyvisa-proxy`. I am using some files "as inspiration" from `pyvisa-sim`.

If `pyvisa-proxy` is working, are you interested in moving it into the pyvisa organization and I would stay maintainer?

MatthieuDartiailh, 2022-04-02:
If you are ready to commit to maintaining it, yes I would be open to move your package under the organisation once it is mature enough.

closed 2022-05-03 by MatthieuDartiailh

MatthieuDartiailh, 2022-05-12:
1.12.0 is out so you can depend on it.

casabre, 2022-05-12:
Awesome 😎! @MatthieuDartiailh thanks for letting me know 😊

PR #661 (2022-05-03): Adding support for pyvisa_proxy #660

- [x] Closes #660
- [x] Executed ``black . && isort -c . && flake8`` with no errors
- [x] The change is fully covered by automated unit tests
- [x] Documented in docs/ as appropriate
- [x] Added an entry to the CHANGES file