pyvisa/pyvisa #662: pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed

arisezxcvbnm, 2022-04-05:
I got a issue here.

I am using ubuntu 18.04 with an RS232 cable to link on a ElectronicLoad Dingchen DCL8006

The SCPI list for DCL8006 is [here](http://www.czdingchen.com/Uploads/file/20180322/1521706734917408.pdf) (and it is Chinese version so I am not sure will that be read or not. If anyone want to know about it, I can explain it.)

I am sure I am using the right code without spelling issue for that.

Here is my code:
""""""""""""""""""""""""""""""""""""""""""""
import pyvisa 
import time

import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s '
          '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

class TEST:
    def __init__(self):
        print("Test the script!!!")
        self.rm = pyvisa.ResourceManager()
        self.load01 = self.rm.open_resource("ASRL33::INSTR",baud_rate=9600)
        self.load01.timeout = 500
        self.load01.write_termination = "\r\n"
        self.load01.read_termination = "\r\n"
        self.load01.StopBits = 1
        self.load01.send_end = True
        a = ["*IDN?","FETCh:VOLtage?"]
        for i in a:
            b = i
            if ("?" in b):
                self.command = self.load01.query(b)
                print(self.command)
            else:
                self.command = self.load01.write(b)
        self.load01.close()

""""""""""""""""""""""""""""""""""""""""""""""""""

And here is the result:

"""""""""""""""""""""""""""""""""""""""""""""""""""""

2022-04-05 21:42:24,897 MainThread      DEBUG    resource       :293      ASRL33::INSTR - opening ...
2022-04-05 21:42:24,898 MainThread      DEBUG    highlevel      :182      viOpen(4097, 'ASRL33::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x7efd0ce9fcc0>') -> 0
2022-04-05 21:42:24,898 MainThread      DEBUG    resource       :318      ASRL33::INSTR - is open with session 1
2022-04-05 21:42:24,898 MainThread      DEBUG    highlevel      :182      viSetAttribute(1, 1073676321, 9600) -> 0
2022-04-05 21:42:24,898 MainThread      DEBUG    highlevel      :182      viSetAttribute(1, 1073676314, 500) -> 0
2022-04-05 21:42:24,898 MainThread      DEBUG    highlevel      :182      viSetAttribute(1, <ResourceAttribute.termchar: 1073676312>, 10) -> 0
2022-04-05 21:42:24,899 MainThread      DEBUG    highlevel      :182      viSetAttribute(1, <ResourceAttribute.termchar_enabled: 1073676344>, 1) -> 0
2022-04-05 21:42:24,899 MainThread      DEBUG    highlevel      :182      viSetAttribute(1, 1073676310, <VisaBoolean.true: 1>) -> 0
2022-04-05 21:42:24,899 MainThread      DEBUG    highlevel      :182      viWrite(1, b'*IDN?\r\n', 7, 'c_uint(7)') -> 0
2022-04-05 21:42:24,899 MainThread      DEBUG    messagebased   :436      ASRL33::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
**2022-04-05 21:42:24,961 MainThread      DEBUG    highlevel      :182      viRead(1, <ctypes.c_char_Array_20480 object at 0x7efd0ce9fcc0>, 20480, 'c_uint(40)') -> 0
DingChen,DCL8006 ,L806X312315A0001,1.7**
2022-04-05 21:42:24,962 MainThread      DEBUG    highlevel      :182      viWrite(1, b'FETCh:VOLtage?\r\n', 16, 'c_uint(16)') -> 0
2022-04-05 21:42:24,962 MainThread      DEBUG    messagebased   :436      ASRL33::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
**2022-04-05 21:42:25,463 MainThread      DEBUG    highlevel      :182      viRead(1, <ctypes.c_char_Array_20480 object at 0x7efd0ce9ff40>, 20480, 'c_uint(0)') -> -1073807339
2022-04-05 21:42:25,463 MainThread      DEBUG    messagebased   :445      ASRL33::INSTR - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')**
Traceback (most recent call last):
  File "main.py", line 45, in <module>
    main = MAIN()
  File "main.py", line 9, in __init__
    self.normalRun()
  File "main.py", line 17, in normalRun
    Test.TEST()                
  File "/home/arise/Documents/Coding/Python/Glary-Power-Project/PSO-Circuit-Test/PSO-Circuit-Test-Master-0-0-2H/lib/test/test.py", line 25, in __init__
    self.command = self.load01.query(b)
  File "/home/arise/.local/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 644, in query
    return self.read()
  File "/home/arise/.local/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "/home/arise/.local/lib/python3.8/site-packages/pyvisa/resources/messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "/home/arise/.local/lib/python3.8/site-packages/pyvisa/ctwrapper/functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "/home/arise/.local/lib/python3.8/site-packages/pyvisa/ctwrapper/highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "/home/arise/.local/lib/python3.8/site-packages/pyvisa/highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
**pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.**
2022-04-05 21:42:25,468 MainThread      DEBUG    highlevel      :3093     Closing ResourceManager (session: 4097)
2022-04-05 21:42:25,468 MainThread      DEBUG    resource       :332      ASRL33::INSTR - closing
2022-04-05 21:42:25,468 MainThread      DEBUG    highlevel      :182      viDisableEvent(1, <EventType.all_enabled: 1073709055>, <EventMechanism.all: 65535>) -> 1073676291
2022-04-05 21:42:25,468 MainThread      DEBUG    highlevel      :182      viDiscardEvents(1, <EventType.all_enabled: 1073709055>, <EventMechanism.all: 65535>) -> 1073676292
2022-04-05 21:42:25,469 MainThread      DEBUG    highlevel      :182      viClose(1,) -> 0
2022-04-05 21:42:25,469 MainThread      DEBUG    resource       :335      ASRL33::INSTR - is closed
2022-04-05 21:42:25,469 MainThread      DEBUG    highlevel      :182      viClose(4097,) -> 0

"""""""""""""""""""""""""""""""""""""""""""""""""""

So far I have try everything I have seen around this git, nothing change.
I can get the response from those *IDN? or *ESR?
But for any SCPI command with ? that have a query, it time out.

And I can command my load to do anything like Load one, set Voltage, Current, .......etc.
But if I want query, it time out.
any help will be good , thanks.

-------------------------------------------------------------
update:

pyvisa-info

""""""""""""""""""""""""""""""""""
Result:

Machine Details:
   Platform ID:    Linux-5.4.0-107-generic-x86_64-with-glibc2.27
   Processor:      x86_64

Python:
   Implementation: CPython
   Executable:     /usr/bin/python
   Version:        3.8.13
   Compiler:       GCC 7.5.0
   Bits:           64bit
   Build:          Mar 16 2022 17:28:59 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: /usr/lib/x86_64-linux-gnu/libvisa.so.21.5.0:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
   py:
      Version: 0.5.2
      ASRL INSTR: Available via PySerial (3.4)
      USB INSTR:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      USB RAW:
         Please install PyUSB to use this resource type.
         No module named 'usb'
      TCPIP INSTR: Available 
      TCPIP SOCKET: Available 
      GPIB INSTR:
         Please install linux-gpib (Linux) or gpib-ctypes (Windows, Linux) to use this resource type. Note that installing gpib-ctypes will give you access to a broader range of funcionality.
         No module named 'gpib'

----------------------------------------

result from STATus:RUN?:

""""""""""""""""""""""""""""""""""""""""""""""""

2022-04-05 22:22:28,401 MainThread      DEBUG    messagebased   :436      ASRL33::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2022-04-05 22:22:28,902 MainThread      DEBUG    highlevel      :182      viRead(1, <ctypes.c_char_Array_20480 object at 0x7fc4f0cf1d40>, 20480, 'c_uint(1)') -> -1073807339
2022-04-05 22:22:28,902 MainThread      DEBUG    messagebased   :445      ASRL33::INSTR - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
Traceback (most recent call last):

Other the same.

MatthieuDartiailh, 2022-04-05:
Thanks for the update issue. One thing that may happen is that you try to read the response too fast for the instrument. Try using the `delay` argument of query to manually wait after writing before reading.

arisezxcvbnm, 2022-04-06:
> Thanks for the update issue. One thing that may happen is that you try to read the response too fast for the instrument. Try using the `delay` argument of query to manually wait after writing before reading.

""""""""""""""""""""""""""""""""""""""""""""""""""""""""
....
....
....
a = ["STATus:RUN?"]
        for i in a:
            b = i
            if ("?" in b):
                self.command = self.load01.query(b,delay=2)
                time.sleep(1)
                print(self.command)
            else:
                self.command = self.load01.write(b)
        self.load01.close()

""""""""""""""""""""""""""""""""""""""""""""""""""""""

Still got same timeout

"""""""""""""""""""""""""""""""""""""""""""""""""""""""
2022-04-06 10:52:34,330 MainThread      DEBUG    resource       :293      ASRL33::INSTR - opening ...
2022-04-06 10:52:34,332 MainThread      DEBUG    highlevel      :182      viOpen(4097, 'ASRL33::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x7f03d34d7cc0>') -> 0
2022-04-06 10:52:34,332 MainThread      DEBUG    resource       :318      ASRL33::INSTR - is open with session 1
2022-04-06 10:52:34,332 MainThread      DEBUG    highlevel      :182      viSetAttribute(1, 1073676321, 9600) -> 0
2022-04-06 10:52:34,332 MainThread      DEBUG    highlevel      :182      viSetAttribute(1, 1073676314, 500) -> 0
2022-04-06 10:52:34,332 MainThread      DEBUG    highlevel      :182      viSetAttribute(1, <ResourceAttribute.termchar: 1073676312>, 10) -> 0
2022-04-06 10:52:34,332 MainThread      DEBUG    highlevel      :182      viSetAttribute(1, <ResourceAttribute.termchar_enabled: 1073676344>, 1) -> 0
2022-04-06 10:52:34,332 MainThread      DEBUG    highlevel      :182      viSetAttribute(1, 1073676310, <VisaBoolean.true: 1>) -> 0
2022-04-06 10:52:34,333 MainThread      DEBUG    highlevel      :182      viWrite(1, b'STATus:RUN?\r\n', 13, 'c_uint(13)') -> 0
2022-04-06 10:52:36,334 MainThread      DEBUG    messagebased   :436      ASRL33::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2022-04-06 10:52:36,835 MainThread      DEBUG    highlevel      :182      viRead(1, <ctypes.c_char_Array_20480 object at 0x7f03d34d7cc0>, 20480, 'c_uint(0)') -> -1073807339
2022-04-06 10:52:36,835 MainThread      DEBUG    messagebased   :445      ASRL33::INSTR - exception while reading: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.
Buffer content: bytearray(b'')
Traceback (most recent call last):
"""""""""""""""""""""""""""""""""""""""""""""""""""""""""

arisezxcvbnm, 2022-04-08:
I collect some of working code on DCL8006, and I also try to read bytes or read() but the issue looks the same, most of command don't work. Not be able to to make SCPI command happened, or time out.

,,,python

import pyvisa
from pyvisa import constants
import time

import logging
FORMAT = ('%(asctime)-15s %(threadName)-15s '
          '%(levelname)-8s %(module)-15s:%(lineno)-8s %(message)s')
logging.basicConfig(format=FORMAT)
log = logging.getLogger()
log.setLevel(logging.DEBUG)

rm = pyvisa.ResourceManager()
info = rm.resource_info("ASRL33::INSTR")
print(info)
load01 = rm.open_resource("ASRL33::INSTR",baud_rate=9600)
load01.open_timeout = 500
load01.write_termination = "\r\n"
load01.read_termination = "\r\n"

a = ["*IDN?","CURRent 5","CURRent?","VOLTage 2","VOLTage?","FETCh:CURRent?"]
for i in a:
    time.sleep(0.5)
    b = i
    try:
        if ("?" in b):
            command = load01.write(b)
            time.sleep(0.01)
            message = load01.read()
            print(message)
            print(load01.last_status)
        else:
            command = load01.write(b)
            print(load01.last_status)
    except:
        pass
load01.close()
,,,

and here is the result:

,,,
2022-04-08 09:45:19,721 MainThread      DEBUG    highlevel      :2908     No visa library specified, trying to find alternatives.
2022-04-08 09:45:19,722 MainThread      DEBUG    highlevel      :2912     Environment variable PYVISA_LIBRARY is unset.
2022-04-08 09:45:19,725 MainThread      DEBUG    util           :153      Not loading dll_extra_paths because we are not on Windows or Python < 3.8
2022-04-08 09:45:20,025 MainThread      DEBUG    highlevel      :93       Automatically found library files: ['/usr/lib/x86_64-linux-gnu/libvisa.so.21.5.0', None, None, None, None]
2022-04-08 09:45:20,026 MainThread      DEBUG    util           :94       No user defined library files
2022-04-08 09:45:20,027 MainThread      DEBUG    highlevel      :2872     The IVI implementation available
2022-04-08 09:45:20,027 MainThread      DEBUG    util           :153      Not loading dll_extra_paths because we are not on Windows or Python < 3.8
2022-04-08 09:45:20,307 MainThread      DEBUG    highlevel      :93       Automatically found library files: ['/usr/lib/x86_64-linux-gnu/libvisa.so.21.5.0', None, None, None, None]
2022-04-08 09:45:20,308 MainThread      DEBUG    util           :94       No user defined library files
2022-04-08 09:45:20,319 MainThread      DEBUG    highlevel      :170      Library signatures: 90 ok, 0 failed
2022-04-08 09:45:20,320 MainThread      DEBUG    highlevel      :201      Created library wrapper for /usr/lib/x86_64-linux-gnu/libvisa.so.21.5.0
2022-04-08 09:45:20,331 MainThread      DEBUG    highlevel      :182      viOpenDefaultRM('<ViObject object at 0x7fe566342dc0>',) -> 0
2022-04-08 09:45:20,332 MainThread      DEBUG    highlevel      :3042     Created ResourceManager with session 4097
2022-04-08 09:45:20,334 MainThread      DEBUG    highlevel      :182      viParseRsrcEx(4097, 'ASRL33::INSTR', 'c_ushort(4)', 'c_ushort(33)', <ctypes.c_char_Array_256 object at 0x7fe566289e40>, <ctypes.c_char_Array_256 object at 0x7fe566289bc0>, <ctypes.c_char_Array_256 object at 0x7fe566289a40>) -> 0
ResourceInfo(interface_type=<InterfaceType.asrl: 4>, interface_board_number=33, resource_class='INSTR', resource_name='ASRL33::INSTR', alias=None)
2022-04-08 09:45:20,335 MainThread      DEBUG    highlevel      :182      viParseRsrcEx(4097, 'ASRL33::INSTR', 'c_ushort(4)', 'c_ushort(33)', <ctypes.c_char_Array_256 object at 0x7fe566289bc0>, <ctypes.c_char_Array_256 object at 0x7fe566289e40>, <ctypes.c_char_Array_256 object at 0x7fe566289140>) -> 0
2022-04-08 09:45:20,335 MainThread      DEBUG    resource       :293      ASRL33::INSTR - opening ...
2022-04-08 09:45:20,338 MainThread      DEBUG    highlevel      :182      viOpen(4097, 'ASRL33::INSTR', <AccessModes.no_lock: 0>, 0, '<ViObject object at 0x7fe566342dc0>') -> 0
2022-04-08 09:45:20,338 MainThread      DEBUG    resource       :318      ASRL33::INSTR - is open with session 1
2022-04-08 09:45:20,338 MainThread      DEBUG    highlevel      :182      viSetAttribute(1, 1073676321, 9600) -> 0
2022-04-08 09:45:20,338 MainThread      DEBUG    highlevel      :182      viSetAttribute(1, <ResourceAttribute.termchar: 1073676312>, 10) -> 0
2022-04-08 09:45:20,338 MainThread      DEBUG    highlevel      :182      viSetAttribute(1, <ResourceAttribute.termchar_enabled: 1073676344>, 1) -> 0
2022-04-08 09:45:20,839 MainThread      DEBUG    highlevel      :182      viWrite(1, b'*IDN?\r\n', 7, 'c_uint(7)') -> 0
2022-04-08 09:45:20,850 MainThread      DEBUG    messagebased   :436      ASRL33::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
2022-04-08 09:45:20,896 MainThread      DEBUG    highlevel      :182      viRead(1, <ctypes.c_char_Array_20480 object at 0x7fe566342dc0>, 20480, 'c_uint(40)') -> 0
[... 8 lines cut ...]
StatusCode.success
2022-04-08 09:45:22,421 MainThread      DEBUG    highlevel      :182      viWrite(1, b'VOLTage 2\r\n', 11, 'c_uint(11)') -> 0
StatusCode.success
2022-04-08 09:45:22,922 MainThread      DEBUG    highlevel      :182      viWrite(1, b'VOLTage?\r\n', 10, 'c_uint(10)') -> 0
2022-04-08 09:45:22,933 MainThread      DEBUG    messagebased   :436      ASRL33::INSTR - reading 20480 bytes (last status <StatusCode.success_max_count_read: 1073676294>)
[... cut ...]
2022-04-08 09:45:25,461 MainThread      DEBUG    resource       :332      ASRL33::INSTR - closing
2022-04-08 09:45:25,463 MainThread      DEBUG    highlevel      :182      viClose(4097,) -> 0
,,,

and if I take out the [try except] format, the trace back will be the same as the timeout trace which point on the start of this thread. Hope this can be helpful, thanks.
I will also try to test some other scpi device on my computer to make sure if this is my computer's problem or the device's problem, although I think these was less likely but yeah.

MatthieuDartiailh, 2022-04-08:
Looking at this, it appears your code is fine. The only thing that comes to my mind glossing over the manual (I do not read chinese), is that you need to send another command first to get the instrument to measure something. A natural guess would be TRIG but the best would be to figure it out from the manual.

arisezxcvbnm, 2022-04-09:
I did recheck the manual, it seems it did not mention any command as the pre-require to make machine work properly.

I will:
*Goes to the factory get some technical support see if any other get same issue like me or if there are any other pre-require setting which I did not know yet.
*Test the same code on different device.(I also got another two SCPI machine, I will test the same code on them to check if that is python problem or machine problem
*Test the code on different computer, see if that is my OS, or my package installation issue.(Although this one is less likely to be the issue but just make sure.)

After I done this all, I will make an update here again, thank you for your support.

arisezxcvbnm, 2022-04-09:
Issue solved!

After I turn to factory support, they say the SCPI codes has been changed..................................
And they forgot to update the manuel...............................................
=_= (It take me almost a week to dig in 15 hrs per day.............. )

I have no word....Orz

Anyway, sorry to bother you for so long. Thanks for your support, I got my work back on line now.
Thanks you very much.

MatthieuDartiailh, 2022-04-09:
Happy you solved it

closed 2022-04-09 by MatthieuDartiailh