pyvisa/pyvisa #664: Keithley 2000 Trigger Issue

MarkusHumer, 2022-04-13:
Hello,

I am trying to trigger a voltage measurement with 100 readings on my Keithley 2000. 
The idea is to start the measurement using a bus command ("GET") and then later read the data all at once. 

I tried to follow the example (https://github.com/pyvisa/pyvisa/blob/main/docs/source/introduction/example.rst), however I run into a problem once I try to initialize the device, and consequently when I try to trigger using a bus command. 

```
import csv
import logging
import pyvisa as visa
from time import sleep
from time import perf_counter
import numpy as np
import pandas as pd
from datetime import datetime as DateTime

rm = visa.ResourceManager()
rm.list_resources()

K2000 = rm.open_resource("ASRL7::INSTR")

K2000.baud_rate = 19200
K2000.data_bits = 8
K2000.parity = 0
K2000.term_chars = "\r"
K2000.read_termination = "\r"
K2000.timeout = 20000

K2000.query("*IDN?")

# So far so good

# Clear Status, event registers, error queue
K2000.write("*CLS")
# Reset Device
K2000.write("*RST")

# Set DC Voltage Function
K2000.write(":CONF:VOLT:DC")
# Voltage Range 5V
K2000.write(":VOLT:DC:RANG 10")
# 7 digits measurement resolution
K2000.write(":VOLT:DC:DIG 7")

# Display on
K2000.write(":DISP:ENAB ON")
# Reading Buffer delete
K2000.write("TRAC:CLE")
# Reading buffer. Answer: Bytes available, Bytes reserved
K2000.query("TRAC:FREE?")

# Define Number of Readings
K2000.write("TRAC:POIN 100")
# query Number of Readings
K2000.query("TRAC:POIN?")

# Define Source of Readings
K2000.write("TRAC:FEED SENS")
# Query Source of Readings
K2000.query("TRAC:FEED?")

# Buffer control. Next = fills buffer and then stops
K2000.write("TRAC:FEED:CONT NEXT")
K2000.query("TRAC:FEED:CONT?")

# Sample Count... Defines how many times the operation loops around the trigger model Should be 1!!
K2000.query("SAMP:COUN?")

# Trigger Source 
K2000.write("TRIG:SOUR BUS")
# Trigger Delay
K2000.write("TRIG:DEL 0")
K2000.query("TRIG:DEL?")

# Now initialize....
K2000.write("INIT")

# Now trigger...
K2000.assert_trigger()

```
This results in an error...
VisaIOError: VI_ERROR_INV_SETUP (-1073807302): Unable to start operation because setup is invalid (usually due to attributes being set to an inconsistent state).

I also tried starting the measurement using K2000.write("GET") or K2000.write("'TRIG"), which resulted in error -113 "undefined header".

What am I doing wrong?

Thank you very much,
best regards,
Markus

MatthieuDartiailh, 2022-04-13:
Can you post the full traceback of your error ?

Also note that the example is using a GPIB resource not a serial one and I am pretty sure serial resource have no notion of bus command (since it is a peer-tp-peer connection).

MatthieuDartiailh, 2022-04-18:
As per the VISA specification of viAssertTrigger:
For a session to a Serial device or TCP/IP socket, if VI_ATTR_IO_PROT isVI_PROT_4882_STRS, the device is sent the string “*TRG\n”; otherwise, this operation is not valid.

So you either need to `io_protocol` to `constants.IOProtocol.protocol4882_strs` or simply send `"*TRG"` using `write`. According to its manual the Keithley 2000 supports it.

Hope this helps.

MarkusHumer, 2022-04-20:
Hi Matthieu,

Thank you very much for your support, indeed "*TRG" seems to work. Either I thought "GET" would do the same or I forgot the star before... 

Also there seems to be an issue (Device does not respond to the "TRAC:DATA?" command) when the "Trace Points" and the "Sample Counts" are not the same, but keeping both numbers equal (as in the example) seems to do the trick. 

Now it works the way it should, so thank you again,
best regards,
Markus

MatthieuDartiailh, 2022-04-20:
Happy to help

closed 2022-04-20 by MatthieuDartiailh