pyvisa/pyvisa #665: open_resource() by a function in a method

FernandoFGF, 2022-04-20:
Hi, im a bit noob with pyvisa and python but Im trying to iniciate open_resource() in a function from a method, but when I write:
`rta = open_resource('TCPIP::192.168.100.101::INSTR')` that said me rta cant be resolve by pylance
I want that the function return me rta from the method, is this posible?

MatthieuDartiailh, 2022-04-20:
open_resource is a method of the ResourceManager class. You need to first create a ResourceManager instance and then call its open_resource method. The documentation has plenty of examples.

FernandoFGF, 2022-04-20:
I did that:
`rm = pyvisa.ResourceManager()
rta = rm.open_resource('TCPIP::192.168.100.101::INSTR')`
but when I include it in a method this dont work.

MatthieuDartiailh, 2022-04-20:
Can you please post your code otherwise it is really hard to figure out what is going on ?

FernandoFGF, 2022-04-20:
lab_module.py:
```
def init_pyvisa():
    rm = pyvisa.ResourceManager()
    rta = rm.open_resource('TCPIP::192.168.100.101::INSTR')
    return rta
```
waveform_DAQ.py:
```
import lab_module as lm
rta=lm.init_pyvisa()
```
something like that

FernandoFGF, 2022-04-20:
and then I could:
lab_module.py:
```
def trigger(rta):
return rta.query("TRIGger:A:LEVel1?")
```
waveform_DAQ.py:
`trigger= lm.trigger(rta)`

MatthieuDartiailh, 2022-05-12:
Can you make a minimal reproducer that I could run ? Otherwise I cannot investigate this really.