pyvisa/pyvisa #666: [COM] Communication issue with Keithley 2182A Nanovoltmeter using Pyvisa

RobinPYM, 2022-04-24:
I'm trying to write a python script using pyvisa to control my Keithley 2182A to measure voltage. However, every time I run the script, I have a timeout error. I tried to run my code line-by-line in IDLE and it worked. But if I run "keithley.read_bytes(1)" or "keithley.read()" following a write command line in IDLE, I get the same timeout error. I think I might be facing some reading issue because I think the writing process has no problem. Does anyone see a problem? Thank you so much!

Code
------------------
import pyvisa
rm = pyvisa.ResourceManager()
resources = rm.list_resources()
keithley = rm.open_resource(resources[0])

print(keithley.query('*IDN?')) # query what is the device
keithley.write("*rst; status:preset; *cls")  # initialization and reset message
keithley.timeout = 3000
keithley.read_termination = '\n'
keithley.write_termination = '\n'

keithley.write(":sens:chan 1")  # Select Channel 1
keithley.write(":sens:func 'volt'") # Select DCV1
print(keithley.query(":sens:data:fres?"))  # Request a fresh reading

Error
------------------
KEITHLEY INSTRUMENTS INC.,MODEL 2182A,4067825,C04  /A02  

Traceback (most recent call last):
  File "d:\0Study\4.College\Research\UCSB_Materials_Wilson_Group\Keithley\Keithley_2182A_code.py", line 24, in <module>
    print(keithley.query(":sens:data:fres?"))  # Request a fresh reading
  File "C:\Users\24991\AppData\Local\Programs\Python\Python310\lib\site-packages\pyvisa\resources\messagebased.py", line 644, in query
    return self.read()
  File "C:\Users\24991\AppData\Local\Programs\Python\Python310\lib\site-packages\pyvisa\resources\messagebased.py", line 486, in read
    message = self._read_raw().decode(enco)
  File "C:\Users\24991\AppData\Local\Programs\Python\Python310\lib\site-packages\pyvisa\resources\messagebased.py", line 442, in _read_raw
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Users\24991\AppData\Local\Programs\Python\Python310\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Users\24991\AppData\Local\Programs\Python\Python310\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler       
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Users\24991\AppData\Local\Programs\Python\Python310\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

Instrument details
------------------
* Model: Keithley 2182A Nanovoltmeter
* Communication: GPIB to USB using KUSB488B converter

Output of `pyvisa-info`
-----------------------
Machine Details:
   Platform ID:    Windows-10-10.0.19044-SP0
   Processor:      Intel64 Family 6 Model 140 Stepping 1, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\24991\AppData\Local\Programs\Python\Python310\python.exe
   Version:        3.10.4
   Compiler:       MSC v.1929 64 bit (AMD64)
   Bits:           64bit
   Build:          Mar 23 2022 23:13:41 (#tags/v3.10.4:9d38120)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments

MatthieuDartiailh, 2022-04-25:
Hi,
First I would try to move your setting of read and write termination before your IDN query to be sure your parameters are correct. Second your last query looks funny to me FRES is not usual for a multimeter, do you mean FREQ ? Note I did not check the manual so this is just a guess.

RobinPYM, 2022-04-25:
> 
Thanks for the advice. I put the read and write termination before the IDN query and the same error happened when I ran the script. And I double-checked the manual, FRES stands for "return a new (fresh) reading", so that perhaps is not the problem. It's so weird to me that my code won't work when run as a script, but it works when I run the same thing line-by-line in IDLE.

RobinPYM, 2022-04-25:
> 

Another thing is, when I run the script, it shows two error messages on the monitor of my Keithley, and I looked them up on the manual: "Error -230" which means "Data corrupt or stale"; and "Error -420" which means "Query unterminated". But they don't show up when I run "print(keithley.query(":sens:data:fres?"))" line in IDLE.

MatthieuDartiailh, 2022-04-25:
The query unterminated usually means that you forgot to read a response to a query before sending the next order. 

Checking the manual, I was wrong and your query looks fine actually. Is it possible that the reading takes a long time to be acquired (> 3s) and could cause the timeout ?

RobinPYM, 2022-04-25:
I tried timeout = 30000 and it still has the error. I don't think it takes a long time to acquire data because when I run in IDLE line-by-line it pops out instantaneously. 

I found the following example in the manual, and it seems that I need to add something following the fresh command to read from Keithley. Sorry I'm new to SCPI command, do you possibly know what functions should I use in python?

‘ Measure voltage on Channel 1: 
CALL SEND(7,“:sens:chan 1”,status%) ‘Select Channel 1.
CALL SEND(7,“:sens:func ‘volt’”,status%) ‘Select DCV1.
CALL SEND(7,“:sens:data:fres?”,status%) ‘Request a fresh
‘reading. 
reading$ = SPACE$(80)
CALL ENTER(reading$, length%, 7, status%) ‘Address 2182 to talk.
PRINT reading$ ‘Display reading on CRT.

MatthieuDartiailh, 2022-04-25:
The manual would translate in what you are doing I think.

Just to be sure, when you run in IDLE line by line, i.e. with a large delay between two commands everything works ? If that is right, you can try adding a sleep before issuing the query the instrument may take some time to process the previous command.

RobinPYM, 2022-05-01:
I eventually solved the issue by adding an initiation command after the reset line, since the initiation command brings the instrument out of idle state and makes it able to listen to commands. Thank you so much for the help!

closed 2022-05-01 by RobinPYM