pyvisa/pyvisa #667: Support for Cgywin/IVI binary for Cygwin

eorojas, 2022-04-29:
I'm successfully running pyvisa under Windows 10 command window.
But I'd really like to be running under Cygwin.  
I can install pyvisa under Cygwin, but when accessing pyvisa I get the following backtrace.

>   File "/home/Engineering3/automation/gpib_instrument.py", line 28, in __init__
>     self.gpib = pyvisa.ResourceManager().open_resource(address)
>   File "/usr/local/lib/python3.9/site-packages/pyvisa/highlevel.py", line 3015, in __new__
>     visa_library = open_visa_library(visa_library)
>   File "/usr/local/lib/python3.9/site-packages/pyvisa/highlevel.py", line 2924, in open_visa_library
>     wrapper = _get_default_wrapper()
>   File "/usr/local/lib/python3.9/site-packages/pyvisa/highlevel.py", line 2883, in _get_default_wrapper
>     raise ValueError(
> ValueError: Could not locate a VISA implementation. Install either the IVI binary or pyvisa-py.
> 
I admit to not digging very deep, but I hoped to find areference to how to install the "IVI binary".

Also installing pyvisa-py has a different error.

MatthieuDartiailh, 2022-04-29:
PyVISA is a wrapper for an IVI compliant VISA implementation. You can install one from any of the major vendors: National Instruments, Keysight, R&S, Tektronik. The dcoumentation does discuss this but the error message could maybe be improved. HoweverI have no idea what you need to do to get something working with Cygwin (with which I am not familiar beyond the name).

eorojas, 2022-05-04:
I was trying to do something like @fishersa did [here](https://github.com/pyvisa/pyvisa/issues/131), 
I don't get a segment fault but pyvisa is still get the error: "Could not locate a VISA implementation. Install either the IVI binary or pyvisa-py."
`
Python 3.9.10 (main, Jan 20 2022, 21:37:52)
[GCC 11.2.0] on cygwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import ctypes
>>> wpath='C:/Windows/System32/visa64.dll'
>>> wlib = ctypes.cdll.LoadLibrary(wpath)
>>> wlib
<CDLL 'C:/Windows/System32/visa64.dll', handle 7ffb85c20000 at 0x6fffffec51f0>
>>> hasattr(wlib, '_functions')
False
>>>
`
I don't get the segfault, but __functions_ is not present.

I've also tried setting the path to the DLL in .pyvisarc and I get file not found.

any ideas welcome.

MatthieuDartiailh, 2022-05-09:
I have no idea how deep Cygwin emulation of linux runs but typically on Windows the visa DLL need to be opened using WinDLL. Also the DLL itself does not export _functions, this attribute is set on the wrapper around the ctype object by PyVISA. you can try to have a look at ctwrapper/highlevel.py at line 161 and try to do things manually to see where things block.

PR #703 (2022-11-22): Extended FAQ to include using PyVISA on Cygwin on Windows.

- [ ] Closes #667 (insert issue number if relevant)

I have documented in the FAQ how to use PyVISA with Cygwin by setting either a .pyvisarc or by specifying the visaXX.dll