pyvisa/pyvisa #672: CI: disable workflows for forks / restrict workflow to pyvisa org

LongnoseRob, 2022-05-19:
Lately I am getting messages from github,that some of the CI workflow jobs have failed whenever I update my fork of pyvisa.
I think the failure itself is related to Codecov not being set-up for my repo.

Checking online for similiar issues, I came across this [issue](https://github.com/prisma/prisma/issues/3539), and especially [this reocmmendation](https://github.com/prisma/prisma/issues/3539#issuecomment-690260573)

Would it be possible to change the workflow/job-config in a similiar matter for pyvisa?

MatthieuDartiailh, 2022-05-20:
Do you want to fully disable the workflows or simply protect the parts that cannot succeed outside of the org ? In the first case it would be easier to simmly disable things on your side. Otherwise feel free to make a PR.

LongnoseRob, 2022-05-22:
For the moment I simply disabled the worflows in my fork.

closed 2022-05-22 by LongnoseRob