pyvisa/pyvisa #675: [COM] Communication issue with Keithley 6221 using Pyvisa

lmorningstar13, 2022-06-28:
Instrument details
------------------
* Model: Keithley 6221 AC & DC Current Source
* Communication: GPIB
* Link to the documentation (if available): https://www.testequipmentdepot.com/keithley/current-sources/ac-dc-current-source-6221.htm

Output of `pyvisa-info`
-----------------------
Machine Details:
   Platform ID:    Windows-10-10.0.22000-SP0
   Processor:      Intel64 Family 6 Model 166 Stepping 0, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\laqm\anaconda3\python.exe
   Version:        3.9.12
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Apr  4 2022 05:22:27 (#main)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1378277095
         Spec. Version: 7340288
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: Keysight Technologies
         Impl. Version: 1378277095
         Spec. Version: 7340288
   py:
      Version: 0.5.3
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: libusb0
      USB RAW: Available via PyUSB (1.2.1). Backend: libusb0
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR: Available via Linux GPIB (None)
      GPIB INTFC: Available via Linux GPIB (None)

Description of issue
-----------------------
Hello, I am attempting to use pyvisa to communicate with Keithley instrument 6221. 
I have tried running the following code:

import pyvisa as visa
rm = visa.ResourceManager('@py')
print(rm.list_resources())

However, my list of resources is empty. I believe I have everything I need installed such as pyvisa, pyvisa-py, pyusb, libusb, IO Libraries Suite Downloads.
I think this may be a pathing issue since I'm using windows, but I'm also new to windows so I may be wrong. If it is a pathing issue, is there any suggestions on how to fix that sort of issue?

Thank you! Any help would be appreciated because I am completely stumped on this.

MatthieuDartiailh, 2022-06-28:
First on Windows I would recommend using the ivi backend (i.e. not passing any argument to the ResourceManager). This will use Keysight VISA which you have installed. If your instrument is listed in Keysight Connection expert then it will appear. Please note that typically TCPIP instrument are not listed if you do not add them manually but can still be accessed.

lmorningstar13, 2022-06-28:
Thank you! I'll let you know if I have any success

MatthieuDartiailh, 2022-07-01:
Did you fully uninstall Keysight IO Library ? Typically USB-GPIB adaptors only work with the VISA implementation from the manufacturer. If you did not uninstall it you should use Keysight Connection Expert to select which implementation to use for GPIB devices. This can be done from the VISA conflict manager accessible from the gear on the top right of Connection Expert window.

If you did uninstall, does the Tek implementation comes with an interactive control you could use to check your install ? Also did you restart after installing (some VISA implementation requires it and won't work before restarting).

lmorningstar13, 2022-07-06:
I believe I have fully uninstalled Keysight IO library. Just to be careful, I uninstalled all GPIB installations. I then reinstalled Keithley and got an error saying the library was missing so I then reinstalled  Ni-Visa and Ni-488. My most recent error is the following:

VisaIOError: VI_ERROR_RSRC_NFOUND (-1073807343): Insufficient location information or the requested device or resource is not present in the system.

Would you happen to have any suggestions on how to solve this? Anything else I should install?

In case this information is relevant, I can communicate with K6221 using the KI-488 Diagnostic tool (just GPIB command).

MatthieuDartiailh, 2022-07-07:
I read one of your previous message too quickly and thought you had installed VISA from Keithley (and not just the drivers). Typically the VISA implementation needs to match the USB-GPIB adapter. In your case you should install TekVISA (after uninstalling NIVISA). (Tektronik and Keithley are a single company but the VISA  implementation is under Tek name). 

Let me know how it goes.

lmorningstar13, 2022-07-07:
Thank you for the suggestion. I have completely uninstalled NiVisa and have now installed TekVisa. I got a new error which is:
VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid

Thank you so much for your help!

lmorningstar13, 2022-07-08:
Thank you again for your help! I got it to work. I installed Tekvisa system connector and then in my attempts to try anything, I did rm = visa.ResourceManager('@py') again and it worked! Thank you so much for your help!

closed 2022-07-08 by lmorningstar13

MatthieuDartiailh, 2022-07-08:
Ok that's a bit unexpected but if gpib_ctypes (used by the pyvisa-py backend) pick up the right GPIB dll fine. I am still surprised that no other VISA impl managed to pick the right GPIB config though.