pyvisa/pyvisa #678: Problems when using pyvisa1.12 to interact with keysightN5225B in windows

changhexuefei, 2022-07-13:
Hello, during the communication between Windows using Pyvisa1.12 and Keysight instrument N5225B, the error of access violation reading 0x00000176D2B06123 appeared.

MatthieuDartiailh, 2022-07-16:
I will need more details to even have a chance of providing any help.

Can you post the output of pyvisa-info and describe in what situation this happened ?

changhexuefei, 2022-07-16:
When I request curve data and marker data from Keysight N5225B multiple times, there is an access violation exception, but I add the "time.sleep(2)" method after each request, whether my interval is not enough caused by long?

changhexuefei, 2022-07-30:
Hello, I think I should have found the reason for the access violation. I used the HiSLip communication protocol. After I changed the protocol of Instrument (VXI-11), the communication between PyVisa and Keysight N5225B is normal.

closed 2022-08-03 by changhexuefei