pyvisa/pyvisa #680: most VISA info links broken

saltsucker, 2022-07-21:
https://pyvisa.readthedocs.io/en/latest/faq/faq.html#where-can-i-get-more-information-about-visa

The only one that kind of works is: https://www.ivifoundation.org/Downloads/Specifications.htm
But that isn't even correct anymore, because most of those links are broken.
If you visit:
https://www.ivifoundation.org/
you can see it's quite different now.

MatthieuDartiailh, 2022-07-21:
Thanks for the report. Feel free to make a PR if you feel so inclined.

saltsucker, 2022-07-22:
Unfortunately, I am very new to this, so I don't even know what would be what :(. I am still trying to figure what VISA commands I can send through your API. I can't find a single resource that lists out the VISA API. I don't the difference between IVI and VISA and NI-VISA, etc. Idk why it has to all be so convoluted. I look at my PSU's data sheet and there aren't any VISA commands listed in the sheet. Not even the "VISA" keyword. I did your tutorial and I can list resources and get the info about the PSU, but that's it. I just want to figure out how I can turn it off and on, and set voltage and amps. That's it. But, they don't make it easy.