pyvisa/pyvisa #681: Once again viOpenDefaultRM -1073807360

Cpgcode, 2022-07-22:
Python Code:
`import pyvisa

pyvisa.log_to_screen()
rm = pyvisa.ResourceManager()
print(rm.list_resources())`

Reinstalling python didn't fix the issue as mentioned in other Threads. What can I do next?

Log:
```
2022-07-22 09:03:23,207 - pyvisa - DEBUG - No visa library specified, trying to find alternatives.
2022-07-22 09:03:23,223 - pyvisa - DEBUG - Environment variable PYVISA_LIBRARY is unset.
2022-07-22 09:03:23,240 - pyvisa - DEBUG - No user defined configuration
2022-07-22 09:03:23,243 - pyvisa - DEBUG - Automatically found library files: [None, 'C:\\Windows\\system32\\visa32.dll', 'C:\\Windows\\system32\\visa32.dll', 'C:\\Windows\\system32\\visa64.dll', 'C:\\Windows\\system32\\visa64.dll']
2022-07-22 09:03:23,243 - pyvisa - DEBUG - No user defined library files
2022-07-22 09:03:23,244 - pyvisa - DEBUG - The IVI implementation available
2022-07-22 09:03:23,244 - pyvisa - DEBUG - No user defined configuration
2022-07-22 09:03:23,246 - pyvisa - DEBUG - Automatically found library files: [None, 'C:\\Windows\\system32\\visa32.dll', 'C:\\Windows\\system32\\visa32.dll', 'C:\\Windows\\system32\\visa64.dll', 'C:\\Windows\\system32\\visa64.dll']
2022-07-22 09:03:23,247 - pyvisa - DEBUG - No user defined library files
2022-07-22 09:03:23,251 - pyvisa - DEBUG - Library signatures: 90 ok, 0 failed
2022-07-22 09:03:23,252 - pyvisa - DEBUG - Created library wrapper for C:\Windows\system32\visa32.dll
2022-07-22 09:03:23,256 - pyvisa - DEBUG - viOpenDefaultRM('<ViObject object at 0x00000177100F7E40>',) -> -1073807360
```

MatthieuDartiailh, 2022-07-22:
Can you also post the output of `pyvisa-info`?

Cpgcode, 2022-07-22:
> 

yes, edited

MatthieuDartiailh, 2022-07-22:
Thanks since PyVISA cannot figure out what VISA library is installed I would suggest to fully uninstall your VISA library (NI, Keysight, etc), and then re-install it. Also be sure to restart your computer after the installation process since usually the library does not work at all before a restart.

Cpgcode, 2022-09-06:
> 

This didn't work either, seems like reinstalling the OS is the only option left.

MatthieuDartiailh, 2022-09-06:
Before that you can try to re-install Python in case you have a weird issue with ctypes. Did the output of pyvisa-info changed at all ?