pyvisa/pyvisa #683: [COM] Communication issue with thermometer / heat bath using RS-485

juanzen, 2022-07-28:
I am having issues with the read_bytes method specifically, I can get these devices to work on pyserial without any issues so I maybe making a mistake somewhere in my implementation.

to explain the issue simply I am supposed to receive 8 bits back from the device after sending a command, when using pyvisa, the first time I execute read_bytes(8) it will work okay, but any subsequent calls within the same process will have missing bits or misplaced bits(order will switch each repetition after the first one) and sometimes read_bytes will fail because the number of bits it read is less than expected, in this case most of the time it will fail at 5 for some reason.

to give an image of what happens when using the RS-485 thermometer I get the following reply the first time

[1, 3, 4, 2, 8, 0, 234, 251, 198] where 234 is the current temperature(you need to divide by 10 in this case so it is actually 23.4 C)

anyways in my code using pyvisa this will work, but the second time I perform the read in the same process it will time out.. if I change the amount of read bytes it will work, say try to do read_bytes(1).. that will fail with a number lower than the first time

program output as is:

[1, 3, 4, 2, 5, 0, 235, 171]
Traceback (most recent call last):
  File "c:\path\redacted\pythonfile.py", line 98, in <module>
    main()
  File "c:\path\redacted\pythonfile.py", line 92, in main
    reply_test_raw = connect_with_rs485('ASRL3::INSTR', 9600, 1, 'none', 8, test_msg_byte)
  File "c:\path\redacted\pythonfile.py", line 70, in connect_with_rs485
    reply = rs485_dev.read_bytes(8)
  File "C:\Python 3.7\lib\site-packages\pyvisa\resources\messagebased.py", line 371, in read_bytes
    chunk, status = self.visalib.read(self.session, size)
  File "C:\Python 3.7\lib\site-packages\pyvisa\ctwrapper\functions.py", line 2337, in read
    ret = library.viRead(session, buffer, count, byref(return_count))
  File "C:\Python 3.7\lib\site-packages\pyvisa\ctwrapper\highlevel.py", line 222, in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore
  File "C:\Python 3.7\lib\site-packages\pyvisa\highlevel.py", line 251, in handle_return_value
    raise errors.VisaIOError(rv)
pyvisa.errors.VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.

if I comment out the second read call and run the program several times
[Running] python -u ""c:\path\redacted\pythonfile.py"
[1, 3, 4, 2, 4, 0, 235, 250, 5]

[Done] exited with code=0 in 0.923 seconds

[Running] python -u ""c:\path\redacted\pythonfile.py"
[1, 3, 4, 2, 4, 0, 235, 250, 5]

[Done] exited with code=0 in 0.918 seconds

[Running] python -u ""c:\path\redacted\pythonfile.py"
[1, 3, 4, 2, 4, 0, 235, 250, 5]

[Done] exited with code=0 in 0.915 seconds

as far as I can tell, reading only once per run works, so I don't understand the difference between running the read several times in the same process and running the whole script again once.

I have no issues doing consecutive reads in pyserial, but I would learn to be able to do this in pyvisa as well, Thank you for any help and insight you could provide.

I also went as far to compare the waveform itself on an oscilloscope, the command sent through pyvisa and pyserial is identical, so is the reply both times, it is just the read_bytes() method not working as expected when called several times.

here is my code example, I tried to make it as short as possible, but it is still a bit long sorry.

    import pyvisa
    from pyvisa import constants
    import time

    def crc_8(message_list):

        x = 0xFFFF
        for data in message_list:
            x = data ^ x
            for _ in range(8):
                x , carry_bit = x >> 1, x & 1

                if carry_bit == 1:
                    x = x ^ 0xA001
        return x

    def connect_with_rs485(res_name, baudrate: int, stop_bits: int, parity: str, byte_size: int, command_byte_array: bytearray):
        '''
        Communicate with RS-485 device using Pyvisa given settings and a raw byte array for a command,
        this will perform CRC on the message and attach the bits to the end of the command

            inputs:
            res_name = resource name in pyvisa.resourceManager
            baudrate = device baudrate
            stop_ bits = stop bit setting
            parity = parity setting expect 'N' or 'none' for none 'O'/'odd'
                    for odd and anything else for even
            byte_size = bite_size
            command_byte_array = bytearray of the bits in the desired command

            output:
            result = bytearray of the reply from the device given the command

        This function can cause exceptions if the device is not connected
        or if there is a setting mismatch

        '''

        cmd_bytes = command_byte_array
        res_man = pyvisa.ResourceManager()

        rs485_dev = res_man.open_resource(res_name)
        rs485_dev.baudrate = int(baudrate)
        stop_bits = int(stop_bits)

        if stop_bits == 1:
            stop_bits = constants.StopBits.one
        else:
            stop_bits = constants.StopBits.two
        rs485_dev.stop_bits = stop_bits

        if parity == 'N' or parity == 'none' or parity == '':
            parity = constants.Parity.none
        elif parity == 'O' or parity == 'odd':
            parity = constants.Parity.odd
        else:
            parity = constants.Parity.even
        rs485_dev.parity = parity

        rs485_dev.data_bits = int(byte_size)
        crc_int = crc_8(cmd_bytes)
        crc_lsb = crc_int & 0xFF
        crc_msb = (crc_int & 0xFF00) >> 8
        cmd_bytes.append(crc_lsb)
        cmd_bytes.append(crc_msb)
        rs485_dev.timeout = 200
        rs485_dev.write_raw(bytes(cmd_bytes))
        reply = rs485_dev.read_bytes(9)
        rs485_dev.close()
        res_man.close()
        return reply

    def main():

        adr_code = 0x01
        fc_code = 0x03
        start_adrh = 0x00
        start_adrl = 0x00
        datal_h = 0x00
        datal_l = 0x02

        test_msg_list = [adr_code, fc_code, start_adrh, start_adrl, datal_h, datal_l]
        test_msg_byte = bytearray(test_msg_list)

        reply_test_raw = connect_with_rs485('ASRL3::INSTR', 9600, 1, 'none', 8, test_msg_byte)
        reply_test_int = [int(bit) for bit in reply_test_raw]
        print(reply_test_int)
        reply_test_raw = connect_with_rs485('ASRL3::INSTR', 9600, 1, 'none', 8, test_msg_byte)
        reply_test_int = [int(bit) for bit in reply_test_raw]
        print(reply_test_raw)

    if __name__ == "__main__":
        main()

Instrument details
------------------
* Model: Yamato Kagaku heat bath / RS-485 thermometer
* Communication: RS-485 through USB converter DN411I/611I also RS-485 thermometer showed the same issue
 This is the USB-adapter  DSD TECH USB to RS485 RS422 Converter, Windows 10, 8, 7, XP, MAC and Compatible Built-in FTDI: Chip
 I also tested another USB-RS485 adapter and I got the same issue.
* Link to the documentation (if available): 

Output of `pyvisa-info`
-----------------------
Machine Details:
   Platform ID:    Windows-10-10.0.19041-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Python 3.7\python.exe
   Version:        3.7.4
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Jul  8 2019 20:34:20 (#tags/v3.7.4:e09359112e)      
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments

juanzen, 2022-07-28:
adding pertinent manual pages

[image]
[image]
[image]

MatthieuDartiailh, 2022-07-28:
First of all thanks for an amazing issue format. I wish more people would go to the length you did when filling an issue.

More to the point. From the top of my head, VISA only supports RS-232 so you may be able to get things to work on RS-485 since it does not add much on top of a standard serial communication but I cannot be sure. Also since your instrument only support one communication protocol and that is not truly part of VISA I am not sure you have anything to win by usong pyvisa you may be better served by pyserial.

Looking at what you did and the manual apart from specifying that the instrument uses CRC (through the control_flow attribute) I am not sure what else you can try honestly. Do you do anything special in pyserial to account for CRC ?

juanzen, 2022-07-28:
thanks, honestly I mostly followed the base template to be honest though.

To answer your question, In my pyvisa code I am calculating the crc  by hand using the exact same function and way of building the byte array. I am still trying to figure out the actual python crc library. So to my knowledge no, not like I have seen the rs485.py source code carefully

one thing that is different that i don't quite get is that pyserial doesn’t mind the size of the byte read, I could set it to read 100 bytes and it will return the proper size, nine in this case, since there are no termination characters I am not sure how it is getting the right size to return.

because of the nature of the issue and seeing the waveforms, I honestly thought must be using the read method wrong,  “why does it work perfectly the first time? Also can work fine if I run the script itself several times,  but two reads in succession fail” I tried adding a long time in between using time.sleep with some seconds no change either so something changes between the first and second run method…. I know all the write commands work even in succession, because of other commands that have more visible effects. 

I was using pyvisa to control other instruments along this one so I wanted to get everything unified.

Thank you for your insight.

juanzen, 2022-08-04:
I have tried all 4 settings of flow_control, the first read operation will work as expected same as before all subsequent read attempts in the same process will fail.