pyvisa/pyvisa #686: [COM] Communication issue with Tabor AWG

gauravag99, 2022-08-12:
Instrument details
------------------
* Model: Tabor WX2184 AWG
* Communication: TCPIP
* Link to the documentation: PDF attached
[How to Properly Manage Tabor’s AWG Arbitrary Memory.pdf](https://github.com/pyvisa/pyvisa/files/9315584/How.to.Properly.Manage.Tabor.s.AWG.Arbitrary.Memory.pdf)

Issue:
I am trying to upload a waveform into a Tabor AWG. The instrument accepts data in the form:
[image]

I have a 'uint16' numpy array named `final_data` to be loaded into the AWG ready. However, the instrument hangs and stops communication everytime I upload the waveform. The code variations I have tried are the following:

```
# 1st try, here final data is manually converted to from 16bit uints to two 8bit uints
    inst.write(f'*OPC?;:TRAC:DATA#{head1}{head2}')
    inst.write(str(final_data.tobytes())[2:-1]) #remove the " b' " from the numpy array, 

#2nd try, here final data is manually converted to from 16bit uints to two 8bit uints
    inst.write(f'*OPC?;:TRAC:DATA#{head1}{head2}')
    inst.write(final_data.tobytes())

#3rd try, final_data is uint16
    #Leads to hang but can close connection
    inst.values_format.use_binary('d', False, np.array)
    inst.write_binary_values(f'*OPC?;:TRAC:DATA#{head1}{head2}',final_data)

#4th try, final_data is uint16
    inst.write_binary_values(f'*OPC?;:TRAC:DATA#{head1}{head2}',final_data, is_big_endian=False)

#5th try
    inst.write_binary_values(f'*OPC?;:TRAC:DATA#{head1}{head2}', final_data.tobytes(), datatype = 's')  # Gives a packing error in python itself from struc library

# 6th try
    #gives an error in instrument
    inst.values_format.use_binary('B', False, np.array)
    inst.write_binary_values(f'*OPC?;:TRAC:DATA#{head1}{head2}',final_data)

    # inst.write_binary_values(f'*OPC?;:TRAC:DATA#{head1}{head2}', final_data, datatype = 'B') ## THIS HANGS!!!

    # inst.write_binary_values(f'*OPC?;:TRAC:DATA#{head1}{head2}',final_data, datatype = 'B',is_big_endian=False)  ## THIS HANGS!!!!!
```

I would appreciate it a lot if anyone can nudge me towards the right direction as I am new to this.

If anyone has coded for Tabor, would you please mind sharing scripts? My script is attached:

<details>
  <summary>Full Code</summary>
```
import numpy as np
import visa
import matplotlib.pyplot as plt

def check_for_errors(inst,loading=False):
    '''
    This function checks for errors on Tabor.
    If loading data, Tabor might be busy so this function will wait for 10 seconds and try again.
    '''

    inst.timeout = 2000 # just let it be there, no logic
    if loading == True:
        inst.timeout = 10000 #set timeout as 10 seconds, not too much querying
        try:
            print(inst.query(":SYST:ERR?"))
            inst.timeout = 2000  # set it back to 2 sec
        except visa.VisaIOError:
            print("Loading, wait!")
            check_for_errors(inst,loading=True)
        except KeyboardInterrupt:
            sys.exit(0)
    else : print(inst.query(":SYST:ERR?"))

def gen_sin(freq,n_samples,SCLK):
    wvf = np.array([np.sin(2*np.pi*freq*i/eval(SCLK)) for i in range(n_samples)])

    return wvf

def process_wfs_into_one(arraylist,dummypts=16):
    '''
    provided a list of arrays with individual waveforms,
    this will output a singlw waveform will all above appended,
    with # dummy points between all the waveforms,
    such that the points are at voltage level of next waveform.
    '''
    processed_wf = np.array([])
    for i in range(len(arraylist)):
        if (i !=0):
            temp = np.append(np.zeros(16)+arraylist[i][0],arraylist[i])
        else : temp = arraylist[i]
        processed_wf = np.append(processed_wf,temp)

    return processed_wf

def convert_array_to_unsign_n_bits(data,volt_pm_level=1,bits=14):
    '''
    Convert provided data to n bit level, given +- voltage level.
    '''
    data_tp = data/np.max(np.abs(data))
    data_tp = data_tp* 2**bits / (volt_pm_level*2)
    data_tp = (data_tp + 2**(bits-1)).clip(0,2**bits-1)

    return data_tp.astype('uint16')

def convert_data_to_two_byte_words(data):
    '''
    Converts 16 bit words into two 8bit words.
    '''
    newdata = np.zeros((data.shape[0],2),dtype='uint8')
    for i in range(len(data)):
        newdata[i,0] = divmod(data[i],2**8)[0]
        newdata[i,1] = divmod(data[i],2**8)[1]

    return newdata

def interleave_channel_data(datach1,datach2,intrlv_pts=16):
    '''
    Provided data for two channels, it will output interleaved stream with interval
    'intrlv_pts'. Channel 2 always placed in front of channel1.
    data output -> CH2|CH1|CH2|CH1.....
    '''
    datach1 = np.reshape(datach1,(int(datach1.size/intrlv_pts),intrlv_pts))
    datach2 = np.reshape(datach2, (int(datach1.size / intrlv_pts), intrlv_pts))
    data = np.hstack((datach2,datach1))  #first 16 points are to be channel 2
    data = data.flatten(order='C')
    return data

if __name__=='__main__':

    #init settings for tabor
    volt_ampl = 2
    SCLK = '75e6'

    ###################### GENERATE AND PROCESS WAVEFORM ###################
    #Data for channel 1
    #generate sinusoids for channel 1
    wvf1 = gen_sin(10e3,int(2**10),SCLK)
    #combine both waveforms with dummy 16 points in between
    finalwfm = process_wfs_into_one([wvf1,wvf1,wvf1],dummypts=16)
    # data for channel 2
    wvf3 = gen_sin(1e3, int(2**10), SCLK)
    finalwfm2 = process_wfs_into_one([wvf3,wvf3,wvf3],dummypts=16)

    #interleave the waveforms
    interleaved_wfm = interleave_channel_data(finalwfm,finalwfm2)
    #convert to unsigned 16bit
    final_data = convert_array_to_unsign_n_bits(interleaved_wfm)
    #finally convert to somethign you can send to instrument
    final_data = convert_data_to_two_byte_words(final_data)
    final_data = final_data.flatten(order="C")
    final_data = np.zeros_like(final_data)
    ##########################################################################

    rm = visa.ResourceManager()
    inst = rm.open_resource('TCPIP::192.168.0.204::5025::SOCKET',write_termination = '\n',read_termination='\n')
    inst.write("*RST")  # reset system
    inst.write("*CLS")  # clear old errors

    check_for_errors(inst)

    inst.write(":INST:SEL 1")
    inst.write(f":FREQ:RAST {SCLK}")
    inst.write(f":VOLT:LEV:AMPL {volt_ampl}")
    inst.write(":VOLT:LEV:OFFS 0")
    inst.write(":OUTP ON")

    inst.write(":INST:SEL 1")
    inst.write(f":FREQ:RAST {SCLK}")
    inst.write(f":VOLT:LEV:AMPL {volt_ampl}")
    inst.write(":VOLT:LEV:OFFS 0")
    inst.write(":OUTP ON")
    inst.write(':FUNC:MODE USER')
    inst.write(':TRAC:MODE COMB')
    inst.write(':TRAC:DEL:ALL')

    numberofpoints= len(finalwfm)#len(wvf1)+len(wvf2)  # TRY with
    inst.write(f':TRAC:DEF 1,{numberofpoints}')

    inst.write(':TRAC:SEL 1')
    head2 = 2*numberofpoints
    head1 = len(str(head2))

    inst.write(f'*OPC?;:TRAC:DATA#{head1}{head2}')
    inst.write(str(final_data.tobytes())[2:-1])

    inst.write(f'*OPC?;:TRAC:DATA#{head1}{head2}')
    inst.write(final_data.tobytes())

    #Leads to hang but can close connection
    inst.values_format.use_binary('d', False, np.array)
    inst.write_binary_values(f'*OPC?;:TRAC:DATA#{head1}{head2}',final_data)

    inst.write_binary_values(f'*OPC?;:TRAC:DATA#{head1}{head2}',final_data, is_big_endian=False)

    inst.write_binary_values(f'*OPC?;:TRAC:DATA#{head1}{head2}', final_data.tobytes(), datatype = 's')  # Gives a packing error in python itself from struc library

    #gives an error in instrument
    inst.values_format.use_binary('B', False, np.array)
    inst.write_binary_values(f'*OPC?;:TRAC:DATA#{head1}{head2}',final_data)

    # inst.write_binary_values(f'*OPC?;:TRAC:DATA#{head1}{head2}', final_data, datatype = 'B') ## THIS HANGS!!!

    # inst.write_binary_values(f'*OPC?;:TRAC:DATA#{head1}{head2}',final_data, datatype = 'B',is_big_endian=False)  ## THIS HANGS!!!!!

    check_for_errors(inst,loading=True)
    inst.close()
 ```
</details>

MatthieuDartiailh, 2022-08-12:
`write_binary_values` take care of formatting the header for you so doing the following should work:

`inst.write_binary_values(f'*OPC?;:TRAC:DATA',final_data,datatype="H", is_big_endian=False)`

The datatype specify that you are sending unsigned shorts (uint16) and since the manual suggest that high byte should be last.

gauravag99, 2022-08-17:
@MatthieuDartiailh Thanks a lot! This seems to work. I will close this issue.

closed 2022-08-17 by gauravag99

MatthieuDartiailh, 2022-08-17:
Happy to help

MatthieuDartiailh, 2022-08-17:
If you feel so inclined could you let me know if you see anything to improve in the documentation to make things clearer ?