pyvisa/pyvisa #690: Using ResourceManager with NI-VISA?

tstern9, 2022-08-25:
I've downloaded and installed NI-VISA 2022 Q3 (latest as of time of this post). pyvisa seems to default to a DLL file located within the Windows folder. Anyway, the actual VISA drivers that come with the NI package are installed by default in _C:\Program Files\IVI Foundation\VISA\Win64\NIvisa_, at least to best of my understanding (there is a file inside the folder called NiVisaCom.dll).

So anyway, I'm trying this constructor:
`rm = pyvisa.ResourceManager(r"C:\Program Files\IVI Foundation\VISA\Win64\NIvisa")`
And get this error:
Exception has occurred: LibraryError
Error while accessing C:\Program Files\IVI Foundation\VISA\Win64\NIvisa: Could not find module 'C:\Program Files\IVI Foundation\VISA\Win64\NIvisa' (or one of its dependencies). Try using the full path with constructor syntax.

During handling of the above exception, another exception occurred:

  File "G:\My Drive\NES\Code\IEEE488.py", line 3, in <module>
    rm = pyvisa.ResourceManager(r"C:\Program Files\IVI Foundation\VISA\Win64\NIvisa")

My Python info is as follows:
Python 3.10.6 (tags/v3.10.6:9c7b4bd, Aug  1 2022, 21:53:49) [MSC v.1932 64 bit (AMD64)] on win32

I'm not sure what that win32 at the end means to be honest. I'm running Win11 Pro 64bit.

MatthieuDartiailh, 2022-08-26:
Can you post the output of pyvisa-info ? Usually a NI-VISA install does put a visa.dll in the windows folder that forward to the actual dll location. 

If there is none, you will have to specify extra dll paths. See https://docs.python.org/3/library/os.html#os.add_dll_directory if this work you can store this information in a pyvisa configuration file  https://pyvisa.readthedocs.io/en/latest/introduction/configuring.html#configuring-the-ivi-backend.

MatthieuDartiailh, 2022-08-26:
Note also that the RM expect the path to the DLL not to a folder.

tstern9, 2022-08-26:
pyvisa-info output:

Machine Details:
   Platform ID:    Windows-10-10.0.22000-SP0
   Processor:      Intel64 Family 6 Model 151 Stepping 2, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\tstern\AppData\Local\Programs\Python\Python310\python.exe
   Version:        3.10.6
   Compiler:       MSC v.1932 64 bit (AMD64)
   Bits:           64bit
   Build:          Aug  1 2022 21:53:49 (#tags/v3.10.6:9c7b4bd)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 22021376
         Spec. Version: 5244928
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 22021376
         Spec. Version: 5244928

So NI installed it inside the system32 folder, as you said. I've tried using the path for the 64bit file and I get no error.
I'm getting a list of resources (just this really ('ASRL3::INSTR',)). I have an NI GPIB-USB-HS connected via USB and can not see it on the list. DAQMAX driver sees it. What could be wrong?

MatthieuDartiailh, 2022-08-26:
Do you have an instrument connected to it ? list_resources will only list INSTR resources by default so if you have no instrument connected to it it makes sense that you do not see anything. You can try listing all resources by passing `"?*"` to list resources in which case you should see a GPIB INTFC resource.

tstern9, 2022-08-26:
That solved it - Thank you!

closed 2022-08-26 by tstern9