pyvisa/pyvisa #692: [COM] Communication issue with Tabor WX2184 using TCPIP

gauravag99, 2022-09-08:
# Issue

I have to write a table as binary data to the instrument with 4 columns with values :  `<uint32><uint16><uint8><uint8>....`
Since the  `write_binary_values` function only allows for one data type I tried to do the following (minimal example):

```
inst.write("*CLS")  # clear old errors

looparr = np.array([2,3,1],dtype='uint32')
segnumber = np.array([1,2,3],dtype='uint16')
jumpflag = np.array([0,0,0],dtype='uint8')
dummyarr = np.array([0,0,0],dtype='uint8')

testseqlist = [ looparr[0], segnumber[0], jumpflag[0], dummyarr[0],   #this is the four column table
                looparr[1], segnumber[1], jumpflag[1], dummyarr[1],
                looparr[2], segnumber[2], jumpflag[2], dummyarr[2]]

inst.write('*OPC?;:SEQ:DATA#224',termination=None)
for i in range(0,12,4):
    inst.write_raw(message=testseqlist[i].tobytes())
    inst.write_raw(message=testseqlist[i+1].tobytes())
    inst.write_raw(message=testseqlist[i+2].tobytes())
    inst.write_raw(message=testseqlist[i+3].tobytes())
```

and also tried this:

```
inst.write('*OPC?;:SEQ:DATA#224',termination=None)#,values=None,header_fmt='empty'
for i in range(0,12,4):
    inst.write_binary_values(message='',values=testseqlist[i],datatype="I",header_fmt='empty',termination=None)
    inst.write_binary_values(message='',values=testseqlist[i+1],datatype="H",header_fmt='empty',termination=None)
    inst.write_binary_values(message='',values=testseqlist[i+2],datatype="B",header_fmt='empty',termination=None)
    inst.write_binary_values(message='',values=testseqlist[i+3],datatype="B",header_fmt='empty',termination=None)
```

Please let me know what can be done to send the data

Instrument details
------------------
* Model: Tabor AWG WX2184
* Communication: TCPIP
* Link to the documentation (if available): 
[How to Properly Manage Tabor’s AWG Arbitrary Memory.pdf](https://github.com/pyvisa/pyvisa/files/9523866/How.to.Properly.Manage.Tabor.s.AWG.Arbitrary.Memory.pdf)

Part of the documentation germane to the issue:
[image]
 

Output of `pyvisa-info`
-----------------------
```
Machine Details:
   Platform ID:    Windows-10-10.0.10240-SP0
   Processor:      Intel64 Family 6 Model 158 Stepping 10, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\ProgramData\Miniconda3\python.exe
   Version:        3.9.12
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Apr  4 2022 05:22:27 (#main)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
   py:
      Could not instantiate backend
      -> Wrapper not found: No package named pyvisa_py
```

MatthieuDartiailh, 2022-09-13:
Even if you do not include the the termchar most instruments do not take well to multiple writes. So I would suggest you try the following:
```python
inst.write("*CLS")  # clear old errors

looparr = np.array([2,3,1],dtype='uint32')
segnumber = np.array([1,2,3],dtype='uint16')
jumpflag = np.array([0,0,0],dtype='uint8')
dummyarr = np.array([0,0,0],dtype='uint8')

data = bytes()
for i in range(4):
    data += looparr[i].tobytes()
    data += seqnumber[i].tobytes()
    data += jumpflag[i].tobytes()
    data += dummyarr[i].tobytes()

inst.write_binary_values('*OPC?;:SEQ:DATA', data, datatype="b")
```

gauravag99, 2022-10-04:
Hey thanks! This seems to work brilliantly @MatthieuDartiailh

closed 2022-10-04 by gauravag99