pyvisa/pyvisa #695: Pyvisa unable to detect Keithley KUSB-488B (GPIB to USB) adaptor for SR830 lock-in amplifier although the connection is listed in Device Manager

juggernaut545, 2022-09-29:
Hello. I am new to Pyvisa, NI Visa, and related software so please bear with my lack of technical knowledge.

I have an SR830 Lock-In Amplifier which I am attempting to connect to my computer so that I can read and write data to and from it. I connected the computer with the instrument by using a KUSB-488B wire, with the GPIB side attached to the instrument and the USB attached to the computer. This connection is listed in device manager, as shown below. However, when I run the following basic code 
```
import pyvisa
rm = pyvisa.ResourceManager()
print(rm.list_resources()
```

I am met with the following 
`()`

Here is a screenshot showing that this connection is recognized by the computer itself in device manager. 
[image]

I let Windows automatically search for new versions of the driver, and it seems to be updated. Here is a screenshot showing that the computer is not finding any errors with it. 
[image]

I additionally installed NI-MAX and NI-Visa. However, I recently realized after reading many articles that this Keithley connection would not show up in NI-Max. In spite of this, NI-Visa is not working for me as upon startup I am met with the following error 
`VISA: (Hex 0xBFFF0011) Insufficient location information or the device or resource is not present in the system.`
I am disregarding this error for now, as I do not think it should affect the current problem.

However, I am still perplexed as to why Pyvisa is not detecting this connection if the device manager is. 
For reference, here is the output when I run `pyvisa-info`
```
Machine Details:
   Platform ID:    Windows-10-10.0.22000-SP0
   Processor:      Intel64 Family 6 Model 166 Stepping 0, GenuineIntel

Python:
   Implementation: CPython
   Executable:     C:\Users\laqm\anaconda3\python.exe
   Version:        3.9.12
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Apr  4 2022 05:22:27 (#main)
   Unicode:        UCS4

PyVISA Version: 1.12.0

Backends:
   ivi:
      Version: 1.12.0 (bundled with PyVISA)
      #1: C:\Windows\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 23069952
         Spec. Version: 5244928
      #2: C:\Windows\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: 23069952
         Spec. Version: 5244928
   py:
      Version: 0.5.3
      ASRL INSTR: Available via PySerial (3.5)
      USB INSTR: Available via PyUSB (1.2.1). Backend: N/A
      USB RAW: Available via PyUSB (1.2.1). Backend: N/A
      TCPIP INSTR: Available
      TCPIP SOCKET: Available
      GPIB INSTR: Available via Linux GPIB (None)
      GPIB INTFC: Available via Linux GPIB (None)

```

I appreciate all and any help. Please let me know what additional information I should provide. 

Thank you.

MatthieuDartiailh, 2022-09-29:
USB-to-GPIB adapters are tricky and typically only work with the VISA implementation of their manufacturer. Try uninstalling NI-VISA and installing TekVISA instead. Be sure to cleanly uninstall and restart after uninstalling and after installing the new version.

juggernaut545, 2022-09-29:
Thank you for your response. I greatly appreciate it.
I apologize if this is off-topic, but do you have any general suggestions as to what to do if the TekVisa software is not detecting the instrument either? I at first installed a version that wasn't the newest, so then I installed the newer 2019 version and restarted my computer, but it still does not seem to be showing the instrument. 
Thank you.

MatthieuDartiailh, 2022-09-29:
Can you post again the output of pyvisa-info ?

Does TekVISA comes with something similar to NI-Max or Keysight connection expert where you can see the instruments it detects ?

It may be worth checking if GPIB specific stuff is not part of a separate installer.

juggernaut545, 2022-09-29:
Here is the output:
[image]

TekVISA comes with the OpenChoice Instrument manager which is similar to the NI-Max interface in that I can see which instruments are detected. I attached a picture of it below.
[image]
The instrument should show up in the left window but nothing is being detected. 

I will look into whether the GPIB related things are part of a separate installer, that's a good idea.

Thanks for the help so far.

LongnoseRob, 2022-09-29:
I would recommend to try this driver package for the KSUB-488B
https://www.tek.com/en/support/software/driver/4-0-0

It could also be worth trying a different USB port or a powered USB-Hub for the KUSB-488B

juggernaut545, 2022-10-05:
Thank you for all your help so far.
I was able to get the connection to work by using another computer, as I believe the issue may have been in multiple installations of drivers and the fact that it was Windows 11, since some of the drivers might have only been compatible with Windows 10 and earlier. 
However, I am having another problem with pyvisa.

Below is the code that I am using and the file that it references. I have highlighted the relevant parts (the only parts in the file where visa is called). 
```
import pyvisa
from labdrivers.srs import Sr830

sr830_gpib_address = 8

lock_in = Sr830(sr830_gpib_address)

```
[image]

When I attempt to run a file that has a constructor which inputs a parameter into this class, I receive this error. 
[image]
Even if I change the code in this file to `pyvisa` instead of `visa`, it still displays the same error. I have searched online but none of the solutions seem to fix this issue. Do you have any suggestions as to what I should do? I have uninstalled and reinstalled pyvisa. Could it be due to multiple installations? I realized that I might have had different versions of pyvisa installed, so I uninstalled pyvisa from the anaconda terminal using pip, and also uninstalled it from the control panel on the computer directly. I reinstalled it using `pip install pyvisa`, but the issue is still present. Here is the output of `pyvisa-info`
[image]

I appreciate all and any suggestions.
Please let me know any further information you need.
Thank you.

MatthieuDartiailh, 2022-10-05:
It should be `pyvisa` instead of `visa` (possible conflict with the VISA payment package). If you do the change does the error message state that pyvisa does not have a ResourceManager attribute ? if it still says visa, something odd is going on.

juggernaut545, 2022-10-07:
I tried to edit the other file, but now I am met with a new error (I apologize for this as it seems like a goose chase).
I have looked on the project issues as there was a post about this, but I was not sure what steps to take. 
I am running this simple code and the error I am met with is this.

```
import pyvisa

rm = pyvisa.ResourceManager()

print(rm.list_resources())

```

(Beginning of Error)
```
Traceback (most recent call last):

  File ~\untitled0.py:12 in <module>
    print(rm.list_resources())

  File ~\anaconda3\lib\site-packages\pyvisa\highlevel.py:3129 in list_resources
    return self.visalib.list_resources(self.session, query)

  File ~\anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py:296 in list_resources
    raise e

  File ~\anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py:290 in list_resources
    ) = self._find_resources(  # type: ignore

  File ~\anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py:782 in _find_resources
    ret = library.viFindRsrc(

  File ~\anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py:226 in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore

  File ~\anaconda3\lib\site-packages\pyvisa\highlevel.py:251 in handle_return_value
    raise errors.VisaIOError(rv)

VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.

runcell(0, 'C:/Users/nidhi/untitled0.py')
Traceback (most recent call last):

  File ~\untitled0.py:12 in <module>
    print(rm.list_resources())

  File ~\anaconda3\lib\site-packages\pyvisa\highlevel.py:3129 in list_resources
    return self.visalib.list_resources(self.session, query)

  File ~\anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py:296 in list_resources
    raise e

  File ~\anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py:290 in list_resources
    ) = self._find_resources(  # type: ignore

  File ~\anaconda3\lib\site-packages\pyvisa\ctwrapper\functions.py:782 in _find_resources
    ret = library.viFindRsrc(

  File ~\anaconda3\lib\site-packages\pyvisa\ctwrapper\highlevel.py:226 in _return_handler
    return self.handle_return_value(session, ret_value)  # type: ignore

  File ~\anaconda3\lib\site-packages\pyvisa\highlevel.py:251 in handle_return_value
    raise errors.VisaIOError(rv)

VisaIOError: VI_ERROR_INV_OBJECT (-1073807346): The given session or object reference is invalid.
```

MatthieuDartiailh, 2022-10-07:
Why are you back with NI-VISA ? I have a feeling that the driver for your adapter is doing something quite unexpected and this is causing the crash.

juggernaut545, 2022-10-10:
I was able to resolve the errors. I reinstalled NI-Visa, restarted my computer, and uninstalled and reinstalled Pyvisa several times. It may have been something with the driver acting up, but I am not completely sure. There was an issue with the .dll win32 installation, which I reinstalled as well which fixed it. I am now able to connect successfully to my machine and read and write values from it. 
Thank you for all your help. I greatly appreciate it as having someone directly to relate your issues to as a newcomer is much more helpful and supportive.

MatthieuDartiailh, 2022-10-10:
Great could you maybe make a PR against the doc explaining how you got things to work (in the FAQ section). I think it is not the first time Keithley adapters have proven hard to get going.

juggernaut545, 2022-10-10:
Is the FAQ you are referring to [this one](https://pyvisa.readthedocs.io/en/latest/faq/index.html)?

MatthieuDartiailh, 2022-10-10:
Yes we could have a section about USB-GPIB adapters with a dedicated entry for Tek ones since they seem a bit peculiar.

juggernaut545, 2022-10-10:
I am still getting used to github, so should I edit and propose changes in this specific [section](https://github.com/pyvisa/pyvisa/blob/main/docs/source/faq/faq.rst)?

juggernaut545, 2022-10-11:
I have created a pull request for adding the GPIB-USB error to the FAQ.

closed 2022-11-11 by juggernaut545