pyvisa/pyvisa #700: Unexpected ATN assert and Unlisten command during read_bytes

abergerSRS, 2022-10-25:
I am communicating with an instrument using the `GPIBInterface` and am issuing a `read_bytes(count)` call to read out a byte stream from an instrument located at address 16. I have found that *during the read operation*, the GPIB controller asserts the ATN line and issues the following commands: `UNL`, `LA0`, `TA16`, and `SC0` (UNListen, Listener Address 0, Talk Address 16, SeCondary Address 0). This interrupts the byte stream and causes the controller to miss bytes from the Talker's byte stream. This sequence of events occurs randomly/unpredictably during the `read_bytes` operation. What would cause this sequence of commands?

I will note that, in my case `count` can be quite large, so immediately before the `read_bytes`, I adjust the `GPIBInterface.chunk_size` to make sure that for `count` > default chunk size (20480), there is no need to perform any low level operations to expand the buffer size. 

To Reproduce
--------------
1. Open a GPIBInterface resource: `sr620 = rm.open_resource("GPIB0::16::INSTR")`. In my case, the instrument at address 16 is an SRS SR620.
2. Expand the `chunk_size` appropriately: `sr620.chunk_size = 32768`
3. Increase the timeout to make we have waited long enough to capture all 32000 bytes: `sr620.timeout = 6000`
4. Issue a "binary dump" command to the SR620: `sr620.write_raw('BDMP 4000')` to stream 4000 measurements (32000 bytes) as binary data. I expect these measurements to come in at about 1 kHz (i.e. receiving 8 bytes of data every 1 ms)
5. Issue a read command: `sr620.read_bytes(32000)`
6. Capture the GPIB Traffic using NI's GPIB Analyzer to search for the `UNL`, `LA0`, `TA16`, `SC0` sequence of commands.  

| Event # | min     | s    | ms     | us  | ns  | A | H | 87654321 | EOI | ATN | SRQ | REN | IFC | NRFD | NDAC | DAV | D |  |
| -------- | ---       | --- | ---     | ---  | --- | -- | -- | ---------- | ---  | ---   | ---    | ----- | --- | ------- | ------ | ----- | -- | -- | 
| 21804   | 0         | 0    | 1       | 130 | 850 |  | 0 | 00000000 | 0 | 0 | 0 | 1 | 0 | 0 | 1 | 1 | DAB | DAV^ |
| 21805   | 0         | 0    | 0       | 0   | 100 |  | 0 | 00000000 | 0 | 0 | 0 | 1 | 0 | 1 | 1 | 1 |  | NRFD^ |
| 21806   | 0         | 0    | 0       | 0   | 150 |  | 0 | 00000000 | 0 | 0 | 0 | 1 | 0 | 1 | 0 | 1 |  | NDACv |
| 21807   | 0         | 0    | 0       | 0   | 100 |  | 0 | 00000000 | 0 | 0 | 0 | 1 | 0 | 1 | 0 | 0 |  | DAVv |
| 21808   | 0         | 0    | 0       | 0   | 100 |  | 0 | 00000000 | 0 | 0 | 0 | 1 | 0 | 1 | 1 | 0 |  | NDAC^ |
| 21809   | 0         | 0    | 0       | 0   | 200 |  | 0 | 00000000 | 0 | 0 | 0 | 1 | 0 | 0 | 1 | 0 |  | NRFDv |
| 21810   | 0         | 0    | 0       | 10  | 800 | $ | 24 | 00100100 | 0 | 0 | 0 | 1 | 0 | 0 | 1 | 1 | DAB | DAV^ |
| 21811   | 0         | 0    | 0       | 0   | 100 | $ | 24 | 00100100 | 0 | 0 | 0 | 1 | 0 | 1 | 1 | 1 |  | NRFD^ |
| 21812   | 0         | 0    | 0       | 0   | 150 | $ | 24 | 00100100 | 0 | 0 | 0 | 1 | 0 | 1 | 0 | 1 |  | NDACv |
| 21813   | 0         | 0    | 0       | 0   | 100 | $ | 24 | 00100100 | 0 | 0 | 0 | 1 | 0 | 1 | 0 | 0 |  | DAVv |
| 21814   | 0         | 0    | 0       | 0   | 100 | $ | 24 | 00100100 | 0 | 0 | 0 | 1 | 0 | 1 | 1 | 0 |  | NDAC^ |
| 21815   | 0         | 0    | 0       | 0   | 200 | $ | 24 | 00100100 | 0 | 0 | 0 | 1 | 0 | 0 | 1 | 0 |  | NRFDv |
| 21816   | 0         | 0    | 0       | 10  | 800 |  | a | 00001010 | 0 | 0 | 0 | 1 | 0 | 0 | 1 | 1 | DAB | DAV^ |
| 21817   | 0         | 0    | 0       | 0   | 100 |  | a | 00001010 | 0 | 0 | 0 | 1 | 0 | 1 | 1 | 1 |  | NRFD^ |
| 21818   | 0         | 0    | 0       | 0   | 150 |  | a | 00001010 | 0 | 0 | 0 | 1 | 0 | 1 | 0 | 1 |  | NDACv |
| 21819   | 0         | 0    | 0       | 0   | 100 |  | a | 00001010 | 0 | 0 | 0 | 1 | 0 | 1 | 0 | 0 |  | DAVv |
| 21820   | 0         | 0    | 0       | 0   | 100 |  | a | 00001010 | 0 | 0 | 0 | 1 | 0 | 1 | 1 | 0 |  | NDAC^ |
| 21821   | 0         | 0    | 1       | 984 | 300 |  | 0 | 00000000 | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 0 |  | NRFDv |
| 21822   | 0         | 0    | 0       | 101 | 100 | ? | 3f | 00111111 | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 1 | UNL | DAV^ |
| 21823   | 0         | 0    | 0       | 0   | 100 | ? | 3f | 00111111 | 0 | 1 | 0 | 1 | 0 | 1 | 1 | 1 |  | NRFD^ |
| 21824   | 0         | 0    | 0       | 0   | 700 | ? | 3f | 00111111 | 0 | 1 | 0 | 1 | 0 | 1 | 0 | 1 |  | NDACv |
| 21825   | 0         | 0    | 0       | 0   | 100 |  | 20 | 00100000 | 0 | 1 | 0 | 1 | 0 | 1 | 0 | 0 |  | DAVv |
| 21826   | 0         | 0    | 0       | 0   | 100 |  | 20 | 00100000 | 0 | 1 | 0 | 1 | 0 | 1 | 1 | 0 |  | NDAC^ |
| 21827   | 0         | 0    | 0       | 0   | 200 |  | 20 | 00100000 | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 0 |  | NRFDv |
| 21828   | 0         | 0    | 0       | 3   | 550 |  | 20 | 00100000 | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 1 | LA0 | DAV^ |
| 21829   | 0         | 0    | 0       | 0   | 100 |  | 20 | 00100000 | 0 | 1 | 0 | 1 | 0 | 1 | 1 | 1 |  | NRFD^ |
| 21830   | 0         | 0    | 0       | 0   | 600 |  | 20 | 00100000 | 0 | 1 | 0 | 1 | 0 | 1 | 0 | 1 |  | NDACv |
| 21831   | 0         | 0    | 0       | 0   | 100 |  | 0 | 00000000 | 0 | 1 | 0 | 1 | 0 | 1 | 0 | 0 |  | DAVv |
| 21832   | 0         | 0    | 0       | 0   | 100 |  | 0 | 00000000 | 0 | 1 | 0 | 1 | 0 | 1 | 1 | 0 |  | NDAC^ |
| 21833   | 0         | 0    | 0       | 0   | 200 |  | 0 | 00000000 | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 0 |  | NRFDv |
| 21834   | 0         | 0    | 0       | 13  | 850 | P | 50 | 01010000 | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 1 | TA16 | DAV^ |
| 21835   | 0         | 0    | 0       | 0   | 100 | P | 50 | 01010000 | 0 | 1 | 0 | 1 | 0 | 1 | 1 | 1 |  | NRFD^ |
| 21836   | 0         | 0    | 0       | 0   | 700 | P | 50 | 01010000 | 0 | 1 | 0 | 1 | 0 | 1 | 0 | 1 |  | NDACv |
| 21837   | 0         | 0    | 0       | 0   | 100 | \` | 60 | 01100000 | 0 | 1 | 0 | 1 | 0 | 1 | 0 | 0 |  | DAVv |
| 21838   | 0         | 0    | 0       | 0   | 100 | \` | 60 | 01100000 | 0 | 1 | 0 | 1 | 0 | 1 | 1 | 0 |  | NDAC^ |
| 21839   | 0         | 0    | 0       | 0   | 200 | \` | 60 | 01100000 | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 0 |  | NRFDv |
| 21840   | 0         | 0    | 0       | 0   | 950 | \` | 60 | 01100000 | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 1 | SC0 | DAV^ |
| 21841   | 0         | 0    | 0       | 0   | 100 | \` | 60 | 01100000 | 0 | 1 | 0 | 1 | 0 | 1 | 1 | 1 |  | NRFD^ |
| 21842   | 0         | 0    | 0       | 0   | 600 | \` | 60 | 01100000 | 0 | 1 | 0 | 1 | 0 | 1 | 0 | 1 |  | NDACv |
| 21843   | 0         | 0    | 0       | 0   | 100 |  | 0 | 00000000 | 0 | 1 | 0 | 1 | 0 | 1 | 0 | 0 |  | DAVv |
| 21844   | 0         | 0    | 0       | 0   | 100 |  | 0 | 00000000 | 0 | 1 | 0 | 1 | 0 | 1 | 1 | 0 |  | NDAC^ |
| 21845   | 0         | 0    | 0       | 0   | 200 |  | 0 | 00000000 | 0 | 1 | 0 | 1 | 0 | 0 | 1 | 0 |  | NRFDv |
| 21846   | 0         | 0    | 0       | 46  | 350 | ~ | 7e | 01111110 | 0 | 0 | 0 | 1 | 0 | 1 | 1 | 0 |  | NRFD^ |
| 21847   | 0         | 0    | 0       | 47  | 0   | ~ | 7e | 01111110 | 0 | 0 | 0 | 1 | 0 | 0 | 1 | 0 |  | NRFDv |
| 21848   | 0         | 0    | 0       | 0   | 100 | ~ | 7e | 01111110 | 0 | 0 | 0 | 1 | 0 | 0 | 1 | 1 | DAB | DAV^ |

Note that ATN is asserted in Row 21821, and the controller commands are issued after that.

I am using an NI GPIB-USB-HS+ adapter to communicate with the SR620 over GPIB.

I have also tried to set `enable_repeat_addressing = False` and `enable_unaddressing = False`. This seems to make no difference.

**Output of `pyvisa-info`**

```
Machine Details:
   Platform ID:    Windows-10-10.0.19041-SP0
  Processor:      Intel64 Family 6 Model 60 Stepping 3, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\aberger\documents\projects\sr542\tcal-venv\scripts\python.exe
   Version:        3.9.4
   Compiler:       MSC v.1928 64 bit (AMD64)
   Bits:           64bit
   Build:          Apr  6 2021 13:40:21 (#tags/v3.9.4:1f2e308)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
```

MatthieuDartiailh, 2022-10-25:
Could you try to also use NI IO monitor to capture VISA traffic ? I would like to know if the GPIB traffic originates from an explicit VISA command or no.

abergerSRS, 2022-10-25:
Attached is the IO Trace. 

The `BDMP` command is issued at step 10. The initial `viRead` command (with a count of 32008) is issued immediately after that with a timestamp of 10:24:38.1899. `viRead` commands are then re-issued about every 200 ms, with a count that decreases each time. 

Note that using IO Trace is much more invasive. Without IO Trace, the binary dump completes in about 4s (4000 measurements*0.001 s/measurement). With IO Trace, the binary dump takes nearly 10 seconds, such that I am missing many more measurements with IO Trace recording enabled, so I'm not certain that the same sequence of events takes place. Nevertheless, I don't see any explicit VISA commands for the ATN assertion or subsequent commands.

[GPIB_IO_Trace.txt](https://github.com/pyvisa/pyvisa/files/9862593/GPIB_IO_Trace.txt)

MatthieuDartiailh, 2022-10-25:
You are seeing multiple viRead because the amount of data you are trying to read exceed the chunk_size used by pyvisa. You can try increasing it and see if it helps.

MatthieuDartiailh, 2022-10-25:
Also can you update to 1.12 ? With it you should be free of the spurious secondary address commands.

abergerSRS, 2022-10-25:
As I mentioned in the original post, I set the `chunk_size` large enough to capture the entire data set before performing `read_bytes`.

I did update to v1.12, and that does seem to get rid of spurious `UNL + LA0 + TA16 + SC0` commands. Thank you. However, I am still occasionally missing data points. Now it appears that the culprit is that the listener (in this case the host PC) is slow to de-assert NRFD in the middle of one of the 8-byte transfers.

[image]

Here you can see an 8-byte transfer starts at Event # 107162 (almost 1 ms has elapsed since the previous event, as expected for the 1 kHz events). 3 bytes are transferred (0x00, 0x48, and 0x0a), but then at event 107179 it takes 2.194 ms to de-assert the NRFD handshaking line, which means I have missed at least one event (likely two) while processing the current 8-byte transfer. Usually the NRFD de-assert happens about 200 ns after the NDAC assert (for example, see event 107173).

Any ideas as to who the culprit might be? The NI GPIB-USB-HS+ device? Windows? PyVISA?

MatthieuDartiailh, 2022-10-25:
During the read PyVISA is simply waiting for the NI-VISA library to be done and what you observe does not look like a misconfiguration like the previous issue, so I would say PyVISA is not responsible here.

If you can it might be worth testing with a GPIB PCIe board or a different GPIB adapter. But I have keener guess at to what  part might be responsible.

abergerSRS, 2022-10-25:
I was able to get ahold of a NI PCI-GPIB card. It does seem like the missed events occur less frequently, but they are still occurring. Unfortunately, since it's not PCI-GPIB+, I am unable to use GPIB Analyzer to diagnose the fault.

abergerSRS, 2022-10-26:
Sorry, I don't follow your last comment. Do you have any other hints as to who might be holding up the NRFD line? The NI GPIB-USB-HS+ and PCI-GPIB data rates are [spec'd](https://www.ni.com/docs/en-US/bundle/gpib-hw-seri/resource/370426t.pdf) at 1830 kbytes/s and 1536 kbytes/s, respectively, so neither should have any trouble with 8 kbytes/s as I'm trying to accomplish in this example. 

Meanwhile, the SR620 which is sending the data should be able to transmit data at about 1333 measurements/s (with each measurement being 8 bytes, this translates to about 10 kbytes/s, so it should also be able to keep up with this measurement example). And since it's the Listener who is responsible for de-asserting the NRFD line, I have to suspect the NI GPIB cards or Windows.

MatthieuDartiailh, 2022-10-26:
My main point was that the culprit was not pyvisa.

Given your application you may be better served by directly using low level GPIB commands and reading through a INTFC resource. I know this was discussed before and motivated my work on #552 but I do not have time to dig it up.